synthetos/g2 #44: G-code arc planning broken in edge

ryansturmer, 2015-01-30:
Planning for g-code arcs appears to be broken in edge.  

Two files are linked below.  The first is a simple circle, drawn as four g-code arcs.  The second is the same circle, done as an interpolated polyline.

The polyline works fine, but the one composed of the arcs does not work.  This file previews and runs correctly in other tools, and looks to be valid.

This is the file with the arcs:  https://gist.github.com/ryansturmer/3d240a58203675bef997

This is the file with the segments: https://gist.github.com/ryansturmer/7d9d0e487f9fac3a0ace

The arc file appears to only draw a portion of the complete circle.  The image below details the behavior:

[image]

The dotted lines are drawn in by hand after the job had completed.  The solid lines were drawn by the tool.

cmcgrath5035, 2015-02-04:
What build did you get from edge? When I tried 9 days ago, Edge would not build. Master now builds 71.02. Have you tried that?

ryansturmer, 2015-02-04:
Edge build just fine for me.  We can't use master ad it only has the single usb channel so the feedhold performance is poor.  I can do a build for comparison though

cmcgrath5035, 2015-02-04:
"Edge" is a dynamic term - when you got it to build, what build # did you get? If you are using dual USB, your build is likely >71.02, although 71.02 does show up as two USBs, on Linux /dev/ttyACM0 and /dev/ttyACM1.
Master is, of course, dynamic as well; 9 days ago, Master was compiling  build 71.02. In December 2014, Master was building 34.something, very out of date.

Just in case Edge still won't build for me (I give it a try), perhaps it is my build system.
What OS are you building on?

ryansturmer, 2015-02-04:
To be completely unambiguous, I built this exact code:

https://github.com/ShopBotTools/FabMo-G2-Core/commit/26262d85c59502a6ce25b8f5f4a04df123f344bd

It builds successfully for me under Linux and OS X (mavericks)

It is the same code that I encountered the planner failure described in https://github.com/synthetos/g2/issues/45

Which reports itself as 75.02

cmcgrath5035, 2015-02-04:
Hmm, I tried building Edge from the Synthetos Git again, still no joy.
From the git Logs. looks like it is 74.02 as well.
Perhaps someone fixed something over at ShopBotTools.

Way beyond my ability to help - I'll watch to see how Rob or Alden respond.

ryansturmer, 2015-02-04:
Doubt we've fixed anything.  Our edge branch is pretty much concurrent with synthetos.  If you open an issue and post your build log, I may be able to help.  Rob and Alden are pretty responsive, but also pretty busy... this is an area at least where I'm capable of helping out.

cmcgrath5035, 2015-02-04:
Ah,so, so you are ShopBotTools :)

I am goung to do a clean clone of the current edge, perhaps something I 
did mucked it up.
If that won't build, I'll be back.

Tool chain works, Master built just fine.

giseburt, 2015-02-04:
@cmcgrath5035 Are you building on Linux? There's a glitch in the linux arm-none-gcc that I haven't had time to hunt down that may be making the binaries unusable. If so, you may try switching to a newer or older arm-none-eabi-gcc (in the Tools directory Makefile). That's OT for this issue, but I wanted to make sure it was mentioned.

cmcgrath5035, 2015-02-04:
Thanks for heads up Rob. Yes Linux. Master built OK 9 days ago. And that binary ran, and was identical to the 71.02 I built when it was Edge back in Dec.
I'll report back after a clean clone.

giseburt, 2015-02-04:
@cmcgrath5035 Perhaps start a new issue so we can track it separately?

aldenhart, 2016-02-23:
Please use edge build 089 or later

closed 2016-02-23 by aldenhart