synthetos/g2 #45: Motion runaway while running a file

ryansturmer, 2015-02-03:
I had motion control run away on me today, in a not-spectacular but not-ideal fashion.  This was a g-code file generated by v-carve pro, and consisted only of G1 segments, plus a few G0 jogs and some setup. The file failed in the vicinity of line 29743 (of 31164) - this was the last line number reported by G2 when it stopped, but the line that tripped it up, if that was the cause, may have occurred before that.    This was the expected behaivor:

[image]

And this was the outcome (See the departure from the intended toolpath towards the bottom right):

[image]

AT the point of the deviation the tool appeared to stop moving, but on closer inspection, it was seen to be moving very slowly in the direction of the slash (you can tell it was moving slowly by the burning of the wood in that area)  It eventually did stop, though it did not escape the run state, and the line number has ceased to advance, with the position of the tool remaining fixed.

FWIW, the move appears to be tangent to the move it was intending to make at that time.  There are no arcs in the file, just coordinated moves and jogs.

One last thing, the engine (which is the process managing the communication with G2) spit this up, the only error message since the start of the file:

```
Feb 03 06:48:29 FabMo_dev_006 node[138]: info: Running file /opt/fabmo/files/catplex.nc [machine]
Feb 03 06:48:29 FabMo_dev_006 node[138]: info: Got a machine state change: running [machine]
Feb 03 06:57:29 FabMo_dev_006 node[138]: error: -1,JSON_PARSE_ERROR,Could not parse response: '
Feb 03 06:57:29 FabMo_dev_006 node[138]: error: -1,JSON_PARSE_ERROR,Could not parse response: '
Feb 03 06:57:29 FabMo_dev_006 node[138]: [62B blob data]
Feb 03 06:57:29 FabMo_dev_006 node[138]: [62B blob data]
Feb 03 06:57:29 FabMo_dev_006 node[138]: [58B blob data]
Feb 03 06:57:29 FabMo_dev_006 node[138]: [58B blob data]
Feb 03 06:57:29 FabMo_dev_006 node[138]: error: -1,JSON_PARSE_ERROR,Could not parse response: '
Feb 03 06:57:29 FabMo_dev_006 node[138]: error: -1,JSON_PARSE_ERROR,Could not parse response: '
```

The JSON_PARSE_ERROR is emitted whenever the g2 module doesn't unerstand a message that came from the board.  Looks like the board crashed and was sending out some garbage?  I am re-running this file in a less EMI-laden environment to see if the system fails in the same place.

ryansturmer, 2015-02-03:
This testing was performed on the current edge branch.

ryansturmer, 2015-02-03:
g2 configuration:

https://gist.github.com/ryansturmer/58c4d2aabfe1fba093ab

ryansturmer, 2015-02-03:
Running this a second time, with the same G2 configuration failed on a much earlier line, with a repeated status "Failed to get planner buffer"  - Feels like it might be a memory issue?

ryansturmer, 2015-02-03:
Incidentally, I had turned up the logging for that second run.  The "failed to get planner" could have been happening on the first run too, and I wouldn't have seen it.

ryansturmer, 2015-02-03:
So yet another one bites the dust, same circumstances.  I am running this one headless (no CNC machine) and running it from my macbook rather than from the SBC.  This time, the error message sent back is garbled (but not immediately, the error was reported dozens of times before the first garbage message came in.  After the garbage message, the system stopped responding.)

[image]

ryansturmer, 2015-02-03:
The issue appears to be that a hard alarm is called in the motion planner that provokes an infinite recursion that makes the system asplode.

It appears that the planner runs into a full buffer here:

https://github.com/ShopBotTools/FabMo-G2-Core/blob/edge/TinyG2/planner.cpp#l211

Provoking a shutdown of the spindle here: 

https://github.com/ShopBotTools/FabMo-G2-Core/blob/edge/TinyG2/canonical_machine.cpp#l641

Which calls the `mp_queue_command`  function again, here:

https://github.com/ShopBotTools/FabMo-G2-Core/blob/edge/TinyG2/spindle.cpp#L100

Which triggers another alarm, causing the system to go into an infinite recursion.

What I don't understand is how the planner is getting to the full state when it's supposed to be checked ouside the `mp_queue_command` function.

I am continuing to dig!

ryansturmer, 2015-02-03:
At the time of the crash, the number of motion buffers as reported by mb.buffers_available is typically nonzero - in the vicinity of 6 or 7, however the buffer referenced by mb.w is pointing to a buffer in the RUNNING (4) state.  This comment will be followed by some output from gdb that indicates the state of the ring buffer.

ryansturmer, 2015-02-03:
This at least gives the state of the buffer at the time of the alarm:

```
Breakpoint 1, mp_queue_command (cm_exec=0x8da61 <_exec_spindle_control(float*, float*)>, value=0x20087df0, flag=0x20087df0) at planner.cpp:211
211			cm_hard_alarm(STAT_BUFFER_FULL_FATAL);
(gdb) print mb.w
$1 = (mpBuf_t *) 0x200720b0 <mb+3296>
(gdb) print mb.r
$2 = (mpBuf_t *) 0x20072578 <mb+4520>
(gdb) set $i = 0
(gdb) print mb.q
$3 = (mpBuf_t *) 0x200720b0 <mb+3296>
(gdb) print mb.bf[$i++].buffer_state
$4 = 2 '\002'
(gdb) 
$5 = 2 '\002'
(gdb) 
$6 = 2 '\002'
(gdb) 
$7 = 2 '\002'
(gdb) 
$8 = 2 '\002'
(gdb) 
$9 = 2 '\002'
(gdb) 
$10 = 2 '\002'
(gdb) 
$11 = 2 '\002'
(gdb) 
$12 = 2 '\002'
(gdb) 
$13 = 2 '\002'
(gdb) 
$14 = 2 '\002'
(gdb) 
$15 = 2 '\002'
(gdb) 
$16 = 1 '\001'
(gdb) 
$17 = 1 '\001'
(gdb) 
$18 = 1 '\001'
(gdb) 
$19 = 1 '\001'
(gdb) 
$20 = 4 '\004'
(gdb) 
$21 = 0 '\000'
(gdb) 
$22 = 0 '\000'
(gdb) 
$23 = 0 '\000'
(gdb) 
$24 = 0 '\000'
(gdb) 
$25 = 0 '\000'
(gdb) 
$26 = 3 '\003'
(gdb) 
$27 = 2 '\002'
(gdb) 
$28 = 2 '\002'
(gdb) 
$29 = 2 '\002'
(gdb) 
$30 = 2 '\002'
(gdb) 
$31 = 2 '\002'
(gdb) print mb.w.buffer_state
$32 = 4 '\004'
(gdb) print mb.r.buffer_state
$33 = 3 '\003'
(gdb) print mb.q.buffer_state
$34 = 4 '\004'
(gdb) 
```

ryansturmer, 2015-02-24:
Further developments.  I have re-run essentially the same file, with different results (I am running an updated build with fixes.)

First of all, I am running the following build:
https://github.com/ShopBotTools/FabMo-G2-Core/commit/1c9a59274f95e0d5632c7ab253999243357b9f92

The geometry of the file is the same as above, and the result is thus:

[image]

In the file above, the pocket (green) is done first, and more or less looks correct.  Then the cutouts for the three holes (blue) - also looks correct.  Finally, the profile of the outside edge, which is a cutout, and has four tabs, which you can't really see in the diagram.  This starts out pink, in the correct position but then stopped at the red X, moving at a very slow pace along the trajectory of the cyan line, until it picked back up again in speed and in the correct toolpath geometry, although it had lost position at this point.  The trajectory of the errant segment appears to be that of the segment it was cutting at the moment of the failure, and it appears to be both in X and Y (although this is difficult to discern.  You can see that the cut is very slow, as the material surrounding the profile is burned.  

Once the cut resumed the geometry was more or less correct, but with an offset as shown.  **When the cycle completed, the reported positions appeared to be correct** - that is, on a second run of the file, the toolpaths started out in, at least approximately, the correct position.  No offset was seen in the initial passes of the file when run a second time.  The second run of the file appeared to lose position as well, but not in the same manner as the first.

On the second run of the file, (which I did as an air-cut above the table) There appeared to be some queue starvation, especially in the cutting of the outline, which appeared to be very jittery.  I am looking into this, but I'm not sure why this would cause a loss of position, especially with symptoms such as seen with the cyan line above.

File analysis doesn't show anything weird, except perhaps for some exceptionally short moves. There are a number of moves that are less than 1mil, which are detailed below, with the format **length, line number**

I unfortunately do not have logs for all traffic, but my logs do not show any errors returned from g2, and they don't indicate a starvation condition. (System streams more lines when the number of lines left unprocessed drops below 250, so there's plenty of margin for communication delay) 

```
0.0005000000000007888,2910
0.0006708203932503595,3670
0.0005385164807141745,3740
0.000632455532034168,3742
0.0005830951894839327,3818
0.0005830951894851513,3835
0.0005385164807140096,4106
0.000300000000000189,4110
0.0006324555320340275,4183
0.0005385164807140096,4185
0.0005385164807140096,4271
0.0005830951894843135,4288
0.0004123105625620976,4290
0.00044721359550050445,4896
0.00039999999999995595,5245
0.0004123105625616668,5508
0.000300000000000189,5872
0.0002999999999997449,6134
0.0005000000000007888,9044
0.0006708203932503595,9804
0.0005385164807141745,9874
0.000632455532034168,9876
0.0005830951894839327,9952
0.0005830951894851513,9969
0.0005385164807140096,10240
0.000300000000000189,10244
0.0006324555320340275,10317
0.0005385164807140096,10319
0.0005385164807140096,10405
0.0005830951894843135,10422
0.0004123105625620976,10424
0.00044721359550050445,11030
0.00039999999999995595,11379
0.0004123105625616668,11642
0.000300000000000189,12006
0.0002999999999997449,12268
0.00010000000000065512,17971
0.00028284271247458785,19049
0.0007211102550930879,19050
0.000360555127547283,20237
0.0005656854249487047,20238
0.0006403124237437345,20239
0.0005656854249498038,20240
0.0006403124237423474,20241
0.0006403124237437345,20242
0.0006403124237423474,20243
0.0007810249675910627,20244
0.0006403124237437345,20245
0.0007810249675909206,20246
0.0007211102550930879,20247
[... 73 lines cut ...]
0.0007211102550930879,27445
0.0007810249675896981,27446
0.0007211102550930879,27447
0.0007810249675909206,27448
0.0007810249675909206,27449
0.0008602325267044776,27450
0.0007810249675895558,27451
0.0008602325267043487,27452
0.0008602325267043487,27453
0.0008602325267044776,27454
0.0008602325267043487,27455
0.0008602325267043487,27456
0.0008602325267043487,27457
0.0008602325267043487,27459
0.0009433981132055564,27460
0.0009999999999998899,27461
0.0009433981132056741,27462
0.0009999999999998899,27463
0.0009433981132055564,27464
0.0009999999999998899,27465
0.0009999999999998899,27467
0.0009999999999998899,27468
0.000854400374531737,27722
0.000700000000000145,27938
```

giseburt, 2015-03-31:
@ryansturmer Did you determine that the cause of this was static from your dust collection? Or is there still an issue here?

giseburt, 2015-04-01:
@ryansturmer I marked this as invalid and am closing.

To everyone else out there: make sure you have protection against static build up! In particular, ground the dust collection hose, the limit switches, and the toolhead (where possible and safe).

The results of static discharge can be random and seemingly almost-sane behavior. In this case, it was causing it to seemingly randomly skip executing a few instructions, but then resume where it should.

closed 2015-04-01 by giseburt