synthetos/g2 #47: Edge won't compile on Linux

cmcgrath5035, 2015-02-04:
Continuation from bottom of #44 
I am following https://github.com/synthetos/g2/wiki/Compiling-G2-on-Linux-and-OS-X-%28using-the-command-line%29

From cli, I went to my /synthetos directory, ran  -> [I don't know why I am so big :) ]
--------------------
git clone https://github.com/synthetos/g2 new_g2
Cloning into 'new_g2'...
remote: Counting objects: 16398, done.
remote: Total 16398 (delta 0), reused 0 (delta 0)
Receiving objects: 100% (16398/16398), 94.70 MiB | 13.06 MiB/s, done.
Resolving deltas: 100% (9971/9971), done.
Checking connectivity... done.
----------- looks OK so far, so ran
/synthetos/new_g2> git checkout edge
Branch edge set up to track remote branch edge from origin.
Switched to a new branch 'edge'
----------- still looks OK, so ran
/synthetos/new_g2/TinyG2> make PLATFORM=gShield SETTINGS_FILE=settings_default.h
Compiling c CMSIS/Device/ATMEL/sam3xa/source/system_sam3xa.c
    -> build/gShield/CMSIS/Device/ATMEL/sam3xa/source/system_sam3xa.o 
bash: arm-none-eabi-gcc: command not found
Makefile:455: recipe for target 'build/gShield/CMSIS/Device/ATMEL/sam3xa/source/system_sam3xa.o' failed
make: *** [build/gShield/CMSIS/Device/ATMEL/sam3xa/source/system_sam3xa.o] Error 127
----------------- Error, obviously.
Interestingly, the /synthetos/new_g2/Tools directory did not download anything:
.. /synthetos/new_g2> ll Tools
total 4
-rw-r--r-- 1 carl users 1759 Feb  4 15:41 Makefile

I can't call this a compiler error, the compiler did not get loaded into the Tools directory.....?

My prior clone looks like this
.../synthetos/g2_edge_old> ll Tools
total 60044
lrwxrwxrwx 1 carl users       28 Dec 22 23:08 gcc-arm-none-eabi -> gcc-arm-none-eabi-4_8-2014q1
drwxr-xr-x 6 carl users     4096 Dec 22 23:08 gcc-arm-none-eabi-4_8-2014q1
-rw-r--r-- 1 carl users 61469091 Mar 27  2014 gcc-arm-none-eabi-4_8-2014q1-20140314-linux.tar.bz2
-rw-r--r-- 1 carl users     1759 Dec 22 20:10 Makefile

History:
1. I cloned and checked out Edge on Dec 28, procedure worked as written, Edge Make built 72.01
2. I checked out Master on Jan 26. Make  again caused the compile tarball to download and populate the /Tools directory and unpack itself there. Edge had been moved to Master, Make in Master built 72.01 again.

I'm open to suggestions - no rush, not top priority my end

ryansturmer, 2015-02-05:
@cmcgrath5035 Do I understand correctly that you don't have the tools to compile, and that is the issue?

giseburt, 2015-02-05:
Hi @cmcgrath5035 ,

So, for the sake of testing, can you cd into the Tools directory and call `make` to see if it does the right thing? I suspect it isn't a failure of installing the tools but is a failure of detecting that it needs to.

Thank you!
  -Rob

cmcgrath5035, 2015-02-05:
Not exactly. All the base tools are loaded.
The Makefile is supposed to download the compiler if it does not find it.
Rob wants a specific version of the compiler.

That part failed this time (worked prior two runs with new cloned 
workspaces).

cmcgrath5035, 2015-02-05:
Running make from the tools directory properly downloaded the compiler.

Running make from new_G2/TinyG2 compiled and built.

So something must be corrupted in my old Edge, but in a way that git 
does not detect.

Fire drill over, Thanks

ryansturmer, 2015-02-05:
According to rob, there's possibly a gcc issue on linux that's causing the build to pass (but not produce working code)  I can build mine, but have yet to test ,as I'm chasing down other issues.  I'll let you know if I find that the code it produces actually works.

cmcgrath5035, 2015-02-05:
OK, I think I have an idea what might be going on.
1. I realized today that 'make clean' only cleans up the build environment for Platform gshield. 
Is this by intent? 
I am running make for both Platform gshield and Platform G2v9i. For now, if I am doing new build with Platform G2v9i, I run make clean and then manually delete the directories TinyG2/bin/G2v9i and TinyG2/build/G2v9i. This causes make PLATFORM=G2v9i to rebuild everything
2. Here is a results table for 8 different builds (all are after make clean plus manual clean as in 1. above)
This applies to Master and Edge as of Feb 4, 2015. All builds on Linux
tinyG2 Master
make PLATFORM=gShield SETTINGS_FILE=settings_default.h                         Builds OK
make PLATFORM=gShield SETTINGS_FILE=settings_shapeoko2_cjm_v1.0.h  Builds OK
make PLATFORM=G2v9i SETTINGS_FILE=settings_default.h                            Builds OK
make PLATFORM=G2v9i SETTINGS_FILE=settings__shapeoko2_cjm_v1.0.h   Builds OK
tinyG2 Edge
make PLATFORM=gShield SETTINGS_FILE=settings_default.h                         Builds OK
make PLATFORM=gShield SETTINGS_FILE=settings_shapeoko2_cjm_v1.0.h  Build Fails
make PLATFORM=G2v9i SETTINGS_FILE=settings_default.h                            Builds OK
make PLATFORM=G2v9i SETTINGS_FILE=settings__shapeoko2_cjm_v1.0.h   Build Fails

3. So Edge builds do not like my SETTINGS_FILE=settings__shapeoko2_cjm_v1.0.h
I created this file from (what is now Master) settings_default.h.
I made two changes/additions to settings_default.h. My goal is to have the compiled-in settings for G2 match my machine, to minimize the disruption of a power-up (overcome lack of Persistence)
A. I entered motor and axis parameters that are correct for my ShapeoOko2 (not a fully stock Oko2)
B. I added a Parameter field for PWM parameter overrides. I do not like the defaults being used in all builds, see https://github.com/synthetos/TinyG/issues/121. Those parameters spin up my spindle on reset - not good.

Summary- My build issue with EDGE seems related to my custom SETTINGS_FILE=settings__shapeoko2_cjm_v1.0.h. here is a link to it: https://dl.dropboxusercontent.com/u/50261731/settings_shapeoko2_cjm_v1.0.h. 
The make clean issues with Makefile created a lot of confusion but have a workable manual workaround.

Next Steps - I will next do a detailed comparison of Master settings_default.h and EDGE settings_default.h to see if there is an obvious change that I need to roll in to my custom settings.

I can't get to it right away, I wanted to get this out so you guys don't spend time on a problem you might not be able to reproduce.

And since Rob mentioned it - I have not tried to load or run any of these, so there is nothing here that reads on the potential issue of compiler building file.hex that don't run.

ryansturmer, 2015-02-05:
Can you post the compiler error messages for the builds in your build table that fail?

cmcgrath5035, 2015-02-05:
This file is for make PLATFORM=gShield SETTINGS_FILE=settings_shapeoko2_cjm_v1.0.h Build Fails
https://dl.dropboxusercontent.com/u/50261731/G2_Build_log.txt

Similar look for the G2v9i build - I can redo it if you need (lost the screen)

cmcgrath5035, 2015-02-05:
I just ran diff on settings_shapeoko2.h in the current Edge directory and the Edge directory from end of December. They are identical.
So, if the compile failure is due to my settings file,  it will be in the area where I added in #defines for P1xx parameters.
file /settings/settings_default.h does not have #define for any P1xx parameters, so defaults are from /TinyG2/settings.h.
I copied the applicable parameter #defines from settings.h to settings_shapeoko2.h.
I then modified values in settings_shapeoko2.h to match my machine and saved as a new file, settings_shapeoko2_cjm_v1.0.h. 
The added lines are 206-225 in settings_shapeoko2_cjm_v1.0.h

For my builds of G2 72.01, the parameters looked correct when I ran both the gShield (Due) and tinyGv9 builds with CoolTerm.

giseburt, 2015-02-05:
It’s saying that X_SWITCH_TYPE_MIN, Y_SWITCH_TYPE_MIN, Z_SWITCH_TYPE_MIN, A_SWITCH_TYPE_MIN, B_SWITCH_TYPE_MIN, and C_SWITCH_TYPE_MIN (along with the corresponding *_MIN) are not defined.

The X-SWITCH_TYPE_* are defined in the default settings here: https://github.com/synthetos/g2/blob/edge/TinyG2/settings/settings_default.h#L115-L116 <https://github.com/synthetos/g2/blob/edge/TinyG2/settings/settings_default.h#L115-L116>

Indeed, those are missing from the file you linked to. You should be able to just copy those into your file and get it to compile.

For the record, I don’t think this is directly related to this issue, since the problem really isn’t that it won’t compile, it’s that it won’t run.

-Rob

cmcgrath5035, 2015-02-05:
I'll give that a try.when I have time.
Not urgent to me.

Thanks

cmcgrath5035, 2015-02-07:
OK, with your hint I found the error in my ways.
I mentally saw that changes to the /setting files on github were 4 months old but did not check that the settings  that did compile were 6 months old. I sort of wish github could build the local clone with your modify dates rather than the date of the clone or pull.

Anyway; both of the make Failures above now  Build.  That's the good news.

And, as you expected, at least the G2v9i version does not run.

Here is the download dialog:
stty -F /dev/ttyACM0 1200
carl@PVE-LinuxSRV5:~/Playground/CNC/tinyG_G2/src/github.com/synthetos/g2_edge_new/TinyG2/bin/G2v9i> bossac  --port=ttyACM0  -U true -e -w -v -i -b -R G2v9i.bin
Erase flash
Write 123348 bytes to flash
[==============================] 100% (482/482 pages)
Verify 123348 bytes of flash
[==============================] 100% (482/482 pages)
Verify successful
Set boot flash true
Device       : ATSAM3X8
Chip ID      : 284e0a60
Version      : v1.1 Dec 15 2010 19:25:04
Address      : 524288
Pages        : 2048
Page Size    : 256 bytes
Total Size   : 512KB
Planes       : 2
Lock Regions : 32
Locked       : none
Security     : false
Boot Flash   : true
Reset not supported for this CPU

Looks OK to me.

BUT:BAD
The tinyGv9 is now stuck with all lights on (as if still in 1200 Baud mode) even after reset button and Power Cycle.

BUT:WORSE
All lights are on, but nobody is home. 
None of (Reset button, power cycle and USB remove/reinsert ) bring up a /dev/ttyACM0 or ACM1 port, so no access to bootloader or anything else.
Appears tinyGv9 is bricked.
I do have an ATMEL Ice Basic, have never opened the box. BOSSAC worked well for me.

I'll try reloading Master 72.02 using ICE when I have some spare minutes. 
That will have a learning curve.

Look again at the last line of the BOSSAC session above ; "Reset not supported for this CPU"
Might that be the issue? I don't recall (and did not keep) the messages from my last successful load.

dhylands, 2015-02-28:
I'm also having a similar issue. I'm using Ubuntu 14.04, and have the linaro toolchain already installed for work I do on MicroPython.

To summarize what I've done:
git clone the TinyG2 repo
git checkout edge
cd into the g2/TinyG2 directory
make
Run my flash.sh script:
2036 >./flash.sh 
+ stty -F /dev/ttyACM0 1200
+ bossac --port=ttyACM0 -U true -e -w -v -i -b -R ./bin/gShield/gShield.bin
Erase flash
Write 125924 bytes to flash
[==============================] 100% (492/492 pages)
Verify 125924 bytes of flash
[==============================] 100% (492/492 pages)
Verify successful
Set boot flash true
Device       : ATSAM3X8
Chip ID      : 285e0a60
Version      : v1.1 Dec 15 2010 19:25:04
Address      : 524288
Pages        : 2048
Page Size    : 256 bytes
Total Size   : 512KB
Planes       : 2
Lock Regions : 32
Locked       : none
Security     : false
Boot Flash   : true
CPU reset.

and the flashing seems to be successful (using the programming port), but the orange LED is on solid.

I only have the Arduino Due with no shield or anything else plugged into it. I'm just connecting to the programming port via USB (i.e. no extra power supply).

When I build, I see one compiler warning:
./planner.cpp:690:9: warning: #pragma GCC target is not supported for this machine [-Wpragmas]
 #pragma GCC pop_options
         ^

and 3 linker warnings:
build/gShield/CMSIS/Device/ATMEL/sam3xa/source/gcc/startup_sam3xa.o:(.vectors+0x64): warning: undefined reference to `SMC_Handler'
build/gShield/CMSIS/Device/ATMEL/sam3xa/source/gcc/startup_sam3xa.o:(.vectors+0x90): warning: undefined reference to `USART3_Handler'
build/gShield/CMSIS/Device/ATMEL/sam3xa/source/gcc/startup_sam3xa.o:(.vectors+0xe8): warning: undefined reference to `EMAC_Handler'

I've tried different versions of the toolchain, but everything seems to behave the same. I've never seen the orange LED blink with the TinyG2 firmware.

Any clues as to what I need to do to get past this?

Also is there a mailing list, irc channel, or forum for the TinyG2? Should we use the TinyG forum? Or just post issues here?

cmcgrath5035, 2015-02-28:
TinyG forum is OK too, but this problem probably easier here(better interface to work with)

My Original compile issue was that I had not migrated my custom settings file to the new needs of the new build. You are beyond that. 

Look in your github.com/synthetos/g2/Tools directory and see if your compiler is this: .....github.com/synthetos/g2_edge_new/Tools/gcc-arm-none-eabi-4_8-2014q1-20140314-linux.tar.bz2. If so, you are likely in the same boat I was - you eDUE may be bricked  due to a compiler issue.
See https://github.com/synthetos/g2/issues/53 for my adventure in using AtmelICE.

I had no choice, my tinyGv9 was just hard bricked, but finally ice recovered,
Here is an old build for DUE (71.02, I think) that might load to a dDUE, because of the Programming Port. https://dl.dropboxusercontent.com/u/50261731/gShield.bin

dhylands, 2015-02-28:
Interesting. My Due doesn't seem to be totally bricked. I can flash it using the Arduino IDE and get LED flashing and USB/serial comms working.

My Tools directory is essentially empty because I already had the toolchain in my PATH. I'll try removing it and see what gets downloaded.

dhylands, 2015-02-28:
I tried gcc-arm-none-eabi-4_7-2014q2 and it behaves the same. Does anybody a version of toolchain which works?

dhylands, 2015-02-28:
Using your bin file, I now see /dev/ttyACM0 and /dev/ttyACM1 when I plug into the native port. And the RX LED flashes slowly.

And if I run: picocom --baud=115200 --flow h /dev/ttyACM0 I get some chattter.

Later this evening, I'll try going back in time and seeing if the problem is toolchain related or code related.

cmcgrath5035, 2015-02-28:
The slow flashing LED sounds correct.
This was compiled last December 20ish.

It seems I had the same version compiler back then, so it may be a 
feature of the newest edge that hits the defect.

In a v9, ALL the software is in the Synthetos build, including the USB 
interface.

The Due Architecture is what is saving you - programming port has it's 
own USB  interface.

Read up in the G2 Wiki on DualUSB, on Due they can be dual physical and 
logical connections, on V9 (only one USB), they are dual logical but 
single physical.

dhylands, 2015-03-01:
ok - progress.

hash https://github.com/synthetos/g2/commit/1721ffac9f581fe27dd4bb460236586f20793b32 is the last one that works (definition of works is the RX LED flashes slowly)

The very next commit: https://github.com/synthetos/g2/commit/3af2b03f3485842e09d7712c1216a040e1f51696 fails.

Now to see if I can figure out why it fails.

dhylands, 2015-03-01:
I created PR #55 which makes things work for me under linux. Hopefully it doesn't break other platforms. We'll see what travis says.

cmcgrath5035, 2015-03-01:
So with this change G2 Edge now builds and runs on Linux - what fwbuild do you get?
And you consider the build "working" OK , you post in #54 implies your tinyG is functional even though the Linux driver is complaining about 'maxpacket 256" ?

dhylands, 2015-03-01:
Is fwbuild the number like 71.02 If it is, I see a 78.01 on my chilipeppr UI (on the right under Feedhold).
https://www.dropbox.com/s/aumi82fi3u0g3mx/Screenshot%20from%202015-03-01%2009%3A30%3A00.png?dl=0

This is my first time using TinyG and ChiliPeppr. I can issue G commands and I see them being sent via serial-json to the board and then I see the UI update to reflect the new position. I don't have any electronics hooked up to my Due board yet - oh - let me hook up my logic analyzer to the step/dir pins.

My understanding of USB stuff is that maxpacket should be limited at 64. I'll take a look at the driver and see what it says.

dhylands, 2015-03-01:
Doh - bulk should be 512. Here's where the driver error message originates from:
http://lxr.free-electrons.com/source/drivers/usb/core/config.c#L258

I found a reference to back up that assertion: https://books.google.ca/books?id=gzUOUeEbozQC&pg=PA53&lpg=PA53&dq=bulk+endpoint+max+packet+size&source=bl&ots=HYzCU-FsWJ&sig=slGeo_gV3B3UWglrrl5pn5ZEwrg&hl=en&sa=X&ei=flPzVPjGOYTVoASWp4D4AQ&ved=0CC4Q6AEwBDgK#v=onepage&q=bulk%20endpoint%20max%20packet%20size&f=false (this is from Jan Axelson's USB Mass Storage book)

cmcgrath5035, 2015-03-01:
Yes, 78.01 is the value of tinyG parameter $fb.
So Edge is moving fast, Rob's a busy boy.

So the USB Error flagged in #54 is something tinyG2 needs to fix?
But it's current setting is working for you, it would appear.

dhylands, 2015-03-01:
@cmcgrath5035 I did a quick hack to bump the bulk descriptors up to 512 bytes (and now I don't see the maxpacket warnings)
```
diff --git a/TinyG2/motate/utility/MotateUSBHelpers.h b/TinyG2/motate/utility/MotateUSBHelpers.h
index 2ffe9b2..56c7fe9 100644
--- a/TinyG2/motate/utility/MotateUSBHelpers.h
+++ b/TinyG2/motate/utility/MotateUSBHelpers.h
@@ -613,8 +613,8 @@ namespace Motate {
         uint16_t suggestedSize = 512;
         if (otherSpeed) {
             suggestedSize = 64;
-        } else if (limitedSize) {
-            suggestedSize = 256;
+//        } else if (limitedSize) {
+//            suggestedSize = 256;
         }
 
                if (USBDeviceSpeed == kUSBDeviceHighSpeed) {
```

dhylands, 2015-03-01:
I think that tthe maxpackets thing may depend on the host peripheral. I saw conflicting reports of bulk packet size. Some places say upto 512 bytes, some places exactly 512 bytes (for highspeed).

I'm not familiar enough with the guts USB to know how significant it is. It seems to working on my zaReason ultralap. I have several other machines around, I should try it on all of them and see if it works on those.

giseburt, 2015-03-02:
With PF #55 merged in, can someone confirm that the latest edge compiles on Linux and runs properly? (Notwithstanding possible USB errors tracked in #54.) Once I hear a "confirmed" we can close this. :)

cmcgrath5035, 2015-03-02:
See bottom of #47, Dave did several linux compiles and some 
experimentation with USB.settings
He seems to be running OK.

dhylands, 2015-03-03:
I just compiled https://github.com/synthetos/g2/commit/9457bb706773578951170dcdb3fc9a6594a6e521 and was able to flash it, and my orange RX LED is blinking, and ChiliPeppr can interact with it.

giseburt, 2015-03-03:
Very good news, @dhylands . Thank you for helping us find this one!

-Rob

closed 2015-03-03 by giseburt