synthetos/g2 #48: Edge chokes on dwell (G4)

ryansturmer, 2015-02-05:
In the same way described in #44, the G4 command appears to cause G2 to crash irrevocably.  A simple file such as what follows is all it takes:

```
G1X1F120.0
G4P3.0
G1X0F120.0
```

Sending this down all in one packet (on the data channel) actually causes the program to hang before the tool is even set in motion.  On an older build g2@3b00ffad  the hang occurs in the trapezoid planner on the check for a non-NULL motion buffer.  On the tip of edge, the program proceeds, but there is no motion, and subsequent commands are replied to with er:9 "File not open"

This feels related to #45

ryansturmer, 2015-02-06:
The "file not open" thing is a red herring.  This occurs on my build because after it locks up, I attempt to restart my tool, and one of the first things it does is a {gun:0} command, which provokes the persistence function to try to write a persistent setting, which is not taken because persistence isn't implemented.  It's strange that I only get that error after this crash, and not during a "normal" startup, though.

giseburt, 2015-03-31:
@ryansturmer I believe this is fixed, and has been for a while. Could you confirm?

Thank you!

closed 2016-02-23 by aldenhart