synthetos/g2 #50: Spindle Spped Control Alternative

cmcgrath5035, 2015-02-18:
If I could figure out how to label this an enhancement, I would

 From what I see in low cost PWM drivers for spindle motors, an analog voltage (normally POT) control is almost always supported, PWM signal sometime but not always.

An nice enhancement would be for the Spindle speed (now known as Spindle PWM) to be either PWM or a D/A converted analog output. Range would be up to the user to implement .

ryansturmer, 2015-02-20:
Since we're on the topic, some spindle controllers are actually a CAN or an RS-485 transceiver, so in some situations a spindle speed change might be rendered as a transaction over a UART.

pcats, 2015-03-05:
The UART spindle speed change is fairly easy to implement. I am using G2 running on a Due for a huge laser cutter, in which the laser is an intelligent peripheral with power output controlled by a serial command. A colleague and I managed to subvert the spindle speed command to be a laser power command instead, it took relatively little effort getting it running on UART0. Ideally I'd like to put it on one of the USARTs instead but I ran into some annoying issues with that, decided I had better things to do with my time at the moment, and left it. I will have to revisit it at some point.

The mods we made to the gcode parser were fairly simple, mainly finding some code to initialise the UART, splicing it in at the top, then adding a call in the parser for the S command to instead send a serial string. It's dirty but it works.

stg, 2015-03-30:
While having analog output is probably a good idea, I found running an analog voltage through all the cabling was too noisy to provide a clean control signal for the VFD and there was a ~5% voltage drop too... just seemed to unreliable in my particular environment.

What you can do if you need this feature right away is to do PWM all the way to the controller, clean it up with a schmitt triggered logic buffer followed by an RC low pass filter (good start may be 10k, 10µF = ~1.6Hz) and possibly an amplifier to bring it into the correct range (I needed 0-10V). Some logic families support higher voltages, so driving the buffer at say 10V might negate the need for an amp.

That said, I'd love to see both analog out and serial out as most of my controllers do RS485 - modbus or proprietary which would yield better precision and a higher degree of control.

jamesgao, 2015-04-15:
Also to tag along with this, is there any way to add a servo output as spindle speed control? I'm using a hobby RC motor which takes the special 50hz PWM-like output to control speed.

Engineer3DPrototyping, 2015-12-28:
I didn't see this post before i just made mine but another option that may be easier to implement is changing from fixed frequency variable duty cycle to fixed duty cycle variable frequency. Now only if I knew C

aldenhart, 2016-02-23:
Update on this thread - we are incorporating various spindle control methods in the GPIO overhaul that is currently in progress. If you have specific units you wish to control please highlight them. I cannot guarantee that we will support everything we receive, but the design should at least accommodate them.

Engineer3DPrototyping, 2016-02-25:
Maybe this is just my particular unit(Hitachi WJ200) but the the RS485 or modbus protocols would work but the one that I think would be easiest to implement because I have no starting point for the above protocols is what Hitachi calls the pulse train input. This is where the speed is controlled by a 50% duty cycle variable frequency pin(spindle PWM possibly modified to use a timer1 like library?)  A second pin for direction CW or CCW and a third Enable/diable pin. The outputs would go to a n channel mosfet to low side switch the 24V for the VFD logic or just about any other voltage required. This setup would be very similar to the current configuration of the spindle control pins of thiyg2 but I am not a computer scientist.  I ended up using a pwm to analog(0-10V) circuit for a few minutes but it overloaded the due's voltage regulator with everything else hooked up.    

 link to the hitachi manual for those who want more info.
 http://www.hitachi-america.us/supportingdocs/forbus/inverters/Support/WJ200_Instruction_NT325X.pdf

aldenhart, 2018-06-02:
Closing this issue. The current state of the spindle speed is that the spindle control is PWM-able, and the gQuintic board uses this to drive a 0-10v DAC output (in addition to offering raw PWM)

closed 2018-06-02 by aldenhart