synthetos/g2 #51: Arduino 1.6.0

anthonywebb, 2015-02-18:
I am trying to flash my due using arduino 1.6.0 and getting:

##########################################################################################
Arduino version is too old and does not include support for the due.  Get a newer version!
Download a newer ARM version here: http://arduino.cc/en/main/software
Exiting......
##########################################################################################

I know 1.6 does support the due, but it looks like bossac is not in the same spot, anyone know where it went in 1.6?

anthonywebb, 2015-02-18:
Actually, I installed the old 1.5.7 BETA and bossac is not there either.  Can someone point me to what version of the IDE I need to install in order to flash my due?

Thanks!

anthonywebb, 2015-02-18:
Sent you a pull request to fix this issue

cmcgrath5035, 2015-02-19:
I found and used a Bossa_1.3 .deb package and converted to .rpm for my Linux system.
I compiled using Linux tools, so had no need for AtmelStudio6.2, which I did download into a WinVM, but have not used it(successfully).
I am not fluent in Arduino, so am not sure what Arduino 1.6 is, but assume it is the Arduino IDE.
All the G2 Wiki items suggest AS6.2.
If all you need is to flash a binary, Bossa ver 1.3 is what you need, and it is not on github, or perhaps I hould say it was not on github in December.
HTH, obviously depends on your OS how to get it.
If perchance you are a Linux user, read #47

aldenhart, 2016-09-08:
Closing this in favor of the PR, which I am confirming has already been made

closed 2016-09-08 by aldenhart