synthetos/g2 #53: Atmel Studio 6.2 Issues - README

cmcgrath5035, 2015-02-21:
A recent disaster bricked my tinyGv9 (see #47 for gory details).
So I went to the shelf and dusted off my new, still in box, Atmel-ICE.
Prior to this , BOSSAC ver 1.3 worked so well I had not needed ICE.

Following the Wiki instructions, I set off to load and run Atmel Studio 6, the prescribed interface to Atmel-ICE. If you head to Atmel today, it will offer for download Atmel Studio 6.2 build 1548 Service Pack 2. As suggested, grab the version that includes .Net updates, just in case.

Just a note - I am a Linux user, run Windows only when absolutely necessary and then in Virtual Box VMs. So almost by definition, my Windows machines are not ready for heavy duty development work (.Net., etc, etc.) If your are an experienced Windows developer, you are likely accustomed to this abuse.

AS6.2 is Windows only.  I tried a Win 8.1 Virtual Box VM, then a Win7 Virtual Box VM, then a native boot into Win7. After install, none would connect to my Atmel-ICE.
Being new to AS6, I flounder a bit not knowing what to expect from the interface.

For now, assume there is a fatal bug in this version of the AS6.2 package.

If you just want to move forward, review this POST : http://www.avrfreaks.net/comment/1456731#comment-1456731 then proceed to download  http://distribute.atmel.no/tools/AS6/driver-atmel-bundle-7.0.666.exe. Just run the .exe, it will update/overwrite the batch of USB drivers distributed with AS6.2.
If you are lucky, as I was, you can now connect to Atmel-ICE.
Compared to this AS install fiasco, flashing tinyGv9 or DUE with ICE is trivial. Follow the Wiki.

I will update this as I triage where I am at. 
So far, I can only guarantee that this now works on native Win7.
I'll test the VMs tomorrow, or soon.

ril3y, 2015-02-21:
Yah let us know.

cmcgrath5035, 2015-02-24:
I have now installed http://distribute.atmel.no/tools/AS6/driver-atmel-bundle-7.0.666.exe on both my Wi7 and Win 8.1 Virtual Box VMs . In AS6.2, the Atmel-ICE now appears as a selectable tool under the Device Programming window.
I did not run a flashing process, but assume that it will work now that the driver mess seems resolved.
So, for now, Win7 native, Win7 and Win8.1 Virtual Box VMs seem operational with AS6.2 and Atmel-ICE

aldenhart, 2016-02-23:
AS6 works. Please see these pages:
https://github.com/synthetos/g2/wiki/Flashing-G2-with-Atmel-Studio6-and-Atmel-ICE
https://github.com/synthetos/g2/wiki/Compiling-G2-on-Windows-(Atmel-Studio-6.2)

closed 2016-02-23 by aldenhart