synthetos/g2 #54: G2/Edge won't talk to linux host, usb errors suspected

buserror, 2015-02-28:
I'm trying G2 on an arduino Due, and the ports ACM0 and ACM1 get created, however, if I picocom -b 115200 /dev/ttyACM0 ( or 1 ) there's nothing, no output, no echo. Chillipeppr also fails to do anything with these ports.
One thing that looks interesting is that the kernel (3.16) complains about:
usb 1-3: config 1 interface 1 altsetting 0 bulk endpoint 0x2 has invalid maxpacket 256
usb 1-3: config 1 interface 1 altsetting 0 bulk endpoint 0x83 has invalid maxpacket 256
usb 1-3: config 1 interface 3 altsetting 0 bulk endpoint 0x5 has invalid maxpacket 256
usb 1-3: config 1 interface 3 altsetting 0 bulk endpoint 0x86 has invalid maxpacket 256

AFAIK, maxpackets are fixed, and should be 512 on bulk packets, can this explain why I can't talk to my board?

dhylands, 2015-03-01:
I see the same output in dmesg with my "working" board.

giseburt, 2015-03-02:
This is not a simple issue. Please see the comment (and referred to links) int he code:

https://github.com/synthetos/g2/blob/edge/TinyG2/motate/utility/MotateUSBHelpers.h#L580-L606

I still need to look into this more, but there a re a few notes:

In the output above it says `interface 1 altsetting 0 bulk endpoint` -- `altsetting` is the key here. This means it's not the primary config setting (which is likely the one being used) but it's instead it's the secondary value.

@buserror Could you send a more complete output? I'm looking for a few things:

  * Did it use the primary or alternate settings eventually?
  * When it asked for the altsettings what speed did it indicate? If it asked for "full speed" settings, then 64 is the largest value we can return.

Another note: We have to be careful about our memory usage. According to the spec it can be "up to 512", and according to the RAM we have in the SAM3X8E[CE] we have 4k. We should have enough in the USBOTG peripheral for four 512 "bulk" endpoints (with two banks each) and two 64-byte "control" endpoints. However, whenever I test it, it crashes the processor. This is why I have the "limitedSize" parameter on `getEndpointSize` so that when you're doing dual-endpoint CDC it'll use 256 as the max size.

dhylands, 2015-03-03:
I was helping @buserror and I think he was having toolchain issues. I was able to build and run both with 256 maxpackets and 512 byte maxpackets.

The 256 size is just a warning from the kernel and I don't think that its a problem (at least not so far). @buserror and I were building from the same git hash, and we think he may have been having toolchain issues. We're both trying to use the due/gShield combo and both run linux.

giseburt, 2015-03-31:
Any update on this? @buserror, are you still having issues with USB?

closed 2016-02-23 by aldenhart