synthetos/g2 #56: G2 gShield/DUE pin mapping seems to be incorrect

dhylands, 2015-03-02:
I checked out the latest g2 edge branch did a build (which defaults to gShield) and flashed my Arduino DUE (I'm using linux so I also needed PR #55).

I hooked my logic analyzer to pins D2 thru D8 on the DUE (according to the gShield schematics this should be the XYZ step/dir pins + enable).

I'm not seeing any steps on D2 at all.
I'm seeing steps on D3 when I ask to move the Z-axis (D3 is the Y axis)
I'm seeing steps on D4 when I ask to move the Y axis (D4 is the Z axis)

The Y&Z direction seem to be swapped as well (and I see no changes on x-dir)

Anybody have any idea what's going on?

giseburt, 2015-03-02:
Alas, this is a result of the default settings. See https://github.com/synthetos/g2/blob/master/TinyG2/settings/settings_default.h

The biggest problem is that the motors are mapped wrong for the gShield. X maps to motor 4, which isn't even available on the shield. It maps "X" (as on the silk screen) -> A, "Y" to Z, "Z" -> Y, "motor 4" (which isn't on the gShield) -> X, and "motor 5" (also not on the shield) -> B.

These really should be changed. These setting can all be changed in runtime, but the Due doesn't have an EEPROM so it cannot save those changes between resets.

You can also choose a different settings file at compile time with adding `SETTINGS_FILE=settings.xyz.h` on the make command line. I would suggest using `settings_shapeoko375.h` as a more reasonable default. You can copy that file and tweak the settings as well to make your own custom settings. Please note that the shapeoko can more rather fast, and the set speeds and jerk values may be too high for many machines.

-Rob

PS: I marked it as a "Question" due to the nature of the request. However, I also marked it a "Bug" since, IMO, it should not be this way, and can be marked as fixed once it's adjusted.

giseburt, 2015-03-03:
This is fixed in the edge branch, commit e82b3e76dabf4b2d229282ed54fb00b99cb1d390.

-Rob

closed 2015-03-03 by giseburt