synthetos/g2 #57: Psuedo EEPROM support?

dhylands, 2015-03-03:
Does it make sense to use a flash page to save what normally gets stored iin EEPROM?

Rather than rewriting flash for every value change, we could introduce some commands to do that.

Marlin has the following G-codes:
    5.1 M500: store parameters in EEPROM
    5.2 M501: read parameters from EEPROM
    5.3 M502: revert to the default "factory settings."
    5.4 M503: Print settings

If we replace the words EEPROM with FLASH, then we could at least have persistent settings. I wouldn't use M-codes, I' just talking about the concept.

Anyways, I figured I'd throw it out as an idea. If there is traction, I'd be happy to work on it (I'm a total noob to the TinyG code base, but an experienced embedded programmer otherwise).

cmcgrath5035, 2015-03-03:
Dave
TinyGv8 runs tinyG fw; you are thinking tinyGv9, whose core uC is same as DUE (different pinout and package)
The tinyGv9 boards have a microSD card carrier. No word yet on progress enabling the microSD for Persistence.

dhylands, 2015-03-04:
Yes - I should have used TinyG2 everywhere I used TinyG.

Are schematics (or preliminary schematics) for the TinyGv9 available someplace? Adding sdcard support to the DUE would also be interesting. I'd rather move in the direction of the TinyGv9 board rather than going off and inventing something new.

It looks like there is lots of flash (for code) available on the DUE. I've worked with FatFS before on the MicroPython project.

Presumably having the ability to load new firmware from the sdcard would also be useful.

giseburt, 2015-03-04:
Hi @dhylands,

We would very much like to use some flash as the persistence so that we have feature parity between G2 and TinyG.

Using an SD card is nice as well, and would be great for a few things, but not all hardware will have it and so we'd like to not *have* to have the SD card hardware to be able to store settings.

The way TinyG does it (with an EEPROM built into the processor) is that when any parameter is changed (using $dollar=1 or {json:1}) then we save the EEPROM changes. For flash, we'd probably not want to save for every chane, but instead have a special command that triggers the "save to EEPROM." This also allows you to "try" settings before committing them.

We don't have a lot of time to work on this, internally, but will help in any way possible if anyone wants to help tackle it!

Thank you!
  -Rob

dhylands, 2015-03-04:
Sure - I'd be interested in looking  into that.

cmcgrath5035, 2015-03-04:
Dave
If you are a Chilipeppr user (and of course not all G2 users are or will be), you might want to look at some of the coding Kevin Hauser is doing in the configuretinyG widget. He has contemplated making stealth backups of parameter changes to the CP cloud resource. Of course the same difficult task remains - how to decide when to commit temporary changes to permanent with clarity as to what is being saved(Kevin is more focused on G1 at the moment, but aware of the G2 flash issues
I assume you are aware that several DUE compatible shields have SD card carriers, attached to the I2C bus I believe. I tried one, just for play, and it was slow but worked. Rob, is the v9 micro carrier connected to I2C in a compatible sort of fashion? v9 schematics are not posted yet.

dhylands, 2015-03-04:
I've never heard of a i2c sdcard (not to say such a thing doesn't exist). Most sdcards support SPI mode, and typically hook up to SPI.

I'm going to start out using chilipeppr. Adding the eeprom emulation in flash should co-exist fine with anything in chilipeppr.

cmcgrath5035, 2015-03-05:
Yup, my error, was SPI.

johnlauer, 2015-03-17:
The way I am solving lack of eeprom right now so I can actually use the v9 is the little cog wheel next to the serial port in chilipeppr. You can program chilipeppr to send commands to the serial port on connect. So I just program all the overrides I want to setup motors, etc. The eeprom essentially is your computer's hard drive.

cmcgrath5035, 2015-03-18:
Yet another alternative is to do a restore from archive using my beta grade restore-from-archive widget.
Working well most of the time with tinyG, I have not tried G2 lately, but it did work with G2 Master about 3weeks ago.
Here is a quick how-to: https://dl.dropboxusercontent.com/u/50261731/archiverestoretinyG%20How%20to.doc

johnlauer, 2015-03-18:
Carl, do you want a way to do the restore automatically each time the user connects to the device? That would seem to achieve what I'm doing with the little cog wheel. You could just listen to the "/com-chilipeppr-widget-serialport/onportopen" pubsub signal and then basically restore the archive data. That would be cooler than my technique since your stuff is slicker. The only thing we'd have to change is that we'd have to load your and Kevin's widget at load time rather than lazy loading it when the user clicks the button. Want me to do that?

cmcgrath5035, 2015-03-18:
To be honest, I don't think that would be a good idea.

Perhaps it would be for G2 users, but for the general population, and 
particularly newbys, I prefer to have folks intentionally go thru a few 
steps to get in the proper mindset.

So much is happening automatically, on startup or in the background, 
folks are not developing any basic understanding of what is going on or 
how stuff works, thus don't have a clue when something goes wrong or 
does not work s it did 'the last time.

I can certainly see instances where this might be helpful, G2 and folks 
who have several machines they maintain.
But folks who have several machines are skilled enough that two or three 
clicks to get configured takes little time anyway.
It's a mindset sort of thing, if users request such a feature then 
definitely would move on it.

In my specific case, where I switch among  setting files for testing and 
version swaps, I am not sure what logic you might use when I started up.

If we implement something like this, there needs to be clear indication 
of what is happening on startup

stg, 2015-03-24:
If I may insert a thought... While flash endurance is typically pretty low (can't seem to find an exact number for the SAM3 series) I've avoided issues with this by utilizing BOD and NMI to trigger a JIT commit to flash just before the device powers itself off. This can be very reliable if set up correctly, and early detection can be implemented externally, signalling an input to trigger an interrupt (preferrable non-maskable, top priority). In my experience there is typically ample time to detect the condition, write the flash and sit in an infinite loop until the µC dies even without special external precautions such as extra capacitors and I've never really had to resort to external early detection. Combined with circular buffer and some intelligence (only writing changed settings) as well as basic wear levelling can easily extend the service lifetime to at the very least tens of years of heavy use.

aldenhart, 2016-02-23:
This request is redundant with https://github.com/synthetos/g2/issues/24 so we are leaving that one open and closing this one. The Persistence issue is still open.

closed 2016-02-23 by aldenhart

diverseg, 2016-06-09:
@johnlauer,
I also like your little cog wheel next to the serial port in chilipeppr.
The eeprom essentially is your computer's hard drive.
Where is this startup script file saved? Is this documented somewhere?

johnlauer, 2016-06-09:
Yeah, that’s the way to solve the lack of EEPROM. That’s the sole reason I did it.