synthetos/g2 #58: Build issues under linux (order of object files on linker command line)

dhylands, 2015-03-03:
I was helping @buserror investigate why his firmware wasn't working properly.

Right now, I've reached the point in the investigation that it seems to be triggered by some differences between make 4.0 and make 3.81, specifically related to the order of the object files produced by $(wildcard).

On my local machine (ubuntu 14.04) if I build with make-3.81, I get the following linker line:

arm-none-eabi-g++  -T"/home/dhylands/g2/g2/TinyG2/platform/atmel_sam/gcc_flash.ld" -Wl,-Map,"bin/gShield/gShield.map" -o bin/gShield/gShield.elf -lgcc -lc  -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -nostartfiles -mcpu=cortex-m3 --specs=nano.specs  -u _printf_float  -mthumb   -lgcc -lc -Wl,--start-group build/gShield/motate/SamTimers.o build/gShield/motate/SamUSB.o build/gShield/motate/SamPins.o build/gShield/platform/atmel_sam/syscalls_sam3.o build/gShield/CMSIS/Device/ATMEL/sam3xa/source/system_sam3xa.o build/gShield/CMSIS/Device/ATMEL/sam3xa/source/gcc/startup_sam3xa.o build/gShield/platform/atmel_sam/cortex_handlers.o build/gShield/platform/atmel_sam/hooks.o build/gShield/motate/AvrUSB.o build/gShield/motate/SamSPI.o build/gShield/./canonical_machine.o build/gShield/./config_app.o build/gShield/./config.o build/gShield/./controller.o build/gShield/./cycle_homing.o build/gShield/./cycle_jogging.o build/gShield/./cycle_probing.o build/gShield/./encoder.o build/gShield/./gcode_parser.o build/gShield/./hardware.o build/gShield/./help.o build/gShield/./json_parser.o build/gShield/./kinematics.o build/gShield/./main.o build/gShield/./persistence.o build/gShield/./plan_arc.o build/gShield/./plan_exec.o build/gShield/./plan_line.o build/gShield/./planner.o build/gShield/./plan_zoid.o build/gShield/./pwm.o build/gShield/./report.o build/gShield/./spindle.o build/gShield/./stepper.o build/gShield/./switch.o build/gShield/./test.o build/gShield/./text_parser.o build/gShield/./util.o build/gShield/./xio.o build/gShield/platform/atmel_sam/Reset.o build/gShield/platform/atmel_sam/UniqueId.o -Wl,--end-group

This produces working firmware.

Using make 4.0, I get the following linker line:

arm-none-eabi-g++  -T"/home/dhylands/g2/g2/TinyG2/platform/atmel_sam/gcc_flash.ld" -Wl,-Map,"bin/gShield/gShield.map" -o bin/gShield/gShield.elf -lgcc -lc  -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -nostartfiles -mcpu=cortex-m3 --specs=nano.specs  -u _printf_float  -mthumb   -lgcc -lc -Wl,--start-group build/gShield/motate/SamTimers.o build/gShield/motate/SamUSB.o build/gShield/motate/SamPins.o build/gShield/platform/atmel_sam/syscalls_sam3.o build/gShield/CMSIS/Device/ATMEL/sam3xa/source/system_sam3xa.o build/gShield/CMSIS/Device/ATMEL/sam3xa/source/gcc/startup_sam3xa.o build/gShield/platform/atmel_sam/hooks.o build/gShield/platform/atmel_sam/cortex_handlers.o build/gShield/motate/AvrUSB.o build/gShield/motate/SamSPI.o build/gShield/./help.o build/gShield/./cycle_homing.o build/gShield/./text_parser.o build/gShield/./controller.o build/gShield/./switch.o build/gShield/./planner.o build/gShield/./plan_arc.o build/gShield/./hardware.o build/gShield/./kinematics.o build/gShield/./cycle_probing.o build/gShield/./pwm.o build/gShield/./plan_zoid.o build/gShield/./stepper.o build/gShield/./plan_line.o build/gShield/./json_parser.o build/gShield/./config_app.o build/gShield/./xio.o build/gShield/./cycle_jogging.o build/gShield/./report.o build/gShield/./config.o build/gShield/./canonical_machine.o build/gShield/./encoder.o build/gShield/./persistence.o build/gShield/./gcode_parser.o build/gShield/./main.o build/gShield/./plan_exec.o build/gShield/./test.o build/gShield/./spindle.o build/gShield/./util.o build/gShield/platform/atmel_sam/UniqueId.o build/gShield/platform/atmel_sam/Reset.o -Wl,--end-group

and this produces non-working firmware.

The only difference between these 2 lines is the order of the .o files.

I need to investigate further to determine if this is a problem with the code (depending on things being in a particular order) or a toolchain issue. 

So I figured I'd open this issue to at least make others aware of the issue, in case they're also experiencing similar problems, and I'll add to this as I discover more.

dhylands, 2015-03-04:
It looks like its an order of initialization problem.

If main.o occurs before xio.o on the linker command line, then the firmware boots up fine. If main.o occurs after xio.o on the linker command line, then the firmware doesn't boot up properly.

I think that what's happening is related to file-scoped C++ objects. The order that the .o files appear on the linker command line controls the order that the constructors for file-scoped objects are called in.

make 3.81 seems to sort the files returned by the filesystem when using $(wildcard). make 4.0 seems to use filesystem order.

I see a couple of ways that we can fix this:
1 - In the Makefile, replace:
```
C_SOURCES   = $(foreach dir, $(SOURCE_DIRS), $(wildcard $(dir)/*.c) )
CXX_SOURCES = $(foreach dir, $(SOURCE_DIRS), $(wildcard $(dir)/*.cpp) )
```
with
```
C_SOURCES   = $(foreach dir, $(SOURCE_DIRS), $(sort $(wildcard $(dir)/*.c)) )
CXX_SOURCES = $(foreach dir, $(SOURCE_DIRS), $(sort $(wildcard $(dir)/*.cpp)) )
```
This then gives a deterministic order, regardless of the version of make being used and causes main.o to appear before xio.o

2 - Construct CXX_SOURCES so that main.cpp is the first in the list. Something along the lines of:
```
CXX_ALL_SOURCES = $(foreach dir, $(SOURCE_DIRS), $(sort wildcard $(dir)/*.cpp)))
CXX_ORDERED_SOURCES = ./main.cpp
CXX_SOURCES = $(CXX_ORDERED_SOURCES $(filter-out $(CXX_ORDERED_SOURCES), $(CXX_ALL_SOURCES))
```

3 - Fix the order of initialization problem.

I think I'll do up a PR for option 2. Seems a bit more explicit. I'll also add some comments.

giseburt, 2015-03-04:
Thank you for hunting this down!

Link order is a PITA, but, we have seen this basic problem before. `FIRST_LINK_SOURCES` is a list of files (the source, not the resulting object) that have to link first.

`FIRST_LINK_SOURCES` already contains device-specific things that have to be early in the link order, so you'll need to add to it. We can try adding main.cpp to `FIRST_LINK_SOURCES` on line 34, but I suspect we'll need it after the device additions.

cmcgrath5035, 2015-03-04:
FYI, My openSUSE_64 system has make 4.0-2.2.3 .
Could be same as Dave's 4.0 (ubuntu?).
I am just watching  for now, let me know if you want another test run.

dhylands, 2015-03-04:
> FYI, My openSUSE_64 system has make 4.0-2.2.3 .
> Could be same as Dave's 4.0 (ubuntu?).
> I am just watching for now, let me know if you want another test run.

My ubuntu system actually comes with make 3.81. I installed 4.0 by downloading the sources and building it. I was helping out @buserror who had make 4.0 installed.

The way the makefile is currently coded whether make 4.0 produces a viable firmware is basically a crapshoot, and depends entirely on what order the filesystem returns the files in.

dhylands, 2015-03-04:
> FIRST_LINK_SOURCES already contains device-specific things that have to be early in the link order, so you'll need to add to it. We can try adding main.cpp to FIRST_LINK_SOURCES on line 34, but I suspect we'll need it after the device additions.

I'll play around a bit with FIRST_LINK_SOURCES and see what I can get to work. If I get something that works, then I'll update my PR.

dhylands, 2015-03-05:
main.o need to be after the motate objects and somewhere before xio.o

So using FIRST_LINK_SOURCES isn't an option. We could just sort the source files (main is alphabetically before xio so that happens to work), or use a sorted list with main.o at the beginning of the list (this is what PR #59 does).

If you just want to do the sorting, I'm happy to change my PR.

giseburt, 2015-03-31:
This should have been closed with #59. Thanks again, @dhylands !

closed 2015-03-31 by giseburt