synthetos/g2 #60: Accessory axes don't scale properly

ryansturmer, 2015-03-12:
I have only tested this with the B axis, but in our system, I do not get sensible results for the scaling on the B axis.  

Setting the axis up with bam=1 (linear mode) and sensible defaults for tr,mi, and sa, I get unexpected movement distances, which I expect is the axis behaving in a rotational fashion, rather than in a linear fashion as expected.  For instance a G0B10 which I expect to jog the axis to 10 inches, jogs to 0.394 inches.  All moves on this axis move with a similar scaling factor.

tedh-shopbot, 2015-03-13:
This just goes to a concept that I was trying to convince Alden of earlier ... at the lowest motion level there should be only one number for setting up a motor >>  steps/unit. Everything else is higher level; introducing notions like microstepping, gear ratio, rotary vs linear etc ... is just confusing.

aldenhart, 2015-03-17:
Ted and I talked about this on the phone. We intend to expose the raw steps_per_unit values in a future release --- units being mm, inches, or degrees, depending. Our complication and the reason we have not done this so far is that we have drivers on board that have software settable microsteps and other values that affect this value. Currently if you adjust the travel per revolution, motor step angle or microstep settings the steps_per_unit value is automatically calculated and stored. Our intention is to expose steps_per_unit in such a way to preserve this functionality, or allow direct entry of steps_per_unit.

closed 2015-03-17 by aldenhart