synthetos/g2 #62: g2 in win8.1

ryansturmer, 2015-03-19:
No love for G2 in windows 8.1

Enabled the installation of unsigned drivers, and made an attempt to install.  Even with changing https://github.com/ShopBotTools/FabMo-G2-Core/blob/edge/TinyGv2.inf#L12 to TinyGv2.inf we don't get a valid installation.

The control channel installs after some fiddling, and shows up as a COM port in the device manager, but the data port still remains as an "unknown device"

gordonab, 2015-03-19:
[image]

ryansturmer, 2015-03-19:
This inf works much better for us on win8.1:  https://gist.github.com/ryansturmer/b6df971326e4cb6ea557  Would welcome confirmation from another user.

stg, 2015-03-30:
This inf has worked nicely for me on both W7 and W8.

aldenhart, 2016-02-23:
We have given up on Windows8 and do not support it. We have moved on the Windows10. So far so good, but reports are still coming in.

closed 2016-02-23 by aldenhart