synthetos/g2 #63: Compiling fail

qtfp, 2015-03-21:
Dear all, 
    Thanks for this project, I really need this arduino due version of motion control system. Recently we are going to modify our lathe and milling machine to CNC. TinyG2 is a good candidate!!
    Now We try to use arduino due board connecting to the motor driver without Shield, so we need to modify the pin and lots of configuration.  We have tried compiling by the win7+ Atmel Studio 6.2 and  Ubuntu 14.04, but they all failed during compiling.
Here is the failing code in  Ubuntu 14.04:
In file included from ./xio.h:70:0,
                 from ./config_app.cpp:46:
motate/MotateUSBCDC.h:35:22: fatal error: functional: No such file or directory
 #include <functional>
                      ^
compilation terminated.
make: *** [build/gShield/./config_app.o] Error 1

Is there any suggestion to compile it right? 
I compiled them all by the g2 wiki pages.....

giseburt, 2015-03-21:
Hi @qtfp,

I'm glad you like out project! We've put a lot of work into it. :-)

Unfortunately, I'm not sure how to help, since I cannot tell what changes you've made. Do you have a fork with pushed changes we could look at to tell what may have change?

Also, were you able to compile an unchanged clone of the project? Please make sure you can (or can't) compile it without any changes first. That way we can determine if the problem is general or specific to your code changes.

-Rob

qtfp, 2015-03-22:
Hello @giseburt ,
I didn't make any change yet, just did as mentioned here:
https://github.com/synthetos/g2/wiki/Compiling-G2-on-Linux-and-OS-X-%28command-line%29
sudo apt-get install git-core gcc make
mkdir -p ~/src/github.com/synthetos
cd ~/src/github.com/synthetos
git clone https://github.com/synthetos/g2 g2
cd ./g2
git checkout edge
cd TinyG2
make
Then it gave:
qtfp@ubuntu:~/src/github.com/synthetos/g2/TinyG2$ make
Compiling c CMSIS/Device/ATMEL/sam3xa/source/system_sam3xa.c
    -> build/gShield/CMSIS/Device/ATMEL/sam3xa/source/system_sam3xa.o 
Compiling c CMSIS/Device/ATMEL/sam3xa/source/gcc/startup_sam3xa.c
    -> build/gShield/CMSIS/Device/ATMEL/sam3xa/source/gcc/startup_sam3xa.o 
Compiling c platform/atmel_sam/cortex_handlers.c
    -> build/gShield/platform/atmel_sam/cortex_handlers.o 
Compiling c platform/atmel_sam/hooks.c
    -> build/gShield/platform/atmel_sam/hooks.o 
Compiling cpp motate/AvrUSB.cpp
    -> build/gShield/motate/AvrUSB.o 
Compiling cpp motate/SamPins.cpp
    -> build/gShield/motate/SamPins.o 
In file included from motate/SamPins.cpp:33:0:
motate/utility/SamPins.h:39:45: fatal error: type_traits: No such file or directory
 #include <type_traits> // for std::enable_if
                                             ^
compilation terminated.
make: *** [build/gShield/motate/SamPins.o] Error 1

I also tried this:
git clone -b master https://github.com/synthetos/g2 g2
cd g2/TinyG2
make
got this:
Looking for tools...
cd ../Tools && make "ARCH=gcc-arm-none-eabi"
make[1]: Entering directory `/home/qtfp/g2/Tools'
make[1]: Nothing to be done for `all'.
make[1]: Leaving directory `/home/qtfp/g2/Tools'
Compiling c CMSIS/Device/ATMEL/sam3xa/source/system_sam3xa.c
    -> build/gShield/CMSIS/Device/ATMEL/sam3xa/source/system_sam3xa.o 
Compiling c CMSIS/Device/ATMEL/sam3xa/source/gcc/startup_sam3xa.c
    -> build/gShield/CMSIS/Device/ATMEL/sam3xa/source/gcc/startup_sam3xa.o 
Compiling c platform/atmel_sam/cortex_handlers.c
    -> build/gShield/platform/atmel_sam/cortex_handlers.o 
Compiling c platform/atmel_sam/hooks.c
    -> build/gShield/platform/atmel_sam/hooks.o 
Compiling cpp motate/AvrUSB.cpp
    -> build/gShield/motate/AvrUSB.o 
Compiling cpp motate/SamPins.cpp
    -> build/gShield/motate/SamPins.o 
Compiling cpp motate/SamSPI.cpp
    -> build/gShield/motate/SamSPI.o 
Compiling cpp motate/SamTimers.cpp
    -> build/gShield/motate/SamTimers.o 
Compiling cpp motate/SamUSB.cpp
    -> build/gShield/motate/SamUSB.o 
Compiling cpp ./canonical_machine.cpp
    -> build/gShield/./canonical_machine.o 
Compiling cpp ./config_app.cpp
    -> build/gShield/./config_app.o 
In file included from ./xio.h:70:0,
                 from ./config_app.cpp:46:
motate/MotateUSBCDC.h:35:22: fatal error: functional: No such file or directory
 #include <functional>
                      ^
compilation terminated.
make: *** [build/gShield/./config_app.o] Error 1

Thanks for reply :)

giseburt, 2015-03-23:
How very odd. I haven't seen these issues before.

Could you try these commands and let me know the results?

```bash
# make sure we're in the <project>/TinyG2 directory first.

# Let's make sure the git is clean and correct:
git reset --hard
git clean -n
git checkout edge
git fetch --all
git pull

# Now with edge checked out and updated:
cd ../Tools
make "ARCH=gcc-arm-none-eabi"
gcc-arm-none-eabi/bin/arm-none-eabi-gcc -v
# That should dump a bunch of info, the last line of which should read:
# gcc version 4.8.3 20140228 (release) [ARM/embedded-4_8-branch revision 208322] (GNU Tools for ARM Embedded Processors)

# Now we'll compile again, forcing that compiler chain to be the one used:
cd ../TinyG2
PATH=../Tools/gcc-arm-none-eabi:$PATH make VERBOSE=1 COLOR=0
```

If you don't have the correct `arm-none-eabi-gcc` or the compile still fails.

qtfp, 2015-03-25:
qtfp@ubuntu:~/g2/Tools$ make "ARCH=gcc-arm-none-eabi"
make: Nothing to be done for `all'.
qtfp@ubuntu:~/g2/Tools$ gcc-arm-none-eabi/bin/arm-none-eabi-gcc -v
bash: gcc-arm-none-eabi/bin/arm-none-eabi-gcc: No such file or directory
under my vm ubuntu 14.04.1 lts

Don't get any way to compile smoothly yet.
Also is there any possible post processor that solidcam can use?
Thanks for your help!

mikejac, 2015-03-28:
Hi,

I'm experiencing the same :-(

Ubuntu 14.04.2 LTS (GNU/Linux 3.13.0-48-generic x86_64)

Linux ubuntu 3.13.0-48-generic #80-Ubuntu SMP Thu Mar 12 11:16:15 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

mikejac, 2015-03-28:
More info:

Followed these instructions: https://github.com/synthetos/g2/wiki/Compiling-G2-on-Linux-and-OS-X-%28command-line%29

First time I ran make, I got this:

michael@ubuntu:~/src/github.com/synthetos/g2/TinyG2$ make
Compiling c CMSIS/Device/ATMEL/sam3xa/source/system_sam3xa.c
    -> build/gShield/CMSIS/Device/ATMEL/sam3xa/source/system_sam3xa.o
bash: arm-none-eabi-gcc: command not found
make: *** [build/gShield/CMSIS/Device/ATMEL/sam3xa/source/system_sam3xa.o] Error 127

Apparently no arm-none-eabi-gcc available... so I ran this to get the ARM GCC:

sudo apt-get install gcc-arm-none-eabi

$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (4.8.2-14ubuntu1+6) 4.8.2
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

giseburt, 2015-03-29:
I'm looking into this.

@mikejac Were you able to compile after that? It seems like a reasonable workaround.

BTW, we know that this isn't a problem for all linux distros, since Travis CI is able to compile and it's running linux.

I'll update when I know more.

mikejac, 2015-03-29:
Sorry I didn't say so, but no, not able to compile :-(

For me it's not a high priority - I went the Atmel Studio way instead...

giseburt, 2015-03-29:
What error are you getting now? The same or something different?

giseburt, 2015-03-31:
@mikejac I pushed changes recently (commit e5ab2de6505690bc58048acac86d4b513b108d9b) that includes Makefile fixes. Could you pull and give the new makefiles a go?

Thank you!

juanblanco, 2015-04-13:
Hi, this is how I get it work:

!!! If you are using Ubuntu 14.04 and later, please be careful because there are packages with same name but produced by Debian and inherited by Ubuntu. Simply follow the above 3 steps, you may end up with gcc-arm-none-eabi from Ubuntu. So to install gcc-arm-none-eabi from ARM, steps are:
1). sudo apt-get remove binutils-arm-none-eabi gcc-arm-none-eabi
2). sudo add-apt-repository ppa:terry.guo/gcc-arm-embedded
3). sudo apt-get update
4). sudo apt-get install gcc-arm-none-eabi=4.9.3.2015q1-0trusty13
    or
    sudo apt-get install gcc-arm-none-eabi=4.9.3.2015q1-0utopic14

Meanwhile we are working with Debian to consolidate and unify this toolchain.

qtfp, 2015-04-13:
@juanblanco  Thanks for your hint! It actually work!!!

aShure, 2015-12-18:
I think this is an unfortunate chain of bad error messages. The Makefile fetches a 32bit gcc-arm-none-eabi package on the first run and deflates it. It's hardcoded in the Makefile to use that toolchain then, which is not possible if you are on a 64bit OS and throws an error which is misleading: Can't find the file or directory. The usual fix is to install the Toolchain via apt-get, which will fetch the most recent version as a x64 binary. Then you might get an error like 
```
motate/MotateUSBCDC.h:35:22: fatal error: functional: No such file or directory
#include
^
compilation terminated.
```

A fix for this would be to just ``apt-get install lib32z1``, as I briefly explained in the Wiki recently on the Compilation in Linux page and not install an x64 Toolchain via apt-get. The lib32z1 will enable you to run the supplied 32bit Toolchain in /Tools and ensure full compatibility with the G2 codebase.

Maybe someone can add an installation of lib32z1 to the Makefile if an x64 Linux is detected..

closed 2016-02-23 by aldenhart