synthetos/g2 #64: Fail Build Edge version G2 with win8.1 Atmel Studio 6.2

cmcta82, 2015-03-22:
I´m trying to build the last edge version with win8.1 atmel studio 6.2 but it fail with the warning:
 Warning	1	#pragma GCC target is not supported for this machine [-Wpragmas]	D:\Downloads\g2-edge (1)\g2-edge\TinyG2\planner.cpp	692	9	TinyG2

It doesn´t make any Bin file, i use Gshield in solution configuration, i was able to build old version from  edge and master without any problem with the same configuration.

giseburt, 2015-03-22:
That warning is a bug in the arm GCC. It's actually harmless.

The bin and elf files should in the bin/gShield/ directory.

If it's not creating those files could you please post the complete output from make?

Thank you,
  -Rob

cmcta82, 2015-03-22:
The folder is empty and is the only error i have, i will try with master to see if i have the same issue

giseburt, 2015-03-31:
We use Atmel Studio fairly heavily (well, half of us ;), so I'd be surprised to find a problem with just AS 6.1. We use with with Win7 however.

Please let us know if this is still an issue for you.

closed 2016-02-23 by aldenhart