synthetos/g2 #68: homing config values incorrect in G20 mode

lllars, 2015-03-28:
In G20 mode (inches), the values of $xhi and $xhd always display as zero.  The values display correctly in G21 mode.  When in G20 mode it appears as though attempted changes to the values don't stick, for example typing:

$xhi=1
$xhi

returns 0.  However the values actually do stick, as can be seen by switching back to G21.  Despite this issue, homing does appear to work correctly while in G20 mode.  Presumably the $yhi, $yhd, $zhi, and $zhd values are also affected by this bug, but I did not verify that to be the case.

aldenhart, 2015-03-28:
Thanks for pointing this out. 

FYI: I'm working on the wiki documentation for all the changes. Starting here:
https://github.com/synthetos/g2/wiki/Changelog-for-Edge-Branches

aldenhart, 2015-03-29:
Fixed

closed 2015-03-29 by aldenhart