synthetos/g2 #70: Build error under Xcode (6.2)

TFlanigan, 2015-03-30:
I am getting the following build error.

build/G2v9i/dep/persistence.o.d -MT build/G2v9i/./persistence.o -xc++ -c -o build/G2v9i/./persistence.o ./persistence.cpp
arm-none-eabi-g++: internal compiler error: Killed: 9 (program cc1plus)
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
make: *** [build/G2v9i/./persistence.o] Error 4
Command /usr/bin/make failed with exit code 2

Full output follows:
---------------------------
ExternalBuildToolExecution TinyG2\ G2V9i
    cd /Developer/tinyG2/g2-master/TinyG2/.
    export ACTION=
    export ALTERNATE_GROUP=staff
    export ALTERNATE_MODE=u+w,go-w,a+rX
    export ALTERNATE_OWNER=tflanigan
    export ALWAYS_SEARCH_USER_PATHS=YES
    export APPLE_INTERNAL_DEVELOPER_DIR=/AppleInternal/Developer
    export APPLE_INTERNAL_DIR=/AppleInternal
    export APPLE_INTERNAL_DOCUMENTATION_DIR=/AppleInternal/Documentation
    export APPLE_INTERNAL_LIBRARY_DIR=/AppleInternal/Library
    export APPLE_INTERNAL_TOOLS=/AppleInternal/Developer/Tools
    export APPLICATION_EXTENSION_API_ONLY=NO
    export APPLY_RULES_IN_COPY_FILES=NO
    export ARCHS=x86_64
    export AVAILABLE_PLATFORMS="iphonesimulator macosx iphoneos"
    export BUILD_COMPONENTS="headers build"
    export BUILD_DIR=/Users/tflanigan/Library/Developer/Xcode/DerivedData/TinyG2-bbhbwgalqsqkeeauxvduqmqqymis/Build/Products
    export BUILD_ROOT=/Users/tflanigan/Library/Developer/Xcode/DerivedData/TinyG2-bbhbwgalqsqkeeauxvduqmqqymis/Build/Products
    export BUILD_STYLE=
    export BUILD_VARIANTS=normal
    export BUILT_PRODUCTS_DIR=/Users/tflanigan/Library/Developer/Xcode/DerivedData/TinyG2-bbhbwgalqsqkeeauxvduqmqqymis/Build/Products/Debug
    export CACHE_ROOT=/var/folders/jc/t87dxm9n3y52qxmk2dmrg34c0000gn/C/com.apple.DeveloperTools/6.2-6C131e/Xcode
    export CLASS_FILE_DIR="/Users/tflanigan/Library/Developer/Xcode/DerivedData/TinyG2-bbhbwgalqsqkeeauxvduqmqqymis/Build/Intermediates/TinyG2.build/Debug/TinyG2 G2V9i.build/JavaClasses"
    export CLONE_HEADERS=NO
    export CODE_SIGNING_ALLOWED=NO
    export COLOR_DIAGNOSTICS=NO
    export CONFIGURATION=Debug
    export CONFIGURATION_BUILD_DIR=/Users/tflanigan/Library/Developer/Xcode/DerivedData/TinyG2-bbhbwgalqsqkeeauxvduqmqqymis/Build/Products/Debug
    export CONFIGURATION_TEMP_DIR=/Users/tflanigan/Library/Developer/Xcode/DerivedData/TinyG2-bbhbwgalqsqkeeauxvduqmqqymis/Build/Intermediates/TinyG2.build/Debug
    export COPYING_PRESERVES_HFS_DATA=NO
    export COPY_PHASE_STRIP=YES
    export CREATE_INFOPLIST_SECTION_IN_BINARY=NO
    export DEAD_CODE_STRIPPING=NO
    export DEBUGGING_SYMBOLS=YES
    export DEFAULT_COMPILER=com.apple.compilers.llvm.clang.1_0
    export DEFAULT_KEXT_INSTALL_PATH=/Library/Extensions
    export DEFINES_MODULE=NO
    export DEPLOYMENT_LOCATION=NO
    export DEPLOYMENT_POSTPROCESSING=NO
    export DERIVED_FILES_DIR=
    export DERIVED_FILE_DIR="/Users/tflanigan/Library/Developer/Xcode/DerivedData/TinyG2-bbhbwgalqsqkeeauxvduqmqqymis/Build/Intermediates/TinyG2.build/Debug/TinyG2 G2V9i.build/DerivedSources"
    export DERIVED_SOURCES_DIR="/Users/tflanigan/Library/Developer/Xcode/DerivedData/TinyG2-bbhbwgalqsqkeeauxvduqmqqymis/Build/Intermediates/TinyG2.build/Debug/TinyG2 G2V9i.build/DerivedSources"
    export DEVELOPER_APPLICATIONS_DIR=/Applications/Xcode.app/Contents/Developer/Applications
    export DEVELOPER_BIN_DIR=/Applications/Xcode.app/Contents/Developer/usr/bin
    export DEVELOPER_DIR=/Applications/Xcode.app/Contents/Developer
    export DEVELOPER_FRAMEWORKS_DIR=/Applications/Xcode.app/Contents/Developer/Library/Frameworks
    export DEVELOPER_FRAMEWORKS_DIR_QUOTED=/Applications/Xcode.app/Contents/Developer/Library/Frameworks
    export DEVELOPER_LIBRARY_DIR=/Applications/Xcode.app/Contents/Developer/Library
    export DEVELOPER_SDK_DIR=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs
    export DEVELOPER_TOOLS_DIR=/Applications/Xcode.app/Contents/Developer/Tools
    export DEVELOPER_USR_DIR=/Applications/Xcode.app/Contents/Developer/usr
    export DEVELOPMENT_LANGUAGE=English
    export DO_HEADER_SCANNING_IN_JAM=NO
    export DSTROOT=/tmp/TinyG2.dst
    export DT_TOOLCHAIN_DIR=/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain
    export DWARF_DSYM_FILE_NAME=.dSYM
[... 650 lines cut ...]
    -> build/G2v9i/./help.o
arm-none-eabi-g++  -Wall -Wchar-subscripts -Wcomment -Wformat=2 -Wmain -Wparentheses -Wunreachable-code -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wno-unknown-pragmas -Wfloat-equal -Wundef -Wshadow -Wpointer-arith -Wwrite-strings -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wredundant-decls -Wlong-long -Wno-format-nonliteral -Wno-format-security -D__SAM3X8C__ --param max-inline-insns-single=500 -mcpu=cortex-m3 -mthumb -mlong-calls -ffunction-sections -fdata-sections -fno-rtti -std=c++11 -fno-exceptions -u _printf_float -DPLATFORM=G2v9i -g3 -Os -Iplatform/atmel_sam/board/v9_3x8c -I"CMSIS/CMSIS/Include" -I"CMSIS/Device/ATMEL" -I"CMSIS/Device/ATMEL/sam3xa/include" -I./platform/atmel_sam -I. -Imotate -DMOTATE_BOARD="G2v9i" -DSETTINGS_FILE=settings_default.h -g3 -Os -Iplatform/atmel_sam/board/v9_3x8c -I"CMSIS/CMSIS/Include" -I"CMSIS/Device/ATMEL" -I"CMSIS/Device/ATMEL/sam3xa/include" -I./platform/atmel_sam -I. -Imotate -DMOTATE_BOARD="G2v9i" -DSETTINGS_FILE=settings_default.h -MMD -MF build/G2v9i/dep/help.o.d -MT build/G2v9i/./help.o -xc++ -c -o build/G2v9i/./help.o ./help.cpp
mkdir -p "build/G2v9i/." "build/G2v9i/dep" "bin/G2v9i"
Compiling cpp ./json_parser.cpp
    -> build/G2v9i/./json_parser.o
arm-none-eabi-g++  -Wall -Wchar-subscripts -Wcomment -Wformat=2 -Wmain -Wparentheses -Wunreachable-code -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wno-unknown-pragmas -Wfloat-equal -Wundef -Wshadow -Wpointer-arith -Wwrite-strings -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wredundant-decls -Wlong-long -Wno-format-nonliteral -Wno-format-security -D__SAM3X8C__ --param max-inline-insns-single=500 -mcpu=cortex-m3 -mthumb -mlong-calls -ffunction-sections -fdata-sections -fno-rtti -std=c++11 -fno-exceptions -u _printf_float -DPLATFORM=G2v9i -g3 -Os -Iplatform/atmel_sam/board/v9_3x8c -I"CMSIS/CMSIS/Include" -I"CMSIS/Device/ATMEL" -I"CMSIS/Device/ATMEL/sam3xa/include" -I./platform/atmel_sam -I. -Imotate -DMOTATE_BOARD="G2v9i" -DSETTINGS_FILE=settings_default.h -g3 -Os -Iplatform/atmel_sam/board/v9_3x8c -I"CMSIS/CMSIS/Include" -I"CMSIS/Device/ATMEL" -I"CMSIS/Device/ATMEL/sam3xa/include" -I./platform/atmel_sam -I. -Imotate -DMOTATE_BOARD="G2v9i" -DSETTINGS_FILE=settings_default.h -MMD -MF build/G2v9i/dep/json_parser.o.d -MT build/G2v9i/./json_parser.o -xc++ -c -o build/G2v9i/./json_parser.o ./json_parser.cpp
mkdir -p "build/G2v9i/." "build/G2v9i/dep" "bin/G2v9i"
Compiling cpp ./kinematics.cpp
    -> build/G2v9i/./kinematics.o
arm-none-eabi-g++  -Wall -Wchar-subscripts -Wcomment -Wformat=2 -Wmain -Wparentheses -Wunreachable-code -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wno-unknown-pragmas -Wfloat-equal -Wundef -Wshadow -Wpointer-arith -Wwrite-strings -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wredundant-decls -Wlong-long -Wno-format-nonliteral -Wno-format-security -D__SAM3X8C__ --param max-inline-insns-single=500 -mcpu=cortex-m3 -mthumb -mlong-calls -ffunction-sections -fdata-sections -fno-rtti -std=c++11 -fno-exceptions -u _printf_float -DPLATFORM=G2v9i -g3 -Os -Iplatform/atmel_sam/board/v9_3x8c -I"CMSIS/CMSIS/Include" -I"CMSIS/Device/ATMEL" -I"CMSIS/Device/ATMEL/sam3xa/include" -I./platform/atmel_sam -I. -Imotate -DMOTATE_BOARD="G2v9i" -DSETTINGS_FILE=settings_default.h -g3 -Os -Iplatform/atmel_sam/board/v9_3x8c -I"CMSIS/CMSIS/Include" -I"CMSIS/Device/ATMEL" -I"CMSIS/Device/ATMEL/sam3xa/include" -I./platform/atmel_sam -I. -Imotate -DMOTATE_BOARD="G2v9i" -DSETTINGS_FILE=settings_default.h -MMD -MF build/G2v9i/dep/kinematics.o.d -MT build/G2v9i/./kinematics.o -xc++ -c -o build/G2v9i/./kinematics.o ./kinematics.cpp
mkdir -p "build/G2v9i/." "build/G2v9i/dep" "bin/G2v9i"
Compiling cpp ./main.cpp
    -> build/G2v9i/./main.o
arm-none-eabi-g++  -Wall -Wchar-subscripts -Wcomment -Wformat=2 -Wmain -Wparentheses -Wunreachable-code -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wno-unknown-pragmas -Wfloat-e [... cut ...] TINGS_FILE=settings_default.h -MMD -MF build/G2v9i/dep/persistence.o.d -MT build/G2v9i/./persistence.o -xc++ -c -o build/G2v9i/./persistence.o ./persistence.cpp
arm-none-eabi-g++: internal compiler error: Killed: 9 (program cc1plus)
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
make: *** [build/G2v9i/./persistence.o] Error 4
Command /usr/bin/make failed with exit code 2

giseburt, 2015-03-31:
The only time I've seen similar before was when there was another compiler installer elsewhere in the path.

Can you open terminal and send back the results of this command:

  ```bash
  PATH="/Applications/Xcode.app/Contents/Developer/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin" which arm-none-eabi-g++
  ```

Ideally it'll return nothing, indicating that it didn't find one. I suspect it will find one, however.

One other thing is to make sure that there aren't any spaces or shell characters (~!@#$%(){}[]*) in your username or any of the directories up to the and including the one that is the TinyG2 repo. Some utilities don't handle those well, and there's very little to be done about it from our end, unfortunately.

-Rob

TFlanigan, 2015-03-31:
I overlooked installing the proper gcc.
The make file seems to have tried to do the install itself but it didn't
seem to work.
I downloaded and install it by hand.  I set up the paths but I'm still
having issues.

which arm-none-eabi-g++ returns the following

>which arm-none-eabi-g++

/usr/gcc-arm-none-eabi-4_9-2015q1/bin/arm-none-eabi-g++

*XCode fails with the following.*

Compiling cpp motate/AvrUSB.cpp
    -> build/G2v9d/motate/AvrUSB.o
arm-none-eabi-g++  -Wall -Wchar-subscripts -Wcomment -Wformat=2 -Wmain
-Wparentheses -Wunreachable-code -Wsequence-point -Wreturn-type
-Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wno-unknown-pragmas
-Wfloat-equal -Wundef -Wshadow -Wpointer-arith -Wwrite-strings
-Wformat -Wmissing-format-attribute -Wno-deprecated-declarations
-Wredundant-decls -Wlong-long -Wno-format-nonliteral
-Wno-format-security -D__SAM3X8C__ --param max-inline-insns-single=500
-mcpu=cortex-m3 -mthumb -mlong-calls -ffunction-sections
-fdata-sections -fno-rtti -std=c++11 -fno-exceptions -u _printf_float
-DPLATFORM=G2v9d -g3 -Os -Iplatform/atmel_sam/board/v9_3x8c
-I"CMSIS/CMSIS/Include" -I"CMSIS/Device/ATMEL"
-I"CMSIS/Device/ATMEL/sam3xa/include" -I./platform/atmel_sam -I.
-Imotate -DMOTATE_BOARD="G2v9d" -DSETTINGS_FILE=settings_default.h -g3
-Os -Iplatform/atmel_sam/board/v9_3x8c -I"CMSIS/CMSIS/Include"
-I"CMSIS/Device/ATMEL" -I"CMSIS/Device/ATMEL/sam3xa/include"
-I./platform/atmel_sam -I. -Imotate -DMOTATE_BOARD="G2v9d"
-DSETTINGS_FILE=settings_default.h -MMD -MF build/G2v9d/dep/AvrUSB.o.d
-MT build/G2v9d/motate/AvrUSB.o -xc++ -c -o
build/G2v9d/motate/AvrUSB.o motate/AvrUSB.cpp
arm-none-eabi-g++: internal compiler error: Killed: 9 (program cc1plus)
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
make: *** [build/G2v9d/motate/AvrUSB.o] Error 4
Command /usr/bin/make failed with exit code 2

------

*FULL OUTPUT FOLLOWS*

------

ExternalBuildToolExecution TinyG2\ G2V9d

    cd /Developer/tinyG2/g2-master/TinyG2/.

    export ACTION=

    export ALTERNATE_GROUP=staff

    export ALTERNATE_MODE=u+w,go-w,a+rX

    export ALTERNATE_OWNER=tflanigan

    export ALWAYS_SEARCH_USER_PATHS=YES

    export APPLE_INTERNAL_DEVELOPER_DIR=/AppleInternal/Developer

    export APPLE_INTERNAL_DIR=/AppleInternal

    export APPLE_INTERNAL_DOCUMENTATION_DIR=/AppleInternal/Documentation

    export APPLE_INTERNAL_LIBRARY_DIR=/AppleInternal/Library
[... 476 lines cut ...]
-ffunction-sections -fdata-sections -fno-rtti -std=c++11 -fno-exceptions -u
_printf_float -DPLATFORM=G2v9d -g3 -Os -Iplatform/atmel_sam/board/v9_3x8c
-I"CMSIS/CMSIS/Include" -I"CMSIS/Device/ATMEL"
-I"CMSIS/Device/ATMEL/sam3xa/include" -I./platform/atmel_sam -I. -Imotate
-DMOTATE_BOARD="G2v9d" -DSETTINGS_FILE=settings_default.h -g3 -Os
-Iplatform/atmel_sam/board/v9_3x8c -I"CMSIS/CMSIS/Include"
-I"CMSIS/Device/ATMEL" -I"CMSIS/Device/ATMEL/sam3xa/include"
-I./platform/atmel_sam -I. -Imotate -DMOTATE_BOARD="G2v9d"
-DSETTINGS_FILE=settings_default.h -MMD -MF build/G2v9d/dep/AvrUSB.o.d -MT
build/G2v9d/motate/AvrUSB.o -xc++ -c -o build/G2v9d/motate/AvrUSB.o
motate/AvrUSB.cpp

arm-none-eabi-g++: internal compiler error: Killed: 9 (program cc1plus)

libbacktrace could not find executable to open

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

make: *** [build/G2v9d/motate/AvrUSB.o] Error 4

Command /usr/bin/make failed with exit code 2

giseburt, 2015-05-09:
Hi @TFlanigan,

Someone else opened issue #90 and it jogged my memory about this issue.

According tot he output above, you have cloned the repo at: `/Developer/tinyG2/g2-master/TinyG2/TinyG2.xcodeproj`

The `/Developer` directory is used and maintained by XCode, and having your project(s) cloned there may be the source of this bizarre error. In the least you are likely to have permissions errors, and if you use `sudo` in there you are likely to break your XCode. (I wonder if you may already need to [uninstall XCode](http://osxdaily.com/2012/02/20/uninstall-xcode/) and [reinstall XCode](https://github.com/synthetos/g2/wiki/Compiling-G2-on-OS-X-(with-Xcode)#whats-needed).)

Can you try moving the clone into a directory under your home directory? Please make sure that *all directories in the path* from the root contain only alphanumerics (plus `-`, `_`, or `.`, but no spaces).

Thank you,
  -Rob

closed 2016-02-23 by aldenhart