synthetos/g2 #71: Add option to invert enable pins

lllars, 2015-04-01:
Specifically, I need the common enable pin inverted (or maybe un-inverted?).  Currently I've got this working by changing `common_enable.set()` to `common_enable.clear()` and vice-versa in stepper.cpp.  There could be a single variable that inverts all enable pins, or separate variables for each pin.

aldenhart, 2015-04-05:
It's a good suggestion. We will probably add this in order to make the system more general.

anthonywebb, 2015-05-30:
Man, would totally kill for this.  My clearpath servo motors need to be pulled HIGH to enable, instead of the pull LOW default.  I hat to add hardware to my shield just to invert something that could possibly be specified in the config.  Any chance this is on the roadmap @aldenhart ?

Video of the motor in action!
https://youtu.be/qGRKC4irYcU

aldenhart, 2015-05-31:
Yes, it's on the roadmap. We have already refactored the digital input system to be more general. The digital output system will be similarly recast. I don't have a time frame for you as there are some other things we are working on that will need to happen first. 

In the mean time you can hack the code to do this in a fork. Instructions on building and so forth are on the wiki. The changes would be in the stepper.cpp file.

anthonywebb, 2015-07-22:
I've managed to get this pretty close.  What I did was in stepper.cpp I replaced all the common_enable.clear();  with common_enable.set(); and vice-versa (total of 5 lines of code replaced).  I recompiled and re-flashed and it looks like the enable pin is working the way I need, but the motor wont move.  @aldenhart can you think of what else I might need to hack in addition to the above to get it to work?

lllars, 2015-07-22:
There are 7 lines to change by my count.  Maybe you missed a couple?  See my changes to stepper.cpp here https://github.com/synthetos/g2/compare/edge...lllars:edge

anthonywebb, 2015-07-30:
@lllars that did indeed do the trick!  I must have missed the other two.  Working well now.

closed 2016-02-23 by aldenhart