synthetos/g2 #72: Missed acc-/dec-ellerations

stg, 2015-04-06:
From #65:

    When issuing multiple fast feeds with short duration on an empty queue (or double-clicking
    move buttons in Chilipeppr with say 10mm). Replicated easily when machine is still by
    issuing two G0 X10 (G91 mode) immediately after each other.
    
    This will accelerate, feed, decelerate for the first move, but then jump straight back to full
    feed speed at the start of the second command (acceleration is skipped), followed by
    normal deceleration.
    
    This causes loss of tracking every time one accidentally hits the move buttons too quickly.

[edge 82.11]

Today I observed similar behavior while running a program containing many short segments - movements without acceleration are being performed causing loss of tracking.

I can't be sure that queue starvation is not occurring, as I don't know of a sure fire way to detect this (is there a way?), however the queue indicator in Chilipeppr never really goes beyond 2/3, but this also occurs when the queue is indicated to be pretty much full (far left).

The problem is **repeatable**.

Is there anything known that should cause this behavior?

Again, the g-code contains *many short segments in the areas where the problem is observed* and I am guessing this is directly related. Most of the moves incorrectly performed are 0.25mm long, following and followed by moves at 90° bends of similar length.

Let me know if there is interest in the configuration and/or offending g-code.

ril3y, 2015-04-06:
A quick way to test this out.  Use Coolterm and send the file. Make sure
flow control is set to dts /rts in coolterm settings.  This will take
chilipeppr out of the equation.  There is a limit to how fast serial comms
can be sent.  So if you are sending 60+ bytes to make a .0025 move... Well
this is not gong to work quickly.  I would ask what you are using to
generate your gcode and if there are optimizations to be done in the cams
software to eliminate the need for so many lines.

See if cool term works correctly (read no jitters).

giseburt, 2015-04-06:
@stg Could you test with the dge-replan-replan branch to see if it fixes the issue?

Thank you,
  -Rob

stg, 2015-04-06:
@giseburt I will certainly try that branch the next time I have a chance (Wednesday).
@ril3y I will try this as well, and while we are on the subject...

I've read about transmission speed in the wiki, but you keep referring to baudrate as a measurement of transmission speed. I've never heard of baudrate having any effect on a virtual (usb) serial port, unless there is a backend where the usb data is actually transformed into a serial stream at the configured baudrate (as with the FTDI FT232 for example). Are you somehow implementing artifical rate control based on the baudrate setting or does this information only apply to the classic TinyG?

I understand G2 is doing things a bit different than the typical VCP-device. Most devices only receive data when they are able to (buffers available). On G2 it seems data is always being received, but discarded if buffers are full - essentially re-implementing the age old serial port rate control issue on top of USB. I am guessing this is because G2 wants to always be ready to receive feedhold etc. control characters. But - the flow issue is then in turn solved by classic RTS/CTS signalling, which would prevent the PC from sending those control characters on a full buffer anyway, all while the dual endpoints take away the only reason I can see to bypass the rate control inherent to USB in the first place. Simply put I'm a bit confused as to why things work the way they do...?

Guess it's slightly off topic, but it does relate to trying other communication methods, as well as to calculating whether there is a danger of overloading communications given a certain g-code input.

Has anyone measured the typical receive rate capabilities of G2 during heavy load?

aldenhart, 2015-04-07:
I noticed that the latest Coolterm on the Mac (v1.4.5) does not handle files with very short lines very well. I have posted an earlier version (v1.4.0b5) that works well in the G2 / Resources directory. If you are testing with Coolterm (mac) I recommend using this one.

I have also written the author about this. Roger has been incredibly helpful in the past.

aldenhart, 2015-04-07:
Roger got back to me immediately. Here's the URL for previous Coo|term versions.
http://freeware.the-meiers.org/previous/

I'm going to see if we can locate the version where the change occurred and he's going to look into it.

giseburt, 2015-04-07:
@stg Re: flow control and serial port: The VCP in G2 relies on the inherent flow control of the USB protocols, so we needn't worry about that. We never throw away data unless we're explicitly in a condition that mandates it, such as an error condition. In that case we don't really "throw away" data either, but simply stop following commands and only respond to queries. We then have one "special query": `{clear:null}` that clears the alarm state.

This is to handle the case where the host-side has an unknown amount of data queued up in one channel, and an error occurs. The UI can simply send the `{clear:null}` down after the data and guarantee that none of the "unknown" data that was in the buffers gets accidentally executed.

As far as transmission rates goes, I haven't done any performance testing on the raw transmission rates. The transmission rates shouldn't be a concern, since the bottleneck really lies in how fast we can parse ,plan, and execute the commands. I have done performance testing on our parser, and it's more than fast enough for our current processors.

I have been working on our planner performance, and I believe you may have gotten caught in one of the interim states -- something I believe to be fixed (to the best of my testing ability) in edge-replan-replan. Any additional feedback on that branch is greatly appreciated.

stg, 2015-04-07:
@aldenhart Thank you, good info. I will use v1.4.0b5 for testing. Hope you had a good trip, by the way.

@giseburt Ok, sorry - I must have misunderstood what is happening when I send data without RTS/CTS flow control. It appeared to me that many commands were never executed, seemingly discarded due to what I guessed to be buffer overflows. I've just never seen that behaviour with an USB device before, even when they are very slow so I assumed there was a reliance on RTS/CTS instead of the regular USB flow control.

Is it allowed for the control application (UI) to ignore a de-asserted CTS and send certain commands (such as {clear:null}) anyway?

giseburt, 2015-04-07:
@stg RTC/CTS signals are completely ignored. They have to effect at all.

There shouldn't be a buffer overflow either. When there's no room in a buffer on G2 it simply stops reading from the USB. The USB host will continue to offer the same data to the peripheral until it gets read. There's no data loss involved.

We have seen cases in small linux boards (such as the Pi or BBB) where you could put too much data in the USB stack and it would lose data. But we're usually talking about many megabytes before you start to see that.

stg, 2015-04-07:
@giseburt Interesting. That means the intention is for it to work exactly the way I had hoped. I should do some debugging to figure out what then, is happening here on my end, as programs always behave incorrectly when sent using a terminal program - the main reason I switched back to using Chilipeppr.

Sorry for going on about RTS/CTS but I believed I was **told** that I had to enable this for proper communication when using a terminal program. However I can't find that - only the recommendation to try CoolTerm, so I'm thinking I must have confused some piece of TinyG information with G2... have had plenty of (way too) late nights trying to get my machines up to date with the modern world...

lllars, 2015-04-07:
I also noticed that very short moves had no acceleration/decelleration when I was working on backlash comp.  These moves were 0.01" (~0.25mm).  I noticed that all move segments were the same number of steps (~5 microsteps) in these very short moves.  Longer moves (like 2.5mm) clearly ramp up and down, with the first move segments having less than 1 microstep.

I was doing this testing using minicom under linux (so, not chilipeppr or coolterm).  These were G0 moves, which were simply typed into the serial console.

stg, 2015-04-08:
Mine are G1 moves of 0.25mm, which translates to 160 steps and a 90° turn of the motor.
I will try different communications solutions as well as edge-replan-replan later today and report back.

I've also noticed that speeding things up by increasing jerk on all involved axes increases the amount of incorrectly executed moves, while decreasing jerk far below acceptable levels removes the issue.

Have not thought about this fact much but I suppose it does indicates bottleneck issues with communication at some point in the control chain (or, perhaps less likely, a timing glitch/race condition).

Right now I am using a jerk of 500 for all axes, which solves the problem for most, but not all, programs. The machine however has been determined to be capable of a jerk of up to around 5k and I'd really like to reach stable operation at about half that value, reserving 5k for e-stop, homing only.

stg, 2015-04-08:
Problem is now confirmed with both CoolTerm (and others), as well as with edge-replan-replan.
I've also confirmed the problem to occur when the buffers are nowhere near starvation (all status reports in vicinity have qr<8).

giseburt, 2015-04-08:
I have confirmed this behavior. I'll have it isolated and fixed asap.

giseburt, 2015-04-08:
@stg Please test again with 083.04 from edge-replan-replan. In my testing it has fixed this oversight.

We've been so focused on extremely fast moves that we missed this slow-move issue.

This is why this is edge and not master, BTW. Edge code is always experimental.

stg, 2015-04-08:
I can also confirm that these moves are vastly reduced with edge-replan-replan, but as stated - they still persist.

giseburt, 2015-04-08:
@stg In 083.04?

stg, 2015-04-08:
Oh, wow. Sorry I was writing while you posted and missed that. I will try it right away.

stg, 2015-04-08:
Of course - I know that edge is experimental and I certainly expect issues to come up.
But you do wish to receive reports on encountered problems, no?
I always check the code for any reference to problems I encounter to see if they are to be expected and try to make sure it has not been discussed before or is currently being worked on.
If this is undesired or unhelpful at this point, I will cease immediately.

giseburt, 2015-04-08:
@stg I apologize. I appreciate the reports, don't hold back. :) Testing takes time, and so any help is greatly appreciated and accelerates the project! Thank you!

Mainly, I was writing to anyone else that was lurking the issues list to ensure that we're talking about an experimental branch. Stability of our code is very important to us, and we're very very picky about what goes into the master branch.

stg, 2015-04-08:
No need to apologize - I know very well what is involved :)

Since my own open projects are typically small I'm a sucker for feedback and try to give others my best in that regard which is sometimes welcome, other times not so much. That's the only reason I asked.

My biggest regret here is that I am not yet familiar enough with to code base to help out by actually fixing things instead of just pointing out problems.

Also, I can confirm that none of my current programs experience any issues with 83.04 and I am right now smiling from ear to ear, listening to completely glitch-free motion ;)

closed 2015-04-08 by stg