synthetos/g2 #73: ARC Problems

bronek999, 2015-04-08:
I wrote modification of TinyG2 - added SD card+LCD witch touch screen support.
My problem is with ARC G2,G3  commands. G code is about 10kB.

I tried to modify my G code (no arc use, code size +100kB) and everything works OK.

G command is readed from SD (not FROM USB) with this code

		res = f_open(&file, "domec1.nc", FA_OPEN_EXISTING | FA_READ);
		if (res == FR_OK)
		{
			while (f_gets(cmd, 100, &file) != NULL)
			{
				while(mp_get_planner_buffers_available() < PLANNER_BUFFER_HEADROOM)
					controller_run1();	//	one process
				DoCode(cmd);
				controller_run1();	//	one process
			}
			f_close (&file);
		}

		void DoCode(char *code)
		{
			cs.bufp = (char_t *)code;
			cs.linelen=strlen(code);
			dispatch_kernel();
		}

This is only change in original tinyg2 files (controller.cpp)

		void controller_run1()
		{
			_controller_HSM();
		}
		void dispatch_kernel()
		{
			_dispatch_kernel();
		}

BAD : 
[image]

GOOD :
[image]

stg, 2015-04-08:
Which version of G2 did you base your code on?
There have been issues causing very similar behavior in earlier versions.
These issues have already been solved in the later edge versions.

bronek999, 2015-04-09:
Replaced code with latest version. But problem is unsolved :(

stg, 2015-04-09:
There are several branches, "latest version" may refer to any of them.
Can you provide the branch and build number of the version you are running?
You can do this by issuing $fb to read the firmware build.

As an example, I am running a edge-replan-replan-83.04.

bronek999, 2015-04-09:
[fb]  firmware build             82.11
[fv]  firmware version            0.98
[cv]  configuration version       7.00
[hp]  hardware platform           3.00
[hv]  hardware version            0.00
[id]  TinyG ID       0213-2335-5383

downloaded zip file today.

aldenhart, 2015-04-10:
(P|ease pardon the beow. My *eyboard is dying and is missing | - for which I substitute a pipe, and cay for which I substitute *)

When running mutip|e arcs it is important not to read the next command from the SD card (or USB input) before the previous arc b|oc* is fu||y executed. The contro| read must be disab|ed or not read from the SD card. The arc ca||bac* returns STAT_EAGAIN unti| the arc is "done". It's a bit more comp|icated, as the arc sti|| |eaves ine segments in the p|anner, but once the arc ca||bac* returns STAT_O* it's safe to read and process the next arc.

bronek999, 2015-04-10:
Problem solved. Thanks to aldenhart

Rewrite code to
        while (f_gets(cmd, 100, &file) != NULL)
        {
            while(mp_get_planner_buffers_available() < PLANNER_BUFFER_HEADROOM)
                controller_run1();  //  one process
            DoCode(cmd);
            while(STAT_EAGAIN ==  controller_run1());  //  one process
        }
    stat_t  controller_run1()
    {
        return _controller_HSM();
    }

closed 2015-04-10 by bronek999