synthetos/g2 #74: assorted unit errors related to rotational axis

lllars, 2015-04-08:
1. In radius mode ($aam=3), position shows up as degrees when running '?' command.

2. In G20 mode, value entered in G28.3 line gets multiplied by 25.4

3. $ara value shows units as degrees.  It should be mm or inches.

4.  In G20 mode, value entered for $ara shows up divided by 25.4

5. In G20 mode, value entered for $ajm or $ajh shows up multiplied by 25.4

6. In G20 mode, velocity in status message (the one displayed every 250ms during moves) is divided by 25.4

7.  In G20 mode, feed rate entered in G1 command (ie. G1 F10 A360) is multiplied by 25.4

8.  Feed rate in G1 command is interpreted as degrees/min even if in radius mode.  It seems like it should be mm/min (or in/min), no?

lllars, 2015-04-09:
9.  $4tr shows up as mm/inches, should be deg/rev.  It's value gets divided by 25.4 in G20 mode.

aldenhart, 2016-02-23:
Still open. Labeled as bug.

giseburt, 2016-12-05:
Bumping this one, is it still open after the fixed for #123?

aldenhart, 2017-01-07:
Update - this has been put back on the active queue