synthetos/g2 #76: CCW arc goes wild

drsubs, 2015-04-10:
G03 X27.811731 Y4.565054 I0.01976 J-30.977549
this gcode makes my machine go into a large arc(CCW).

It was produced by CamBam

drsubs, 2015-04-10:
additional info:
[fb]  firmware build             82.10
[fv]  firmware version            0.98
[cv]  configuration version       7.00
[hp]  hardware platform           3.00
[hv]  hardware version            0.00
[id]  TinyG ID       0213-2325-8333
chillipebber does not show any thing wrong, but when the arc is finished the tool comes rushing in from off screen.

lllars, 2015-04-10:
That line of gcode should make a large radius arc.  See a description of G3 [here](http://linuxcnc.org/docs/html/gcode/gcode.html#sec:G2-G3-Arc).  The I and J terms are a vector from the current position to the arc's center.  So your radius is ~31 units (inches or mm?).

Does that help?  If not, please post more/all of your gcode file.

lllars, 2015-04-10:
I should clarify that what I wrote above is true for G91.1 mode (where the arc center is specified in relative units).  If you are in G90.1 mode, the arc center is specified in absolute units.

lllars, 2015-04-11:
just figured out that tinyg2 doesn't support G90.1 mode.  It is always in G91.1 mode.

There seems to be a bug wherein trying to specify G90.1 mode results in G90 mode.  G91.1 is similar, it results in G91 mode.

drsubs, 2015-04-12:
No this is an bug here is the g file:
https://dl.dropboxusercontent.com/u/2787926/hejtek.ngc
and a screen shot of chillipeppr and the milled result
the offending line is line #848
screen:
[image]
foto:
[image]

aldenhart, 2015-04-12:
I was able to reproduce the error you showed above. It seems the arc off the bottom-right stroke of the K wants to draw a complete 360 degree circle. It's interesting to note that the start and end points of line 848 are almost the same. I suspect that are being considered the same (erroneously), and this very small arc is being interpreted as a full circle. I am using the latest edge push - build 083.07. I'll track this down. Thanks for the detailed file to reproduce this.

drsubs, 2015-04-14:
Thnx, I´ll move on to 083.08

aldenhart, 2015-04-14:
Hi. I'm not quite ready to close this. I understand the root cause, but I don't yet have a solution out there. I'll close this once I've got a solution out to edge - hopefully in the next few days.

closed 2016-02-23 by aldenhart