synthetos/g2 #78: where is the schematic of the TinyG2

shortfordalong, 2015-04-15:
Hello
    I am the new. And I am a rookie. I download the TinyG and TinyG2. I get the schematic of the TinyG, but I didnot fint the schematic of the TinyG2. where can I get it.
    thank you in advance

shortfordalong, 2015-04-15:
Also I want to buy the development of the TinyG2, but I didnot fint the link.Can you give me the link.

teshko, 2015-04-15:
As far as I saw G2 should run on multiple platforms. Easiest way is to use Arduino Due with Gshield (https://github.com/synthetos/grblShield).

shortfordalong, 2015-04-16:
As I know the TinyG2 run in ATSAM3X8E, but the link you show me is  the board which is only drive the motor. Does the program run in ARDUINO DUE? the development board is like this 
http://arduino.cc/en/Main/ArduinoBoardDue
Looking forward to your reply

cmcgrath5035, 2015-04-23:
@shortfordalong
The tinyG2 Wiki at https://github.com/synthetos/g2/wiki describes the project and has  several items about the build environment. Platforms supported by the developers are the DUE + driver shields such as the gShield (3 stepper drivers) and a development platform called tinyGV9 which is not available yet for purchase. Other SAM3X platforms could probably be used, but you will be on your own to modify the build.

aldenhart, 2015-05-04:
The Arduino Due should be considered to be the reference platform for now. Along with the gShield for the drivers. Please refer to that documentation.

closed 2015-05-04 by aldenhart