synthetos/g2 #79: Comunication lost.

juanblanco, 2015-04-18:
Hi, I am having an odd problem:
when I plug the Due usb wire it opens a serial port (usually ttyACM0) that doesn't work at all. then I unplug and plug again the usb and then I have two serial interfaces (usually ACM1 and ACM2) only one of these works, some times none works and need to unplug the usb and plug again until it works.
then I can communicate to G2, run gcode and everything works ok, If i send a very large serial command or send a lot of gcode instruccitions at once the communication stop working and the Due doesn't respond, then I have to reset the Due and repeat the mentioned steps. I tested it in ubuntu 13.10, 14.04, on windows 7, and on windows 7 of a different computer. I test it with and without the gshield , the problem persist so I think it is not due noise from steppers or spindle.
I'm starting to believe that is some hardware problem of my Due, but I don't understant how a broken board can open a serial interface one time, two another time, and some how with trial and error I always got it  work.
Someone have experimented this problem before?? how many  ports should be open in normal conditions? 
I am reading the g2 code, is there an easy way to change form native usb to another serial for testing purposes?
thank you very much.

ryansturmer, 2015-04-21:
@juanblanco 

What version of the code are you currently running?

juanblanco, 2015-04-22:
Hi, I am using edge compiled a couple of weeks ago.

ryansturmer, 2015-04-22:
Can you query the firmware build (fb) - also, with the due plugged in, in either configuration, copy the output of `lsusb`

cmcgrath5035, 2015-04-23:
@juanblanco
Head over to https://github.com/synthetos/g2/wiki and search on USB, there are two items that discuss the Dual USB implementation. 
What are you using to send the serial commands and/or Gcode - CoolTerm, Chilipeppr, ?

giseburt, 2015-05-04:
@juanblanco Is your problem resolved? If not, do you have any updates?

lutorm, 2016-01-03:
I had a similar problem with a $15 "Chinarduino" from ebay. Whenever first plugged in, it would not respond, unplugging it and plugging it back in would make it start TinyG. My suspicion is that it got stuck in bootloader mode, but I'm not  sure. In any case, that board was quite sketchy, the headers were soldered all crooked and it was clearly not very high quality. I got a "Sainsmart due" for all of $25, and I'm not having that problem any more. This one always boots TinyG right away, I only have to reset it after uploading a new firmware.

ril3y, 2016-01-04:
@lutorm just to be clear your Due was sketchy not the gshield?  Right?

lutorm, 2016-01-04:
Correct. I have my own shield.

closed 2016-02-23 by aldenhart