synthetos/g2 #80: PWM frequency doesn't match input

jamesgao, 2015-04-18:
I'm trying to use a PWM pin to control an ESC for my homemade spindle. Unfortunately, it seems that the PWM system can't achieve the necessary 50hz frequency. Based on the code in SamTimers.h, it seems the lowest achievable frequency is 1600hz or so, given a prescaler of 1024. Is there a way to drop this frequency further?

giseburt, 2015-04-18:
Are you sure you want 50hz, and not 50*K*hz? What spindle are you using?

I doubled checked, and indeed at can handle 50hz. The formula is (when *not* using `DIVA` or `DIVB`, which is truly not uses):

```c++
// Definition of divisors:
// uint32_t divisors[11] = {1, 2, 4, 8, 16, 32, 64, 128, 256, 512, 1024};
int32_t top = (masterClock / divisors[divisor_index]) / frequency;
```

The top value simply sets how high it counts before overflow. The lower that number, the less resolution you have and the less accurate your actual frequency will be. The code will chose a prescaler of 32 for a frequency of 50hz. That leaves a top value of 52,500 which is a rather high resolution to set the duty cycle with. The duty cycle achieved will be within 1/1,000 of a hz of what you set in the gcode.

Short version: There's no technical reason for 50hz to not work.

What are the other values you're setting? The values that need configured for PWM to work correctly are documented [here](https://github.com/synthetos/TinyG/wiki/TinyG-Configuration-for-Firmware-Version-0.97#pwm-group-pulse-width-modulation), in the TinyG wiki.

Also keep in mind that `M3` is clockwise, `M4` is counter-clockwise, and `M5` is spindle stop. `M3` and `M4` need an `S` parameter that is between the "speed low" and "speed high" settings that you gave for the respective direction. For example, if you set `p1csl` (clockwise speed low) to 100, and `p1csh` (clockwise speed high) to 1000, then you must give an `S` value between 100 and 1000, such as:

```
M3 S250 ; turn on the spindle at 250 RPM
```

-Rob

jamesgao, 2015-04-18:
Thanks for the input! Sadly, I don't have a scope so I can't tell you exactly what's wrong, aside that I'm pretty sure something's off with the frequency.

The ESC I'm trying to control is a standard hobby motor ESC, which accepts a 5V "servo" control signal. Basically, it expects to see a .7 - 2ms pulse every 20 ms or so. In other words, you can approximate the control using 50hz pwm, duty cycle of 3.5 - 10%. You initialize the ESC by sending out the low end (3.5% duty cycle at 50hz), and it makes the motor beep an acknowledge code. From there on, 3.5% maps to 0 speed, 10% maps to max speed of the motor.

I have the $p1 commands configured correctly as far as I can tell. Based on a simple voltmeter, I can see p1pof is set correctly -- the Arduino Due I'm using outputs 3.3V, at 3.5% outputs about 0.12 V. Issuing M3 S16000 (the configured max speed), I see .33V on the control line. However, even when I set $p1frq to just 1 hz, I show a static display on my voltmeter. It seems as though $p1frq has no effect. Since the ESC is technically supposed to see a 5V signal, I also tried using a level shifter to get the 3.3V output to 5V, but this also had no effect.

If I initialize the ESC with another arduino mega running the default Servo.h from Arduino, everything runs correctly. I also tried disconnecting the ESC after initialization (this makes it emit some warning beeps indicating loss of controller input), then connecting it to Due pwm output. This speeds up the warning beeps, which I assume this is related to not seeing the right pwm frequency.

I know the real answer is to get a scope and figure out what's going on, so I'll have to hunt one down at some point...

aldenhart, 2015-04-19:
I know that g2 can drive an ESC / RC motor as a spindle, as this is exactly what one of our machine builders uses. The default settings for the spindle/pwm are set up for an ESC. Are you using those?

stg, 2015-04-20:
Not sure if this applies - but I have been unable to configure pwm frequency using $p1frq (but have not tried updating configuration in source) and with the default configuration I get 1kHz (measured with scope) despite it being configured for 100Hz in source, and $p1frq reporting 100 as well.

aldenhart, 2015-04-20:
OK. Let's mark this as a bug and take a closer look

cmcgrath5035, 2015-04-23:
I just built Edge 83.07 and loaded to my DUE.
It is a default gShield build.
$p1frq = 100 and $p1pof =0.1
My cheapo Hantek Oscope sez frequency =1.0kHz, Duty cycle=10%

cmcgrath5035, 2015-04-23:
Hah! So I rebuilt Edge 83.07 with the $p1frq setting (was 100) set to 5, assuming /10 error.
Nope! 
I do now have $p1frq=5, but the PWM frequency is still 1kHz
Happy hunting, Alden

By the way, I tried various 'M03 Svalues' , and PWM is changing appropriately

cmcgrath5035, 2015-04-30:
I just loaded build 83.07 on my V9k.
I have $p1frq=2500, PWM still running at 1KHz

aldenhart, 2015-05-04:
I pushed a fix for this as build 083.08 in edge. Please let me know what your results are.

cmcgrath5035, 2015-05-05:
I built edge 83.08 and downloaded to my tinyGV9k.
pwm freq now =2500Hz = $p1frq.

As far as I am concerned, this item ready to close.

It might be worth noting that changes to $p1frq require a reboot to become effective.
This implies that the desired value oof $p1frq must be in the compiled settings, as there is no persistence(yet) foor settings changes.
tinyG works this way, I assume G2 to be the same........
NOPE!! tinyG2 does NOT require a reboot for change in $p1frq.
I just tested by changing $p1frq to 1000 while in M5 (spindle off) state. On M3, measure 1000 Hz.

Nice fix!

jamesgao, 2015-05-05:
Sounds good! Unfortunately I dismantled my ESC setup, so I can't validate this personally. However, someone else did, so I'll close the issue!

closed 2015-05-05 by jamesgao