synthetos/g2 #81: Motate as included in TinyG2 vs the Motate Github repo?

kitling, 2015-04-19:
So, I've been interested in using Motate for a while. So, finding the repository once again, I start trying getting it to work with my device.

Eventually, I come over to look at this project, and I find the version of Motate bundled seems to be that of an older version, judging by the folder structure. This leaves me confused.

Seeing how my issue and PR at the Motate repo has been unanswered, and the G2 project seems much more active, I ask you: What gives? What's up? Should I be using the version from the repository, or the one from the TinyG2 project?

giseburt, 2015-04-19:
I apologize for the confusion. I answered this in the Motate repo synthetos/Motate#2 as well.

Short for for TinyG: We will be merging in the changes to Motate, which will happen after we have a tested and proven stable edge that is pushed to master.

Also note that G2 cannot support AVR Mega or Tiny, even though Motate does. The G2 stepper generation relies on multiple levels of interrupt supported by hardware, which means that the only AVR supported is XMega at this time. We also support ARM.

(I'm not saying it's impossible, just that it is not supported as part of the current architecture, and that we have no plans on attempting to make it work at this time.)

kitling, 2015-04-20:
Got it. Thanks!

closed 2015-04-20 by kitling