synthetos/g2 #84: malloc problem

bronek999, 2015-04-24:
Why I can't use malloc in this project?

I can use it, but I can't detect avilable heap size.
malloc(200000) return "valid" pointer. (Not NULL)

ryansturmer, 2015-04-24:
I believe we're veering away from dynamic memory allocation in this project - You'll have to allocate some space statically.

bronek999, 2015-04-24:
I have SD card + TFT + touch panel support for Arduino due version of TinyG2.
I need dynamic memory allocation !!!!

ryansturmer, 2015-04-24:
Why specifically? (Just curious, sounds interesting!)

bronek999, 2015-04-24:
I dont want PC near my machine
I wand small, stable,fast and secure system.
Booting windows to mill 1 packet is no my way........

There is no 100% working GUI for tiny.......java based application isn't way :(

ryansturmer, 2015-04-24:
no I mean, what about your SD card + TFT setup requires malloc?

bronek999, 2015-04-24:
> Komu: "synthetos/g2" <g2@noreply.github.com>
> Kópia: "bronek999" <bronek@amega.sk>
> Odoslané: 24.04.2015 16:06:57
> Predmet: Re: [g2] malloc problem (#84)

SD card?  for SORTED file list? files on FAT32 is not sorted by 
filename?

TFT+touch? Graphics buttons, labels,.........

S pozdravom

Branislav Drengubiak
AMEGA s.r.o.
Liptovský Mikuláš

------ Pôvodná správa ------

giseburt, 2015-04-24:
Hello @bronek999,

TinyG (and G2) are real-time extremely-low-latency motion control systems. We designed the system from the bare-metal up to be highly efficient and fast, while easy to communicate with both as a human or machine-to-machine.

There's not really enough processor cycles to be able to guarantee the timing we need of the motion control while still handling graphics and other UI tasks.

We recommend using a small Single-Board-Computer (SBC) to drive the TinyG or G2. You can use a Beagle Bone, Raspberry Pi, or Intel Edison. (The Edison is really small, physically, if that's a concern.) You can then either put a display and input device on the SBC, or drive it wirelessly (over WiFi) with a phone or tablet.

For the latter approach, check out [Chilipeppr](http://chilipeppr.com/tinyg) -- you load a small program on the board to link the web to the tinyg (over USB) and then you control it all from a web browser.

--

About malloc specifically: The malloc that's in this project (and used on the ARM that's on the Due) is that provided by [newlib-nano](https://github.com/32bitmicro/newlib-nano-1.0). See the [documentation](https://github.com/32bitmicro/newlib-nano-1.0/blob/c010b5911834ed9a412bd0a865abdf3eed00a4ee/newlib/libc/stdlib/malloc.c) there on how to fully utilize it. All of the linker script utilities that are necessary should be accounted for. (I believe the implementation is [here](https://github.com/32bitmicro/newlib-nano-1.0/blob/c010b5911834ed9a412bd0a865abdf3eed00a4ee/newlib/libc/stdlib/mallocr.c#L192-L272) but I might have the wrong one.)

-Rob

bronek999, 2015-04-27:
my code for memory allocation is

#include <stdlib.h>
void* p_mem;
p_mem = malloc(200000);

p_mem is set to  0x200...............

I think there is stdlib in TinyG2 by default
tinyg2.h uses #include <stdlib.h>

How can I change it to newlib?

giseburt, 2015-04-27:
It's already set to newlib (aka newlib-nano, they've merged) by the makefile.

Why are you attempting to allocate ~195K? I don't believe the Due has that much ram.

bronek999, 2015-04-27:
Yes.. it have 96kB RAM.... malloc(200000) should return NULL pointer.
but it return "valid" pointer

giseburt, 2015-04-29:
I'm sorry. As far as I can tell that's the correct behavior of malloc in an newlib. If you can find a way to make that work we'll happily accept a pull request.

Thank you!

bronek999, 2015-06-12:
First milling

[url]www.amega.sk[/url]

closed 2016-02-23 by aldenhart

hqkhuong, 2018-03-06:
@bronek999 , could you please check your email. thanks