synthetos/g2 #86: Boot Loader missing after Atmel ICE based reinstall of G2 on tinyGv9k?

cmcgrath5035, 2015-04-29:
Back in January, I bricked my tinyGv9k due to a bad compile (Issue #47) then went a few rounds with Atmel 6.2 and Atmel ICE (Issue #53) but finally won and now have build 71.04 installed.

Today I decided to complete my G2 upgrade to build 83.07(already running on my DUE) but find that my tinyGv9k will not respond to the bossac load process from https://github.com/synthetos/g2/wiki/Flashing-G2-with-Linux.

I'll assume for the moment I could get the ICE out and load 83.07with it.

Is there a different build and/or load process to restore a bricked tinyGv9k that will restore the boot loader?

giseburt, 2015-04-29:
The loader is baked into the RO on the processor, so it cannot be removed. I suppose it's possible that you fried the chip...

There's an Erase button (and a similar jumper on the v9 prototypes) and you must hold the Erase button (short that jumper) and hit reset to erase the flash, which forces the board to use the loader.

Once in this state, you should see a single serial port and the USB listing (lsusb on linux, or System Report on OS X) will show an Atmel CDC device attached. That's the loader and `bossac` should now work.

**Note:** I have found that, after the _first_ load after using the Erase pin you have to completely power the board down and back up in order to get to the loaded firmware. Otherwise it will continue to reset into the loader. After that, normal methods of getting to the loader will not require a power cycle.

cmcgrath5035, 2015-04-29:
OK, thanks Rob. So the reset performed by the ICE didn't have that effect? 
I don't think I fried anything, this was the issue which you thought was compiler, may have been Makefile, now fixed.

I'll try the erase jumper (I see it) and give it a go.

cmcgrath5035, 2015-04-30:
Good News: I have now mastered the "short the jumper then reset" to get a bossac compatible interface.

Other News: the bossac download of my 83.07 build (PLATFORM=G2v9i) processes without error, a power cycle (per your comment) brings tinyG2 up (the heartbeat led). But attempts to connect via SPJS or CoolTerm result in the heartbeat stopping and no response. TinyG2 does restart the heartbeat on reset button.

Please clarify: I now forget what I did last Dec when I first started experimenting with my V9k. I know I built what was then Edge (71.04) and loaded it with bossac, using the 1200 baud technique. But I have no idea what was running on the V9k at the time. 
So, question: Does tinyG2 running on the V9k platform support a new download procedure by setting tty to 1200baud?
I am beginning to suspect the answer is no, the erase step is required first.
I am also beginning to think I didn't really need my Atmel ICE to recover from the bad build in Feb, rather just this erase procedure.

cmcgrath5035, 2015-04-30:
Updated News: So after summarizing last nights adventure above, I powered up my V9k this AM and find that TODAY both CoolTerm and CP/SPJS1.80 will talk to my V9k/83.07 with a little nudging. First, I checked and checked again that the serial port was set at baud=115200, not 1200(I had done that night too).
My only answer is that it took 12 hours for the bits to harden in the flash, and I'm sticking to it.

I ran CoolTerm first, then knowing that it should work fought a bit with CP via SPJS 1.80. It did not want to start at first, but after a tinyG reset (always fun on linux due to port walking ACM0-->ACM1, etc) I managed to get it going. 
I'll now download JohnLs SPJS 1.8.1 to see if that works a bit better.

On both CoolTerm and SPJS, the first interaction between server and host(V9k) results in a "message(or command) not recognized", so there is some initialization garbage somewhere.

giseburt, 2015-05-04:
@cmcgrath5035 What's the latest on this one? Is the problem resolved?

cmcgrath5035, 2015-05-04:
I have been running my tinyGV9k on build 83.07 for a few days, successfully connecting to both CoolTerm and CP/SPJS1.80.

Shortly, I will be building 83.08 to test out Alden's PWM Freq fix.
Will report on how loading goes.

cmcgrath5035, 2015-05-05:
OK, built Edge just now, builds $fb=83.08
I find that with $p1frq=2500, I measure 2500Hz, Alden's fix works.

I have to conclude that https://github.com/synthetos/g2/wiki/Flashing-G2-with-Linux does not apply to the tinyGV9k when a version of tinyG2 is running. To enable bossac for an update from 83.07 to 83.08, I had to do the "screwdriver across the erase solder bumps and reset". After that, the LEDS onboard change, G2 stops heartbeating and lsusb reports an Atlmel device.
Bossac then runs without the 1200 baud dance. Download was fast, and after power cycle G2 heartbeat begins.

Interestingly, CP/SPJS1.80 connected first time.

So pwmfreq is now correct and I am back in business.

closed 2015-05-05 by cmcgrath5035