synthetos/g2 #87: Motor goes to sleep despite MOTOR_ALWAYS_POWERED

jamesgao, 2015-04-30:
I found that on a long pocketing operation with no Z movement, the z-axis goes to sleep. I initially tried setting MOTOR_ALWAYS_POWERED in settings.h. Chilipeppr reflects the change, and the timeout duration is longer. However, the axis still goes to sleep after a while, dropping the tool into the workpiece.

I originally thought perhaps the Z-axis stepper is overheating, so I tried lowering the current. I also tried replacing the stepper driver. Neither has any effect.

cmcgrath5035, 2015-04-30:
I happen to be playing with my V9k/build 83.07
With $1pm=1, X axis ($1) powers and times out (based on $mt)as Y is moved.
Otherwise $1 off, so $1pm=1 not working, I concur.

James - Does setting all to $_pm=3 fix your issue as work around (3 == if any axis in moving, power on)

lutorm, 2015-11-23:
I'm also seeing weirdness with the motor enable behavior. All my motors are set to 2 (on in cycle), but if I make a manual single-axis movement, only that motor is enabled. This is with the current edge.

lutorm, 2015-12-06:
I made a pull request (https://github.com/synthetos/g2/pull/108) with the changes I had to make to stepper.cpp to get not-moving motors with MOTOR_POWERED_IN_CYCLE to be powered on.

giseburt, 2016-12-05:
@lutorm, I'm not sure why this is still open. Moving it to In Progress, but I believe this was fixed a while back.

closed 2017-01-04 by aldenhart