synthetos/g2 #88: g2 doesn't upload code into Arduino DUE

xwalter, 2015-05-02:
Hi to everybody and thanks for your job ! 
Just a problem ....I have downloaded the g2 pack and unzipped .
After that I have opened it with Atmel Studio + Visual Micro that I normally use for my project .
In your g2 main folder I found  g2 solution for atmel studio 6.2 and I click over it 
The Atmel studio open the solution and I can see all of the files, folders and else
I clean and then build the solution -> no errors , 1 warning  ..I think it's ok ....
I try to upload the code into my arduino due connetted , I select arduino due board programming port and the correct serial port which I normally use .But when I press to download the code into DUE I have a message box " Please select a conncted tool and interface and try again " 
What does it means , I have to buy and install a Atmel ICE or something like that ? 
Does the g2 project do not work as other project " 
Thank you for the answer 
Walter

xwalter, 2015-05-03:
I have  Atmel Studio 6.2.1563  Service Pack 2  +  Visual Micro  + Arduino Due board 
But doesn't download Tinyg2 
I dont' know ....:(

xwalter, 2015-05-03:
I have understood ...I have an Arduino Due board but not tinyg2 board and I have windows 
Maybe those are the issue

xwalter, 2015-05-04:
nobody here .... ?

stg, 2015-05-04:
I am on Windows 7 and using Arduino Due with custom shield.
Was a bit tricky to get the board programmed but for me the process is:

1) Connect to __programming port__
2) Press and hold reset for 1 sec
3) Wait 10 sec
4) Disconnect
5) Prepare bossac command so you just need to hit enter to execute
6) Connect to __programming port__
7) Immediately hit enter as the device is discovered
8) Repeat from 4 until you get the timing right

stg, 2015-05-04:
For reference, this is the method I use https://github.com/synthetos/g2/wiki/Flashing-G2-with-Windows

I don't like the modern huge IDEs so I never installed Atmel Studio.
It compiles just fine with msys (mingw) and gnu tools for arm embedded (launchpad) - provided you build a dummy tput binary, as this is used by the Makefile and not available on windows.

xwalter, 2015-05-04:
good morning stg 
Well  I have downloaded the project and I mean It's written by using Atmel Studio because inside the folder there are two files : one is the project and the other one the solution 
This is because I use Atmel Studio ...to study the software and maybe to do some modifications .
I would like to understand the motion algorithms and else , but anyway Im not able to download inside the Arduino Due board ....

xwalter, 2015-05-04:
pratically this project get the max of performances of arduino due as a arm cortex processor and the arduino ide functions are jumped out .
The arduino due board is like a 32 bit arm cortex generic board and to flash it need to have the Atmel ICE .Otherwise it is possible to flash it by using arduino ide bossac ect ect 
What is important to understand is that this is not normal arduino ide work flow .

stg, 2015-05-04:
Yes, I am aware of that.

Still, it is quite possible to flash using bossac, I do this routinely for every new version of g2 that I am interested in. Atmel ICE is not necessary unless you have bricked or reprogrammed the ATMEGA16U2, which is there to allow programming of the SAM3X8E without a dedicated programmer.

stg, 2015-05-04:
As Atmel Studio + Atmel ICE is not the method I am using, I am sorry, but I can not help you with this setup. Using the method above, you could at least get your code running until you find a way to make your preferred method work.

xwalter, 2015-05-04:
ok thanxs for your informations , now I have understood ....
I think before to work and to modify the project I have to find the flashing solution and then work on it

giseburt, 2015-05-04:
Hi @xwalter,

I suspect the problem may be that you don't have any "drivers" installed for the Due or TinyG2. If you install the *.inf file found at the [top level of the repo](https://github.com/synthetos/g2/blob/edge/TinyGv2.inf) then it should then be recognized by Bossac.

I'm sorry that we don't have good instructions on how to install that file. Please also be aware that it won't work on Windows 8 or newer without extra work since it's not a signed driver.

-Rob

xwalter, 2015-05-04:
Hi giseburt , thanxzs for your answer 
So , I', using windows 7 pro . I have all of the drivers installed for arduino uno , mega, due 
Normally I use this configuration to develop and download the code 
-Windows 7 pro 
-Atmel Studio 6.2.1563
-Visual Micro 
-Arduino IDE 1.5.7
I ' have modify my bossac.exe to allow the programming of Arduino Due embedded in UDOO board 
Maybe is this the issue ?

giseburt, 2015-05-04:
I'm not sure how you modified the bossac, but that's certainly a possibility. I don't have an UDOO to test against, so I can't really test it either.

I'll explain the procedure for a real Due, and you may have to translate it to what's on the UDOO:

  1. Hold the ERASE button, then push and release the RESET button, then release the ERASE button
  1. In the system's list of USB devices, you should then see something with a vendor of "Atmel" -- that's the SAM-BA "bootloader". If the driver you have attaches to that device, you should be able to see a serial COM port, and bossac should be able to locate and load the firmware on that device.
  1. After bossac is done, it will either reset back into the loader and show an Atmel device, or it will restart into the G2 firmware and show a Synthetos or "OpenMoko" device. (We are using a free OpenMoko vendor/device ID, so it sometimes shows up in lists.)
  1. If it shows an Atmel device, you will likely need to completely remove power from the Due and then power it back up for the firmware to "take". If that doesn't work, make sure that the `-b` flag is on your `bossac.exe` command line, as that's what tells the Due to reset into the Flash and not back into the SAM-BA bootloader.

xwalter, 2015-05-04:
I mentioned udoo just to inform tha I can program arduino due embedded and arduino due single board ....ok...just a question 
After this operation , the boot loader installed into arduino due will be erased?

giseburt, 2015-05-04:
The SAM-BA loader in the Due is in the ROM, and cannot be removed or changed (to my knowledge). It's not really a bootloader in the sense that it goes into the bootloader then will timeout into the normal program. If you're in the loader, the firmware has been erased and the chip is effectively blank.

xwalter, 2015-05-04:
ok sir thanx

xwalter, 2015-05-09:
with my new configuration I'm able to work with Atmel Studio 6.2 with Arduino platform or ASF 
I created a new external tool based on bossac.exe and a simple windows batch file 
If I open an ASF project based on Board Arduino Due I can download the project without any problem 
If I open and I create a new projet with sketch files I can download the pprojet into Arduino UNO , DUE ect . 
If I want ,instead , to flash the Arduino Due processor embedded into UDOO  I can do without any problems .I can do anything but not flash tinyG2 ......I had also fork and create tinyG2 project into my repository , in my desktop pc . I open the solution and I try to built it 
Some errors occours 

Error	1	'struct Motate::InputPin<106>' has no member named 'setInterrupts'	D:\myDep\g2\TinyG2\gpio.cpp	120	17	TinyG2

Error	2	'struct Motate::InputPin<107>' has no member named 'setInterrupts'	D:\myDep\g2\TinyG2\gpio.cpp	121	17	TinyG2

Error	3	specialization of 'static void Motate::IRQPin<pinNum>::interrupt() [with signed char pinNum = 106]' after instantiation	D:\myDep\g2\TinyG2\motate\utility\SamPins.h	262	55	TinyG2

Message	4	in expansion of macro 'MOTATE_PIN_INTERRUPT'	D:\myDep\g2\TinyG2\gpio.cpp	183	1	TinyG2

Error	5	specialization of 'static void Motate::IRQPin<pinNum>::interrupt() [with signed char pinNum = 107]' after instantiation	D:\myDep\g2\TinyG2\motate\utility\SamPins.h	262	55	TinyG2

Message	6	in expansion of macro 'MOTATE_PIN_INTERRUPT'	D:\myDep\g2\TinyG2\gpio.cpp	184	1	TinyG2

I want to try to create a new ASF project and copy the tiyG2 files , but I don't know what will happen .... Your project is difficult to understand maybe and too complex ....
If you want to program arduino due with ateml studio you have two chances 
1) Use visual micro with arduino ide platform 
2) Use ASF and create e new virtual programmer such as external tool 
Your system is different but is difficult to understand 
Do you have any idea to how complie it without setting file o make or else , but just with simple environments tools supllied by Atmel Studio 6.2 
Why do yo have to use an external programmer like Atmel ICE ?

giseburt, 2015-05-09:
Hi @xwalter,

I'm rather confused about your problems.

I guess we need to get to the meat of it: What are you trying to accomplish?

  * Just trying to flash a Due (the one on the UDOO) with G2 from Windows?
    * You want these instructions, no Atmel Studio or Atmel ICE required:
      [Flashing G2 with Windows](https://github.com/synthetos/g2/wiki/Flashing-G2-with-Windows)
    * If you have an Atmel ICE, you can use these instructions:
      [Flashing G2 with Atmel Studio6 and Atmel ICE](https://github.com/synthetos/g2/wiki/Flashing-G2-with-Atmel-Studio6-and-Atmel-ICE)
  * You want to compile and then flash G2 from Windows?
    * You want these instructions: [Compiling G2 on Windows (Atmel Studio 6.2)](https://github.com/synthetos/g2/wiki/Compiling-G2-on-Windows-(Atmel-Studio-6.2))
      * At the end it explains how to use an Atmel ICE to flash (and debug).

You shouldn't need to create a new project. We have an Atmel Studio project (and "solution") already in the repo. Open `<project directory>/TinyG2/TinyG2.atsln` as the instructions linked to above describe.

You also don't want to have ASF in the project, as that will likely conflict with the Motate functions that provide our hardware interaction.

-Rob

PS: I'm assuming when you say "visual micro" you mean Atmel Studio. **If you do mean another IDE,** then we can't really support another IDE, but it shouldn't be hard to make any IDE that supports a Makefile-based project (like this one) work. I would highly recommend getting one of the solutions we offer and support to work first, before starting to tread a new path.

xwalter, 2015-05-09:
good evening giseburt .
I would like to compile the tinyG2 with Atmel Studio 6.2 , in windows .I saw and follow all of your tutorial but when I compile the project some errors occours .
Visual Micro is installed as a Atmel Studio plug -in ;in this case I can write sketches for arduino and download into the boards .But I can also to program Arduino DUE without visual micro , just using ASF (atmel software framework ) and this is case is like to program Arduino Due board as a normal Atmel development board .For these operation I never use any programmer , no Atmel ICE , no SAM ICE , no SAM BA .I use just arduino avrdude.exe or bossac.exe (for sam) 
What I would like to do is to open tinyG2 with Atmel Studio 6.2 as a normal ASF project , to study it and if I want to download the project into my Arduino Due board , by using external tool that I have already done .
Pratically I want to know if your project  needs to have Atmel ICE or something like that because I read in somewhere the Atmel ICE in your tutorial .
My philosofy in Arduino is not to have and not to buy any programmer and not to spend money for them , otherwise I come back to Microchip dsPic or Pic 
I choose to move to arduino because I don't want to spend money for programmer and to have problems with operating system and usb driver at any time 
Then I have many arduino boards and I program them with atmel studio + visul micro but for critical application I prefer to program them as board , for example by using asf or to program them without arduino ide platform 
The UDOO is an other question , just to say that I have configured my system to program also Arduino due embedded into it .
At the moment I'm interesting to understand your system and work with it with this  
- Windows 7  pro 
- Atmel Studio 6.2.1548
- Arduino IDE  1.5.7
- No Atmel ICE 
- No SAM BA 
- No phisical external programmer 
Do you think is this possible to do ? 
I think also the issue is makefile , gitattribute and else
I want to open the tinyG2 solution without any option , just code and libraries , I don't want to complicate my life with github , makefile ....

xwalter, 2015-05-09:
Try to image this .....
I create a new blank solution in Atmel Studio 6.2 and I import tinyG2 project .Then I check out the use of external makefile , pratically NO external makefile .
I build the project and I have an error : 
Error	1	MotatePins.h: No such file or directory	C:\Users\karol\Documents\Atmel Studio\6.2\g2-master\motate\utility\SamSPI.h	36	24	TinyG2
But I see into the utility folder and SamSPI is present ....I don't know what to do ...

xwalter, 2015-05-09:
Now try to image this situation .....I follow your tutorial , I fork , I clone the project in my repository , and then I open the project by clicking on tinyg2 solution 
Without touch anything I try to compile and I have this errors 

------ Build started: Project: TinyG2, Configuration: G2v9d ARM ------
Build started.
Project "TinyG2.cppproj" (default targets):
Target "PreBuildEvent" skipped, due to false condition; ('$(PreBuildEvent)'!='') was evaluated as (''!='').
Target "CoreBuild" in file "C:\Program Files (x86)\Atmel\Atmel Studio 6.2\Vs\Compiler.targets" from project "D:\myDep\g2\TinyG2\TinyG2.cppproj" (target "Build" depends on it):
	Task "RunCompilerTask"
		Shell Utils Path C:\Program Files (x86)\Atmel\Atmel Studio 6.2\shellUtils
		C:\Program Files (x86)\Atmel\Atmel Studio 6.2\shellUtils\make.exe all 
		make: *** No rule to make target `arduino/cortex_handlers.o', needed by `TinyG2.elf'.  Stop.
	Done executing task "RunCompilerTask" -- FAILED.
Done building target "CoreBuild" in project "TinyG2.cppproj" -- FAILED.
Done building project "TinyG2.cppproj" -- FAILED.

Build FAILED.
========== Build: 0 succeeded or up-to-date, 1 failed, 0 skipped ==========

Pratically I wnat to know how to compile this project , with makefile or without ? I have to change something into makefile ? 
I have already change the device in sam3x8E  but nothing

giseburt, 2015-05-09:
No external programmer is needed. The instructions in [Step 3a](https://github.com/synthetos/g2/wiki/Flashing-G2-with-Windows#step-3a---program-tinyg2-onto-the-due-using-arduino-tools) of [Flashing G2 with Windows](https://github.com/synthetos/g2/wiki/Flashing-G2-with-Windows) explain how to use `bossac` to load the firmware onto the Due without an external programmer.

BTW: We explain how to obtain the `bossac` from the Arduino project simply because it has the added support for the SAM3X8E that's on the Due. The `bossac` on the BOSSA web site and github project (last time I checked) did not.

We support using the Atmel Studio and XCode GUIs to control the compilation, but they both eventually use the Makefile. We only support compilation via Makefile at this time. We provide and maintain Atmel Studio and XCode projects to ease adoption, but we unfortunately can't spend a lot of time helping with other UIs or methods of compiling.

-Rob

xwalter, 2015-05-09:
Ok Rob , but  can I compile thi s project or just download the firmware into arduino due by using bossac as you explain?  
In the properties of project , tab build there is a combo cob to select the configuration ; what configuration I have to select to compile for arduino due without g2shield? 
I can see here ....
-active(G2v9d)
- debug 
-gshield 
-g2v9f
-shopbot Shield
- sbv300

giseburt, 2015-05-09:
@xwalter, that's trying to compile code we haven't had in the project for a long time!

The good news is, this is easy to fix. 

Sine you made a fork (which is only necessary if you're making changes), you need to update your fork with the latest changes from our fork. (The original instructions are [here](https://help.github.com/articles/merging-an-upstream-repository-into-your-fork/). I've modified them to make it easier to do again later by making our fork the `upstream` remote.)

(For an easier alternative if you aren't planning on making changes, just clone our fork directly.)

1. Open a shell (OS X) or github shell (installed with the [GitHub App for Windows](https://windows.github.com/))
1. Change directories to your project repo
1. This command you will need to run **once** (per cloned out repo, which is usually just once):

    ```bash
    git remote add upstream https://github.com/synthetos/g2.git
    ```

1. Then, each time you want to update your repo with code changes from our repo:

    ```bash
    git fetch upstream master edge
    # Replace <branch> with either edge or master.
    git checkout <branch>
    git merge upstream/<branch>
    ```

1. To push changes to your fork, you can:

    ```bash
    git push origin <branch>
   ```

  1. From there you can issue pull requests.

Notes: The `edge` branch is the latest development branch, and may have instabilities. The `master` branch is older code.

Your repo likely doesn't have the `edge` branch, if it's as old as I think it is. To use the edge branch the first time, you'll want to:

  ```bash
  git checkout -b edge upstream/edge
  ```

Once you have a new `master` or `edge` checked out, you should be able to compile. (You may have to quit and restart Atmel Studio if it was open during all that git work.)

-Rob

giseburt, 2015-05-09:
@xwalter To answer your question "what configuration I have to select to compile for arduino due without g2shield":

If you don't wish to use an actual gShield, but instead make your own, you can either use the same pinout as the gShield, or you can create your own target. Instructions are [here](https://github.com/synthetos/g2/wiki/Adding-a-new-G2-board-(or-revision)-to-G2).

xwalter, 2015-05-09:
Rob ok thanks for your informations .
But I would like to have just a project and open it with Ateml Studio 6.2 and compile it 
Must I have to do all of this sptes you mentioned above ?    
Just a question ..... I create a new solution with atmel 6.2 and namedit as tinyG2_new .
Do you think can I copy the files of your application into the new project and compile them ? 
I'm trying to do this but I have problem with motate and what is inside motate folder .
If I create a new project ASF , do you think it is possible to copy and paste those files and compile them ?  If the result will be positive I will publish this new solution for people that use windows and don't want to open the console and type command .....?
Pratically is like to create a new project which is based on tinyg2 because the algorith and the operations are coming from tinyg2 becuase I like tinyg2 project but in this way I'm loosing a lot of time for logistic problems such as command line , makefile , repository and else , I'm not expert in github yet

aldenhart, 2015-05-09:
@xwalter - I use the a Windows7 based AtmelStudio6.2 solution in a virtual machine all the time. I use and Atmel-ICE (about $50) and do not have to use the command line for any of the edit, compile, load, run, debug steps. (Although I do at times, it is not necessary). The ICE also supports real time debugging and hardware breakpoints, which is very, very useful. I do not use Atmel ASF as it just complicates things and we do not support it.

If you follow the instructions on our wiki you should be able to do the same. This page: 
https://github.com/synthetos/g2/wiki/Compiling-G2-on-Windows-(Atmel-Studio-6.2)
...should have everything you need. 

If you want to program without Atmel ICE (about $50) the you should not need the following page: https://github.com/synthetos/g2/wiki/Flashing-G2-with-Windows. Command line operations are required.

If you figure out how to integrate one of the non-ICE programming methods outlined on that page into Atmel Studio please do, and let us know how you did it, if you would. It would be much appreciated.

Thanks for your interest.

xwalter, 2015-05-10:
thanxs for your answer aldenhart .
I'm watching your links .
I like your project so because I would like to understand well

xwalter, 2015-05-10:
aldenhart I have already senn this page  
https://github.com/synthetos/g2/wiki/Compiling-G2-on-Windows-(Atmel-Studio-6.2)
but nothing  ... :(

xwalter, 2015-05-10:
Nobody gonna take my car 
I'm gonna race it to the ground
Nobody gonna beat my car 
It's gonna break the speed of sound
Oooh it's a killing machine

closed 2016-02-23 by aldenhart