synthetos/g2 #90: Unable to compile TinyG2 in Atmel studio

Jaytronics, 2015-05-09:
Hi,
Trying to compile G2 in Atmel studio and always get "make: *** No rule to make target `C:/Users/TIM', needed by `bin/gShield/gShield.elf'.  Stop." Does anyone have any ideas ??

giseburt, 2015-05-09:
Hi @Jaytronics,

What would the rest of that path be? Perhaps something like `C:/Users/TIM'S COMPUTER/TinyG2`?

You may run into trouble compiling (much like most any other compiled project) if you have any non-alphanumeric characters anywhere int he path to the project. This applies to all platforms, not just windows. No matter how hard we try to make it work, we don't write all of the tools we use, and so there are still cases where there are problems.

Safe characters are: `a` through `z`, `A` through `Z`, `-` (dash), `_` (underscore), `.` (period), and `0` (zero) through `9`. This applies to the *full path from the root of the drive to the TinyG project*, so if your user's *home directory* is `ROB's WinNT Box!` (:smiley:), then you will likely see problems no matter where in there your project resides.

Again, this applies to Windows, Linux, and Mac.

-Rob

Jaytronics, 2015-05-09:
Hi Rob,

Thanks for that… Hadn’t thought of that .. I will sort!

Many thanks

Jay

 

No virus found in this message.
Checked by AVG - www.avg.com
Version: 2015.0.5863 / Virus Database: 4328/9526 - Release Date: 04/13/15
Internal Virus Database is out of date.

Jaytronics, 2015-05-10:
Hi Rob,

Really sorry but just another quick one that is confusing me… I am compiling without error but only the elf file is created but not the .bin ?? Any idea would be gratefully received … cheers

giseburt, 2015-05-20:
Sorry @Jaytronics, I missed this. I'm not sure why only the elf would be created.

The build results are under `./build/<target>/` where there should be `<target>.elf`, `<target>.bin`, and`<target>.map`, where `<target>` is the name of the board you're compiling for. By default this will be `gShield`, which will work for most Due targets.

Perhaps you're seeing the `TinyG2.elf` file that is put in the primary directory. That should only be done for AtmelStudio, and without it AS thinks the job failed. It's otherwise problematic, since it's both confusing as well as indeterminate as to which one you'll end up with when building for multiple boards simultaneously.

-Rob

Jaytronics, 2015-05-20:
I did manage to compile to bin but had to edit the makefile as the one on 
the edge is a bit corrupt.. cheers

Sent with AquaMail for Android
http://www.aqua-mail.com

Jaytronics, 2015-05-20:
From memory the objcopy lines are wrong.. hope that helps... cheers

Sent with AquaMail for Android
http://www.aqua-mail.com

giseburt, 2015-05-20:
Could you provide some more detail when you get a chance? I use the makefile every day and don't have any issues, so I'd like to know what's unique to your setup that's causing issues.

Thank you,
  -Rob

kendallgreen, 2015-07-31:
Trying to make TinyG2 for gShield on UDOO dual.  Makefile ends with error "./gcc cannot execute binary file".  I tried executing the binary ./gcc directly and got the same error.   It is located in home/ubuntu/Downloads/g2-edge/Tools/gcc-arm-none-eabi/arm-none-eabi/bin

nafets77, 2016-02-03:
Hi,

Old tread, but not closed so I jump in with the same issue regarding output file. I only get a .elf file in the bin folder after compiling, no .bin file. Any solution to this?

Regards
Stefan

pabloxid, 2016-03-05:
Same behavior here. TynyG2, Windows, Atmel Studio 6.2, DUE, then --> only .elf, no .bin file.

yck011522, 2016-03-05:
Go find this exe called arm-none-eabi-objcopy.exe (I don't remember where I got it)
run the following command from CMD console: (Change the directories for your file location)

C:\Temp\arm-none-eabi-objcopy -O binary C:\g2\TinyG2\TinyG2.elf C:\Temp\TinyG2.bin
C:\Temp\bossac.exe --port=COM6 -U false -e -w -v -b C:\Temp\TinyG2.bin -R

pabloxid, 2016-03-06:
I'm trying to figure out what's wrong with current makefile.

aldenhart, 2016-09-08:
If you are still interested, please take a look at the edge-100 branch and makefile structure.

closed 2016-09-08 by aldenhart