synthetos/g2 #92: Too short pulses

bronek999, 2015-06-01:
I tried to use my own version of TinyG2 with SD card+Touch panel + TFT LCD module support. Everything is working correctly, but STEP pulses are too short for my stepper drivers. I'm using LeadShine 3DM683 drivers, but lot of drivers require minimal 2.5uS pulse. 
Whene I compile and run unmodified version, this problem is same.

[image]

Modifying line gives longer pulses, but it is still short.
//	dda_timer.setDutyCycleA(1.0 - 0.75);		// This is a 75% duty cycle on the ON step part
to
	dda_timer.setDutyCycleA(1.0 - 0.80);		// This is a 75% duty cycle on the ON step part

stg, 2015-06-01:
Had the same problem. I solved this with a modification to stepper.cpp to emit a continous 50/50 pulse train instead of pulses by changing "set" to "toggle" and removing "clear". This requires one to double the microstep parameter. This has been working very well for me, but in theory this could still generate pulses that are too short right at direction changes. However, even the most demanding jobs I've done shows no indication of this occuring.

aldenhart, 2015-06-01:
Would you mind pointing me to the branch with these changes, or better yet providing a pull request?

stg, 2015-06-02:
Just did a pull request, but realized you were most likely referring to broneks changes, not mine.
At least I learned how to do a pull request - and my git fu could use every bit of improvement. ;)

aldenhart, 2015-06-02:
Thanks for doing this. We should figure out a way to make pulse stretching an option and a worked example is invaluable.

bronek999, 2015-06-02:
My way is not solution.
I moved shortest pulses to unused motors. So XZYA works correclty and motors BC pulse is short. (<2uS)

bronek999, 2015-06-10:
My modificaton:   

1.   Add pulse_of_timer(counter) to each motor
typedef struct stRunMotor {				// one per controlled motor
	uint32_t substep_increment;			// total steps in axis times substeps factor
	int32_t substep_accumulator;		// DDA phase angle accumulator
	uint8_t power_state;				// state machine for managing motor power
	uint32_t power_systick;				// sys_tick for next motor power state transition
	uint8_t pulse_off_timer;
	float power_level_dynamic;			// power level for this segment of idle (ARM only)
} stRunMotor_t;

2.    modify DDA timer interrupt  (example only for motor4)

MOTATE_TIMER_INTERRUPT(dda_timer_num)

if (interrupt_cause == kInterruptOnMatchA) {
		if (!motor_4.step.isNull() && (st_run.mot[MOTOR_4].substep_accumulator += st_run.mot[MOTOR_4].substep_increment) > 0) {
			motor_4.step.set();
			st_run.mot[MOTOR_4].substep_accumulator -= st_run.dda_ticks_X_substeps;
			INCREMENT_ENCODER(MOTOR_4);
			st_run.mot[MOTOR_4].pulse_off_timer = 3;
		}
		if(st_run.mot[MOTOR_4].pulse_off_timer)
		{
			st_run.mot[MOTOR_4].pulse_off_timer--;
			if(st_run.mot[MOTOR_4].pulse_off_timer == 1)
				motor_4.step.clear();
		}
     	} else if (interrupt_cause == kInterruptOnOverflow) {
//		motor_4.step.clear();   // COMMENT THIS LINE
..............

aldenhart, 2015-06-23:
Guys, would 5 uSec pulses be long enough?

bronek999, 2015-06-24:
> Dátum:2015. 06. 23  12:52  (GMT+01:00)
> Komu: synthetos/g2 <g2@noreply.github.com>
> Kópia: bronek999 <bronek@amega.sk>
> Predmet: Re: [g2] Too short pulses (#92)

Yes, 5uS is OK for every driver l know

Odoslané zo služby Samsung Mobile

-------- Pôvodná správa --------

stg, 2015-07-12:
Do I remember correctly that the current time is about 3.75µS (ideally)?

My stepper drivers (generic run of the mill chinese 6A drivers, black casing with heat sink, green phoenix contacts) work intermittently at the current pulse length (stops working once warm). So I'd say 5µS will proably work but might still be on the edge for these fairly common drivers.

My (toggle) method is certainly not the solution as this basic version has a flaw in that a single step may be missed at each direction change. This error tends to even out as the odds are technically 50/50 over time - and I've not had problems until recently with a very small radii helix that was accumulating a large error very quickly with devastating results. Correcting this is no longer a simple quick-fix, so for now I'm considering halfing the DDA frequency instead, hoping the increased aliasing won't have too much of a negative impact.

Can i simply modify FREQUENCY_DDA without adverse effects or are there other things to consider as well?

aldenhart, 2015-07-26:
I've been working on an edge release that (among other things) will run pulses at 5 uSec - more if you slow down the step clock. I think that should handle most drivers. What do yours require?

stg, 2015-08-24:
They are specced at 200kHz max, so 2.5µS should've been enough.
However, as is common on the China market the specs are off.

3.75µS is enough when they are cold, but not as they heat.
7.50µS has been tested to work perfectly.
5.00µS unverified - but likely ok.

Been sifting though everything I could find on problems with these drivers:

Budget copies or bad benchmarking aside, they quite common and this problem is mentioned often, especially as the pulse length is shorter with the control system they ship with. Most people seem to have no issues with the original system though. When there are problems, the majority are borderline like mine.

Some have tried KFLOP with adjustable pulse length and report that the default 2.0µS makes it worse, the maximum 3.78µS improves/solves it. Others have built hardware to extend the pulses.

The response in these drivers is obviously all over the place and pretty bad to begin with compared with most everything else, still 5µS would probably work in almost every case. So if that is what we get without jumping through hoops, I think it is very good.

aldenhart, 2015-08-24:
We have been experimenting with a variant of the stepper interrupt routine that does the pulses a different way. The DDA timer is set up as a single phase, not two phase (PWM) as is currently. At the start of the ISR all step bits are turned off. Then the DDA runs to see which (if any) get turned on. 

This guarantees a minimum of 5 uSec when running the DDA clock at 200 KHz. It is also possible to drop the DDA clock down somewhat - say 150 KHz - to lengthen the pulses.

Please let me know if you think this would help your issues and we can post a special edge branch for you to try out.

stg, 2015-08-24:
I am happy with my current setup and don't want to cause you any additional work.
However, if you *want* it tested - I would be more than happy to try it out.

aldenhart, 2015-08-26:
I've been running this as my default now for a while, so thanks for the offer to test but no need to disrupt your setup.

closed 2015-08-26 by aldenhart