synthetos/g2 #94: OSX Install Error

FPOBDOC, 2015-06-03:
I have followed the directions (or think that I did) on a MBP running OS X 10.10.3, Arduino IDE 1.6.3, and am trying to install G2 on a Due.  I followed the directions and am trying to install the 078.03 elf file using the G2-edge package.  The elf file is in the g2-edge directory, which is the same directory from which I'm trying to use the installer.

When in the terminal, I used the instruction:
./DueFromOSX.sh -f TinyG2.elf -p /dev/cu.usbmodem*
And received the following error message:
-bash: ./DueFromOSX.sh: No such file or directory

I tried the other method of installation and  received the following error:
Bills-MacBook-Pro:g2-edge Bill$ export PLATFORM=gShield
Bills-MacBook-Pro:g2-edge Bill$ export SETTINGS_FILE=settings_default.h
Bills-MacBook-Pro:g2-edge Bill$ make
make: *** No targets specified and no makefile found.  Stop.

I've searched online and have struggled to find an answer.  I am admittedly new to this and am excited to try out TinyG2 on my CNC.  My interpretation is that, even though I followed the earlier instructions, I may have a bad installation of Xcode.  Any help would be really appreciated.

ril3y, 2015-06-04:
FPOBDOC,

You are in the wrong directory trying to issue the ./DueFromOSX.sh script.

<code>
ril3y@VOXCORE:~/Development/g2$ pwd
/Users/ril3y/Development/g2
ril3y@VOXCORE:~/Development/g2$ ls
README.md    Reference/   Resources/   TinyG2/      TinyGv2.inf  Tools/
ril3y@VOXCORE:~/Development/g2$ cd Resources/TinyG2-OSX-Programmer/
ril3y@VOXCORE:~/Development/g2/Resources/TinyG2-OSX-Programmer$ ls
DueFromOSX.sh*                      ReadMe.txt                          TinyG2.elf.13.01*                   TinyG2_Due-edge-078.03-default.bin
ril3y@VOXCORE:~/Development/g2/Resources/TinyG2-OSX-Programmer$ ./DueFromOSX.sh
</code>

You can see what I did..  That bash script is in the Resources dir then then TinyG2-OSX-Programmer directory.

FPOBDOC, 2015-06-04:
Thanks, ril3y.  I changed directories and have reinstalled the Arduino IDE (to make sure that it was in the right directory for the .sh script. I'm now getting a new error:

$ ./DueFromOSX.sh -f TinyG2.elf -p /dev/cu.usbmodem*
./DueFromOSX.sh: line 59: arm-none-eabi-objcopy: command not found
Forcing reset using 1200bps open/close on port /dev/cu.usbmodem1421
wait...
Starting programming of file TinyG2.elf -> TinyG2.bin on port cu.usbmodem1421
./DueFromOSX.sh: line 73: /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/bossac: No such file or directory

I went into the script and the problem is with the following line:
arm-none-eabi-objcopy -O binary "${file}" "${file/.elf/.bin}"

giseburt, 2015-06-04:
@FPOBDOC It's assuming that the compilation toolchain has been installed, and is trying to make a raw `.bin` file from the `.elf`.

If you have a `.bin`, you can safely comment out that line (with a `#` since it's a bash script) and run it again, passing the name of the `.bin` file.

It should look like this once commented out:

```bash
#arm-none-eabi-objcopy -O binary "${file}" "${file/.elf/.bin}" 
```

-Rob

Cinezaster, 2015-11-09:
I also had this problem with newer versions of the Arduino IDE. Switch back to arduino 1.5.2 try it again. It is looking for the bossac tool which in newer versions (1.6.x) of the Arduino IDE is not at that position.

justinclift, 2016-01-11:
This might be helpful for other people who hit the same issue, when trying with newer Arduino IDE's (v1.6+): https://github.com/synthetos/g2/pull/118

May take a bit of back and forth discussion until (if) that patch is merged, but the concept in it should be understandable in the mean time anyway. :smile:

aldenhart, 2016-09-08:
Please see PR for this issue #52

closed 2016-09-08 by aldenhart