synthetos/g2 #95: Long uptime issues

stg, 2015-06-17:
I've been seeing some strange issues when running jobs on a board that has been powered for a long time (several weeks) and I am therefore wondering if timer rollover issues have been taken into account?

Typically I've come to always reset the boards every time I need to use them, but sometimes I've been forced to keep them on for long periods of time waiting for tools and materials while needing to keep the system running, as my machine has no absolute positioning system and relies on steppers being powered and G2 remembering the exact position of the currently held microstep.

Attempting to restart the job at that point has yielded strange effects that I do not fully understand.

aldenhart, 2015-06-23:
The timer is an unsigned long that increments every millisecond. This gives a rollover interval of about 49 days (1 dog year). I guess this isn't enough! What symptoms did you experience on rollover, and can you explain the job restart problems a bit more?

stg, 2015-06-23:
Wow - I didn't realize I had it running that long, but looking back at the last time it happened it does seem to match the time frame. Also, this is the only occurance where I could think of no external issues that could be the culprit.

I can't account for the symptoms exactly but here's what I remember. By restarting the job I mean loading the g-code for the next tool and hitting run. The machine started cutting, but with an offset. I stopped the job, reloaded the g-code (thought I had gotten the wrong file) and tried again. This time I got another offset. I stopped again and tried to manually correct the offset (G28.3) but then the spindle would no longer start. This led me to reboot the board and try again. G-code files were verified to be correct upon successfully completing the job after the reboot.

While I cannot say for sure if timer rollover is the issue but I can't come up with any ideas on external causes and I've had similar timer issues myself on a few occations, various projects. Typically it can be easily fixed by rewriting any timer math to make rollovers transparent. Using a trivial blocking delay as an example (for completeness, I'm sure you're already aware):

while(now() < start + delay); // evil code
while((typeof(start))(now() - start) < delay); // rollover is transparent

aldenhart, 2016-02-23:
We have started using a Timeout object in the code that addresses this error. Parts of the code that use the timers will be refactored as those sections are open. So this issue will disappear gradually. I will leave this open for now until all timeout sections have been addressed.

aldenhart, 2016-09-08:
These have been addressed

closed 2016-09-08 by aldenhart