synthetos/g2 #96: Subsystem communication

stg, 2015-08-27:
This is a question, not an issue.

I've installed linear rules with 5µM resolution to all of my axes.

They've been connected to a homebrew box with quadrature decoders and zero mark detection, battery backup, DRO display, the works.

This system now always know exactly where the tool is located and could theoretically be used for closed loop feedback - but that's asking a bit much :)

However, what I would like to do it to be able to implement a custom command to TinyG, say M99 or whatever that sends a request over any exposed uart to my tracking system, which will reply with the exact machine coordinates and update the corresponding internal variables of TinyG.

This will allow me to zero the machine without moving it (well once, then it's tracked and battery backed). To restore origin after stalling, having to e-stop or reboot the system. It will also be able to report any inconsitencies after/while running a job, etc.

I will figure out how to implement this, but any hints you could give on where to start, proper way of doing this, etc. would likely reduce the time it will take me significantly and would as such be appreciated greatly!

aldenhart, 2015-08-27:
I think you can already do this. Use the G28.3 command. It sets the machine coordinates to the values you provide. 

A cautionary note - this command should be run in isolation and NOT embedded in the middle of a series of moves (e.g. in a Gcode file) or unpredictable results can occur. The system does not like to have its world coordinates changed in the middle of a set of operations.

stg, 2015-08-27:
You mean there is a traditional UART that already accepts G-code?

If there is, there would still be the issue of how to command the tracking system (via usb) to send the G28.3 command and how to have this command is execuded immediately instead of being placed at the end of the queue. I need to be able to embed the necessary command sequence into the g-code stream but I have no requirements as to what that command sequence would look like and it does not need to be a single command.

If I understand you correctly, the G28.3 command should never appear in a program?
Shouldn't this be possible, by... I don't know, forcing a stop, replan, start or something when that command is encountered?

aldenhart, 2015-08-27:
There is not a traditional UART that accepts Gcode - I was referring to the USB port. All commands must come in through that channel, so signalling your tracking system for coordinates would have to be something you did external to the controller and then sent the command as USB.

It's technically possible to put the g28.3 - and other setup commands such as the JSON configs - in a Gcode file, but I would consider this bad practice as they are machine setup and not part of the Gcode job. We view the layers of control as below, and think it's a good idea to separate them and not mix them:

- Machine setup and configuration. In which the {} commands and G28.3 belong

- Job setup. Some job setup is common at the head of a Gcode file (G21, G64...). But you might consider this as a separate step if you want to go into the settings and limit the maximum feed rate, jerk, or other parameters depending on the tool and material you are using. You don't want this in the Gcode file.

- Gcode job. The "tape" runs from start the M30 (or M2). I always number them as this ends up saving me time. This is what the USB data channel is used for.

- Run-time operator commands. Real-time commands such as feedhold that are issued from the "front panel". These take precedence over the tape and come in on the USB control channel.

So you could put it in the file, but I would not, and if you do please test very carefully. In retrospect, it would have been better not to have called the machine coordinate setup G28.3. It should have been some JSON like {mset:{x:2.33412.... At some point in the future we might deprecate the G28.3 form and promote a JSON form instead.

stg, 2015-08-27:
It sounds like you feel the best option is customizing the control program (chilipeppr) if I need this functionality as described?

Does the following sound like a safe procedure to you, upon encountering the synchronization command in the g-code stream?

1) Stop the "tape" (much like the pause-on-tool-change option)
2) Wait for the machine to come to a full stop
3) Request tracking data on a separate port
4) Issue the G28.3 command/future equivalent
5) Restart the "tape" :)

Allthough not optimal, this is certainly an acceptable solution.

aldenhart, 2015-08-27:
I would be concerned about resetting the machine coordinates in the middle of a job. Perhaps a better way yo do this would be to use the g92 offsets and just introduce a temporary offset. You can do that inside a job, as many times as you want.

stg, 2015-08-28:
Point taken.
Thank you for your support.

closed 2015-08-28 by stg