synthetos/g2 #97: SPI example

james-boulton, 2015-09-18:
Hi, I'm trying to use Motate within TinyG2 to communicate with some SPI devices on an Arduino Due board. I've setup an SPI device with --

(kSPI_LCDChipSelectPinNumber=52)
Motate::SPI spi;

and I'm trying a very simple test to write one byte via SPI --

spi.init(SystemCoreClock/20, kSPI8Bit | kSPIMode0);
spi.setChannel();
spi.write(0xaa, true);

I have a scope hooked up to SCK and SS (D76 for SCK / D52 for SS), When I write a byte I get the correct activity on the chip select pin, but no activity on SCK or MOSI. It appears as though SCK and MOSI are hiZ and not set as output.

What am I missing here? It looks like it should be working.

Thanks!

james-boulton, 2015-09-21:
Just as a quick note, I've ported over the Arduino SPI code and have SD card access working via that. I suspect its going to be something simple...

bronek999, 2015-10-02:
I have tft display connected to tinyg2 by spi

init routine:
PMC->PMC_PCER0 |= 1 << ID_PIOA;
	PMC->PMC_PCER0 |= 1 << ID_PIOB;
	PMC->PMC_PCER0 |= 1 << ID_PIOD;
	
	//configure for input			//MISO
	PIOA->PIO_PDR |= PIO_PA25;
	PIOA->PIO_ODR |= PIO_PA25;		//Input
	
	PIOA->PIO_PDR |= PIO_PA26;		//MOSI
	PIOA->PIO_OER |= PIO_PA26;		//MOSI	Output
	PIOA->PIO_ABSR &= ~PIO_PA26;	//Peripheral A
	
	PIOA->PIO_PDR |= PIO_PA27;		//SPCK
	PIOA->PIO_OER |= PIO_PA27;		//SPCK	Output
	PIOA->PIO_ABSR &= ~PIO_PA27;	//Peripheral A
	
	PIOD->PIO_OER |= PIO_PD8;		//CS pre displej
//	PIOA->PIO_ABSR &= ~PIO_PA29;	//Peripheral B
	PIOD->PIO_PUER |= PIO_PD8;		//pull-up
	lcd_spi_cs_high();				//

	//Enable clock for the SPI0 peripheral
	PMC->PMC_PCER0 |= 1 << ID_SPI0;

	//Disable the SPI0 peripheral so we can configure it.
	SPI0->SPI_CR = SPI_CR_SPIDIS;

	//Set as Master, Fixed Peripheral Select, Mode Fault Detection disabled and
	//	Peripheral Chip Select is PCS = xxx0 NPCS[3:0] = 1110
	SPI0->SPI_MR = SPI_MR_MSTR | SPI_MR_MODFDIS | 0x000e0000;
	
	//SPCK baudrate = MCK / SCBR = 84MHz / 128 = 656250Hz
	SPI0->SPI_CSR[0] |= 0x0000FF00  | 1 << 1;
	
	//Enable the SPI0 unit
	SPI0->SPI_CR = SPI_CR_SPIEN;

write to SPI

unsigned char LCD_SPIWrite(uint8_t val)
{
	//Wait for previous transfer to complete
	while ((SPI0->SPI_SR & SPI_SR_TXEMPTY) == 0);
	
	//load the Transmit Data Register with the value to transmit
	SPI0->SPI_TDR = val;
	
	//Wait for data to be transferred to serializer
	while ((SPI0->SPI_SR & SPI_SR_TDRE) == 0);
	
	while ( (SPI0->SPI_SR & SPI_SR_RDRF) == 0 ) ;

	return SPI0->SPI_RDR & 0x00FF ;
}

closed 2016-02-23 by aldenhart