synthetos/g2 #98: Adding a LCD

james-boulton, 2015-09-21:
I've been implementing an LCD based user interface on the Due itself (as has been done before), and I just wanted to get an idea of how this would best integrate with the main update loop.

The simplest option would seem to be adding a user interface update function into _controller_HSM(), but I'm wary about the latency of any GUI update code. The other idea I had was moving the call to _controller_HSM() into a fixed update frequency interrupt (say 50hz). This would leave the main thread free for GUI update and ensure the system is kept up to date.

Any input on this would be much appreciated.

aldenhart, 2015-09-21:
James - you are asking the right questions. I added this page to provide some guidance for the questions you ask. It's not much, but hopefully this is a start. Please come back with any questions, and feel free to edit the wiki if you see fit.

https://github.com/synthetos/g2/wiki/Developer-Notes

james-boulton, 2015-09-21:
That's great, thanks for the info. I have the LCD and SD card working now, so it should be "just" down to getting a GUI written and then tweaking latency as needed. Fantastic stuff.

aldenhart, 2015-09-21:
Cool. Let us know how it goes or if you have any questions

bronek999, 2015-10-02:
https://www.youtube.com/watch?v=tF4OMlCDmls

800x480 TFT lcd + sd card + touch panel + VFD control

aldenhart, 2015-10-04:
Fantastic. Thanks for the work.

james-boulton, 2015-10-05:
Thanks for that bronek999. I have a 320x240 LCD touch interface and SD card hooked up to TinyG2 now -- I'll see if its worth pushing the code for others when its fully working.

eltron247, 2016-01-11:
Any update on this? I'd love to know how the LCD is functioning.

bronek999, 2016-01-12:
I had no time for this project last months. But I added analog joystick for manual jogging independent on tinyg2 positions. Now I works about emergency stop procedure. Also tool change procedure. Stop spindle, move up few centimeters, measure tool length,wait for manual tool change, set correct tool ofset, return to work position and continue procesing g code.

Display works now 100%. I had problems with sd card reading with running spindle. Noise. Not it works ok.analog joystick-was problem also. Broken lot of mills.

james-boulton, 2016-01-12:
I have an LCD display hooked up (a little 320x200 one) with a simple interface working. I've not had time to work on this in a while, but will post up the source code somewhere when its working for interested parties.

eltron247, 2016-01-12:
@james-boulton Thanks much! 
@bronek999  The joystick for jogging on a remote pendant was my next task. How is it working out for you? Is it interfaced directly through the board or as a JSON device?

bronek999, 2016-01-12:
At present it is analog joystick connected directly to adc inputs. Steps are generated by pwm. For future I want to use joystick+button board connected by digital interface. Uart for example

eltron247, 2016-01-12:
I have thought about doing this but my approach was going to be using a SPJSON server on an rpi to multicast. With the LCD attached it could function as a full wireless pendant functioning as both send and receive.

closed 2016-02-23 by aldenhart