synthetos/g2 #100: Incorrect hardware platform reported in status report?

Mgilbride, 2015-09-28:
I compiled g2 edge with: make PLATFORM=gShield SETTINGS_FILE=settings_Makeoko2.h

I expect to see hp=2. But see hp=3 as shown below?

{"r":{"fv":0.98,"fb":83.09,"hp":3,"hv":0,"id":"0213-2325-7333","msg":"SYSTEM READY"},"f":[1,0,1]}
tinyg [mm] ok>

diverseg, 2016-06-09:
I as well expect to see 2 = Native Arduino Due, but the status shows
 [hp]  hardware platform           3.00

aldenhart, 2016-09-08:
Taken for verification, now that we have the ability to differentiate hardware.h files per-board as part of he build process.

aldenhart, 2017-01-07:
Update - this item will not be addressed directly, instead in a future release the hardware platform and revision will be changed to strings that are set by the board files. This item will remain open until that release is pushed.

closed 2018-05-13 by aldenhart