synthetos/g2 #102: Variable Homing Behavior

Mgilbride, 2015-09-30:
I am getting variable homing behavior per homing cycle. And I have run out of ideas as to what I may have set up incorrectly. 

I have limits disabled. All three axes home correctly. However, If I do a move on any axis (resets homing flag?) and then re-home the axis, the axis does not follow the full latch back-off or zero back-off. But it usually still homes without issue. If I follow that with another move on the same axis and then re-home that axis I get different behavior again. The axis does not execute the correct latch back-off and it keeps driving into the switch at latch velocity. All homing is at 800mm/min and latching is at 100 mm/min. I do not have my search velocity set to negative as noted in settings comments. But I do have X and Y homing in negative direction and Z homing in positive direction as set in settings. Any suggestions?

The final event's terminal output is as follows:

Firmware: {"r":{"fv":0.98,"fb":83.09,"hp":3,"hv":0,"id":"0213-2325-7333","msg":"SYSTEM READY"},"f":[1,0,1]}

Hardware :Arduino DUE

posx:102.345,vel:0.00,stat:3

Issue a G28.2 X0 command here...
 
posx:99.173,feed:800.00,vel:800.00,coor:0,momo:1,stat:9
posx:95.834
posx:92.494
posx:89.155
posx:85.836
posx:82.476
posx:79.137
posx:75.797
posx:72.458
posx:69.138
posx:65.779
posx:62.439
posx:59.099
posx:55.759
posx:52.419
posx:49.079
posx:45.739
posx:42.399
posx:39.079
posx:35.719
posx:32.378
posx:29.038
posx:25.698
posx:22.378
posx:19.018
posx:15.678
posx:12.338
posx:8.998
posx:5.678
posx:2.318
posx:-1.022
posx:-4.362
posx:-4.905,feed:100.00,vel:100.00
posx:-5.321,vel:0.00
posx:0.000,feed:0.00,coor:1,momo:4,stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
stat:3
I cut power to motors here to reset machine...

yck011522, 2015-09-30:
Just out of creative ideas, could it be noise pollution from motor wirings to switch inputs.

Mgilbride, 2015-09-30:
I have switches wired with shielded cable, and run at 48v DC to relays that
need 48v to actuate. Should be exceptionally immune to noise. However, not
impossible that noise still is an issue. This problem is quite repeatale.
So I still doubt noise is the problem. Thanks for the suggestion.

Mgilbride, 2015-09-30:
I am confused about what is implied in settings files. I am looking at the shapeoko.H file. Settings comments state that a (-) homing search velocity will move search in axis negative direction. But $xhd also seems to set  this same property?

aldenhart, 2015-10-01:
The homing direction sets the direction of movement. The comment on the search velocity is old and incorrect. Please use the direction value. We'll fix the comments. Sorry for the confusion.

Mgilbride, 2015-10-01:
No worries. Thanks.

closed 2015-10-04 by aldenhart