synthetos/g2 #103: Homing behavior

Mgilbride, 2015-10-05:
Sorry Alden, I know you closed my last homing issue but I am still figuring out how to get it working reliably. After a lot of experimenting and reading I am wondering about the homing back off setting. Why are we setting a back off value rather than searching back and forth over the switch and reading state changes on the homing pins to automatically locate home? Then execute the homing pull off.  I think grbl has homing executed rather well in that respect. Very reliable in all of the last releases I used.

aldenhart, 2015-10-05:
Thanks for the note. Perhaps I should not have closed the old issue. 

I know homing works reliably, but perhaps it is confusing to get set up correctly. Here are my settings for the Othermill. This is a screw-driven machine that has very high precision. Input 1 wired to a NC microswitch mounted on the left end of the X axis (i.e. X negative) that has a very short throw of less than 1 mm.

[xam] x axis mode                 1 [standard]
[xvm] x velocity maximum       1500 mm/min
[xfr] x feedrate maximum       1500 mm/min
[xtn] x travel minimum            0.000 mm
[xtm] x travel maximum      145.600 mm
[xjm] x jerk maximum            500 mm/min^3 * 1 million
[xjh] x jerk homing              1000 mm/min^3 * 1 million
[xjd] x junction deviation       0.0000 mm (larger is faster)
[xhi] x homing input              1 [input 1-N or 0 to disable homing this axis]
[xhd] x homing direction          0 [0=search-to-negative, 1=search-to-positive]
[xsv] x search velocity         500 mm/min
[xlv] x latch velocity           25.00 mm/min
[xlb] x latch backoff             1.000 mm
[xzb] x zero backoff              0.400 mm

Homing is set up for digital input 1, which is configured as below. The homing cycle "takes over" the normal operating function of the switch (which is none, in the current config), so the only relevant setting is the input mode, which is normally closed. The other 2 can be ignored.

[di1mo] input mode            1 [-1=disabled, 0=NO,1=NC]
[di1ac] input action            0 [0=none,1=stop,2=halt,3=stop_steps,4=panic,5=reset]
[di1fn] input function          0 [0=none,1=limit,2=interlock,3=shutdown]

If you wanted input 1 to act as a limit switch during operation you would configure it as so:

[di1mo] input mode            1 [-1=disabled, 0=NO,1=NC]
[di1ac] input action            1 [0=none,1=stop,2=halt,3=stop_steps,4=panic,5=reset]
[di1fn] input function          1 [0=none,1=limit,2=interlock,3=shutdown]

The homing sequence is:

1. Test if the opposite switch (X max) and back off of it if it's closed. There is no Xmax, so homing runs past this step
2. Move towards the homing switch in the homing direction - in this case towards negative as configured
3. When the switch is hit decelerate rapidly using the jerk homing setting
4. Move in the positive direction off the switch by the latch backoff amount. Latch backoff must be large enough to ensure that the switch has been cleared
5.  Move back towards the switch at the latch velocity until it closes again.
6. Move away from the switch by the zero backoff amount and set this as the axis zero.

The reason for clearing the switch an re-approaching is that many switches are asymmetric - they have hysteresis between the closing and opening positions. Further, the closing position is more repeatable that the opening position, hence the 2 pass homing.

Please let me know if this clarifies things.

Mgilbride, 2015-10-05:
Thanks Alden. I think I understand it correctly. It's working on my machine. But I am still verifying repeatability. A little confusion comes from changing terms and schemes in settings and firmware. But that comes with the territiory. Just trying to sort it out. 

Mostly, I am asking about the potential benefit of searching for a signal change (i.e. homing switch cycling on and off) with the firmware instead of asking the user to input the homing backoff settings. I understand the need for the backoff and why it works that way. But am wondering if searching back and forth over the switch with firmware might not be a more robust / reliable way of dealing with hysteresis in the switches / various switch types. And that might keep users out of trouble from setting the back off value too narrowly.

aldenhart, 2015-10-05:
Thanks for the heads up on the signal change idea. Sounds like a good addition to the homing routine. Removing the latch backoff value would simplify things.

closed 2015-10-05 by Mgilbride