synthetos/g2 #104: Move Time is NAN - prep isnan

ryansturmer, 2015-10-12:
When running a file, I get the following failure:

```
g2: <-C--1444428812746---- {"sr":{"posx":11.741,"posy":1.501,"posz":0.402,"vel":120.03}} [g2]
g2: <-C--1444428813004---- {"sr":{"posx":11.744,"posy":1.505,"posz":-0.099,"vel":120.00,"line":34}} [g2]
g2: <-C--1444428813073---- {"er":{"fb":83.09,"st":31,"msg":"Move time is NAN - prep isnan"}} [g2]
g2: <-C--1444428813074---- {"r":{},"f":[1,206,42]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
```

The G-code that generates the failure is here: https://gist.github.com/ryansturmer/bd9e2d5f289696c8d32e

And the current configuration that we're using is here:
https://gist.github.com/ryansturmer/efcc38a5077e43165775

ryansturmer, 2015-10-13:
It appears that this is a very small arc (0.0003 in radius)

ril3y, 2015-10-18:
FYI to all.  Alden and Ryan had a google chat about the issue and it looks
like we are working on a fix for this in the near future.

ril3y, 2015-10-18:
FYI.  It looks like @aldenhart and @ryansturmer had a google chat about this issue and are working on a fix for this in the near future.

aldenhart, 2016-02-23:
See edge build 089.xx or later

closed 2016-02-23 by aldenhart