synthetos/g2 #105: Compiling for an arduino Zero would be a plus for this new hardware platform on kickstarter

Cinezaster, 2015-10-23:
Hi guys, was browsing through kickstart projects and came by this hardware board.
https://www.kickstarter.com/projects/1204283/zeropi-arduino-and-raspberry-pi-compatible-develop?ref=thanks
Running G2 on the arduino Zero, and using the Raspberry to run Chillipeppr  and your good to go. 
What do you think?

teshko, 2015-10-23:
It appears to be doable, but not without ZeroPi schematic, and some compromises.
Mind you, arduino Zero and ZeroPi feature different microcontrollers. SAMD21G18 (Zero - 48 pins) vs SAMD21J18 (ZeroPi - 64 pins) vs ATSAM3X8E (Due - 144 pins)

giseburt, 2015-10-23:
Hi Guys,

We've looked at that device specifically, along with the Zero and the SAMD21 line in general.

With the Arduino Zero available now, it would be feasible to port Motate to the SAMD21 and then G2 would only need pin mappings in order to work. It could be prototyped with the gShield.

If anyone is willing to help, we'll support you and be very grateful! (Also, please let us know so we can possibly combine efforts.)

-Rob

Cinezaster, 2015-10-24:
I'll get in touch with the makeBlock guys for the pinouts. 
This can be a starting point 
https://github.com/Makeblock-official/ZeroPi/blob/master/zeroCnc/zeropi.h
They already ported Marlin to their platform. Here is the pinmapping
https://github.com/Makeblock-official/ZeroPi/blob/master/MarlinZero/Marlin/pins.h

anthonywebb, 2015-12-24:
@Cinezaster did you ever make any headway on this with the makeBlock guys?

Cinezaster, 2015-12-24:
@anthonywebb I figured out that we already have the pinout if you look at the links I posted.
If someone needs more info, I'm happy to get in touch with them. With a more specific question.
The board is now in production and will be arriving beginning of January. I was waiting for it to show up, before moving forward
I just asked them directly about the schematics or chip pinouts.

ril3y, 2015-12-24:
Please keep us updated Cinezaster.  We would like to hear about the new
board.  Also join up our google group here:
https://groups.google.com/forum/#!forum/devtinyg

Riley

Cinezaster, 2016-02-16:
Received the ZeroPi. Next week I will try to compile G2 for SAMD21J18. I'll move to the google groups with my questions and troubles.

ewidance, 2016-02-28:
Any updates on G2 compilation?

Cinezaster, 2016-02-29:
Waiting for the schematics of the board. 
And I'm loaded with my daytime job at the moment.

aldenhart, 2016-09-08:
I am closing this issue. The new code in the edge-100 branch provides better support for supports multiple board targets and processor architectures, but a SAM D21 port is not currently on our roadmap.

closed 2016-09-08 by aldenhart