synthetos/g2 #106: Format of received command ie: [0=disable,1=enable]

prrvchr, 2015-10-24:
I try to write a small utility for driving TinyG2 card in FreeCAD.
You can take look at: https://github.com/prrvchr/USBTerminal
I want to put some widget for editing settings and want to be able to determine the type of widget.

[0.05 < mfo < 2.00] -> [0.05:minimum,2.00:maximum]
[1-6 (G54-G59)] -> [1=G54,2=G55,3=G56,4=G57,5=G58,6=G59]
[0=X,1=Y,2=Z...] -> [0=X,1=Y,2=Z,3=A,4=B,5=C]
[0.000=minimum,1.000=maximum] -> [0.000:minimum, 1.000:maximum]
[disabled] -> ???
[input 1-N or 0 to disable homing this axis] -> ???

the goal is to be able to determine the widget (combobox, spinbox...), the list of available choices and eventually minimum and maximum if they apply.
Many settings are in degrees can we determine if they are on an integer or float: deg or deg/rad
Thanks.

aldenhart, 2015-10-26:
Have you taken a look at the JSON commands? That's the recommended way to talk to the board. Thanks for looking into this.

prrvchr, 2015-10-27:
Hi Alden,

Sorry I just take a look to JSON, and I use it now... I can make a
TreeWidget with....and is't more wonderfull.
But I have less information with JSON.
We have two solution: I hardcode erery settngs in my model; and need
changing accordingly to your change...
or we define a more reliable format and if new command or change command
occurs I have nothing to do...
With grbl I could follow the second solution.
In fact, it would be best to have these information in JSON...

I understand that this is not an issue but rather an improvement...

Ps:I have a strange behavior: when I send the comand {"$": n} it locks the
card and I need to power off for restarting and PySerial is over: I need to
retstart application... Until now I have around this problem by sending:
{"q":n} {"m":n} {"o":n} {"sys":n} consecutively. Same probleme if I do
whith miniterm.py: a PySerial terminal application.
And with JSON I cannot determine the end of a query command... while with
command $ I had a prompt return ie: "tinyg..." I have to do a line by line
treatment with PySerial and it is not great, or else I'm missing someting
...

Thank you for taking your time ...
Pierre

closed 2015-10-30 by prrvchr