synthetos/g2 #107: Mist coolant doesn't work?

lutorm, 2015-11-01:
I was just testing the various outputs on the scope and while an M8 would make the coolant output go high, an M7 would not. I was under the impression that both coolant commands would make the pin go high? Or is there a config somewhere determining which command the coolant pin responds to?

Cheers,

/Patrik

anthonywebb, 2016-01-01:
Which firmware are you running? In the default firmware, settings/settings_default.h at line 58 there are some entries for how the coolant is handled:

#define COOLANT_MIST_POLARITY       1                       // 0=active low, 1=active high
#define COOLANT_FLOOD_POLARITY      1                       // 0=active low, 1=active high
#define COOLANT_PAUSE_ON_HOLD       true

http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m7-m8-m9

lutorm, 2016-01-02:
This was with the current edge. I have it working now, but I'm no longer sure whether I made any code changes or if it was just a configuration issue. I'll try to see if there are any coolant changes among all the other changes I have, but it's likely I just didn't have it set up right.

closed 2016-02-23 by aldenhart