synthetos/g2 #110: Configurable maximum acceleration?

lutorm, 2015-12-08:
As I understand it, it not possible to set a maximum path acceleration currently. However, if the move is planned as a constant jerk path from 0-max velocity, the maximum value of the acceleration during the move can get arbitrarily high (depending on the jerk and maximum velocity.)

Since the max acceleration a machine can attain is limited by motor torque, maybe it would be useful to set an upper limit on acceleration?  Without this, the jerk value may have to be lowered so a long move doesn't fail, even though shorter moves that never attain high acceleration values could safely use higher values.

giseburt, 2015-12-08:
Hi @lutorm,

We've put a lot of thought into whether or not we should limit acceleration.

TL,DR; For now, we've decided it's unnecessary.

Long form: Our motion control currently is beyond even "constant jerk," and at the beginning and end of each "block" (roughly mapping to a gcode line), we have zero jerk and acceleration. The *peak* jerk along the motion curve is what you set when you set the maximum jerk for an axis.

Here's an animation I made illustrating our current motion profile, showing a single acceleration curve:

[image]

Of course, a deceleration curve would be the same, but reversed.

As you can see, the peak acceleration in the center of the move is not "bound" artificially, but is limited by the jerk and the length over which the acceleration occurs. This means that you *do* limit the acceleration, but only indirectly by limiting the peak jerk and the maximum velocity of an axis.

Also, most of the literature that refers to the torque of a motor in relation to acceleration is referring to *sustained* constant acceleration with "infinite jerk" at the beginning and end of the velocity ramp. We've found the the limiting factors genuinely are jerk and velocity.

With too high of velocity you start to get mechanical "chatter" and friction issues.

We haven't found another jerk-limited motion control system. All of the others have to limit their acceleration so that the (impossible) infinite jerk at the beginning and end of a velocity ramp doesn't exceed the torque capability of the machine. "Infinite jerk" is impossible, so the jerk *will* occur. It just happens over an unknown length of time at the beginning and end of a velocity ramp, and is roughly proportional to your acceleration rate. So, in order to limit jerk in a system that doesn't directly manage jerk, you *must* limit acceleration.

Now I'll admit this is mostly theory backed by years of testing and experience. However, if anyone has the equipment and time to prove scientifically that acceleration does or does not need to be limited directly and independently, we would help in whatever way we can, and would be very grateful.

-Rob

lutorm, 2015-12-08:
Hi Rob,

Thanks for the answer -- that helps, and that animated graph is great!  I was not aware that the planned jerk also smoothly went to zero at the ends.

As for the "inifinite jerk" systems, given the finite resolution of a stepper system it seems hard to define the actual output jerk at the beginning/end of a move: first you have no pulses, then you have one. Formally that represents both an infinite jerk, acceleration, and velocity, since it's a step function in position. To even estimate the jerk, it seems you'd need at least 4 points since it's the 3rd derivative of position.

It's also not at all clear to me what torque "means" for a stepper motor that's trying to accelerate. My question is more based on that of an ideal physics model of a motor as a constant-torque device. You may well be right that for an actual stepper motor, in an actual mechanism with friction and resonances, an acceleration limit is not practically relevant. (It would seem it would also make the motion planning quite a bit more complicated, which may be enough of a reason all by itself.)

It's easy to construct a theoretical example of a machine with negligible friction, large inertia, and low torque motors, where the resulting acceleration would violate the torque limit though. Imagine a rotary axis with the motor on the axis of a large flywheel. Such a system should have a very high velocity limit, but depending on the mass of the flywheel and the torque of the motor, it may have to accelerate slowly to get there. This would be a system close to the "sustained constant acceleration" that you mentioned.

Regards,

/Patrik

closed 2016-02-23 by aldenhart