synthetos/g2 #111: Compiling on x64 Ubuntu

aShure, 2015-12-17:
TL;DR: Install lib32z1.

g2 compiled fine with a system where all the necessary tools are already installed, but starting from scratch on a fresh Ubuntu Wily x64 doesn't seem to work without some tweaks. First of all, if you try to use the arm-none-eabi package from the most recent aptitude update, you'll end up with the version (``gcc-arm-none-eabi_15%3a4.9.3+svn227297-1_amd64``) which will throw the usual non descriptive error like breaking at `include <type_traits>`. I didn't investigate this further, because the Makefile includes a handy download script that should fetch a working arm-none-eabi package to the ../Tools directory. However, after downloading and deflating it, Make can't find it: ``No such file or directory``. I thought the issue might be because it's outside of the build dir. I still think it's bad practice but validating each step, it should indeed work. It's just a horribly non descriptive error message, the file is indeed at the right place and accessible. The only thing missing is a 32bit lib to run the binary.

``file arm-none-eabi-gcc``
``arm-none-eabi-gcc: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.2, for GNU/Linux 2.6.8, stripped``

This should solve it:

``apt-get install lib32z1``

ril3y, 2015-12-17:
Great work.  Would you mind updating the wiki?

aShure, 2015-12-17:
Done: https://github.com/synthetos/g2/wiki/Compiling-G2-on-Linux-and-OS-X-%28command-line%29

closed 2015-12-17 by aShure