synthetos/g2 #115: Which pins can be used for IO?

anthonywebb, 2016-01-01:
I am building a PNP that uses TinyG2 for motion control.  I am trying to find a couple pins I can toggle on/off a vacuum pump.  Worst case I can use the spindle/coolant pins, but wasn't sure if there was a more general purpose IO pins I could use for that?  Looks like in other systems the M42 command has been used for this? http://reprap.org/wiki/G-code#M42:_Switch_I.2FO_pin

The other thing that would be ideal is to have an input I could tie to a vacuum pressure sensor to know if a part is on the tip or not.

Is there anywhere that I can learn more about IO in G2, hoping it is already in place.  So many people talking about Smoothie being far more flexible, but I really do prefer the tinyg assuming it is capable of meeting my needs.

giseburt, 2016-01-02:
Hi there. We have an extensive IO system in place: https://github.com/synthetos/g2/wiki/Digital-IO-%28GPIO%29

Not the generic outputs are not implemented yet, so you're limited to coolant/spindle for now. We've gotten a few nudges on that recently so it's a priority. 

As for inputs, you can put inputs in your status report, and it'll automatically send a report when an input changes. This is in addition to the other actions a input can have, so you can get a report AND trigger a feed hold, or you can just get a report.

Other than generic outputs not yet implemented, does that sound like what you need?

-Rob

anthonywebb, 2016-01-05:
@giseburt thank for the heads up!  Sounds like exactly what I need.  I do need 4 generic outputs, so I will need to wait until those are in place before I can shift over to tinyg2 entirely.  But I love that the inputs can be part of the status report.

BTW, I'm not sure where the best place to put questions around the serial JSON api.  I left one over in the tinyg project but are we better to keep conversation in this project? https://github.com/synthetos/TinyG/issues/151

anthonywebb, 2016-01-07:
@giseburt @aldenhart here is what I am trying to do.  I need at least 5 axis motor control and to control 4 solenoids for my PNP project.  (I may need to add an input in here as well) but I'd rather not reinvent the wheel and your grbl shield is so nice.  So, we have a due breakout board for tinyg2 that will allow you to plug in 2 grbl shields.  Just need somewhere to map those 4 outputs for the solenoids.  But in general, for people who need to control more than 3 or 4 motors, this shield should work too.

[image]

anthonywebb, 2016-01-15:
@giseburt @aldenhart even just a brief state of the union for getting IO into G2 would be great, I'm trying to decide what platform I should drive my PNP project from.  Thanks!

aldenhart, 2016-01-15:
Anthony, Generalized outputs are on the short list. I expect sometime in the next month or so. We are pushing out a lot of changes that have been in the works for a while. Once that's done it's planned for the next round of features.

anthonywebb, 2016-01-15:
Great to hear!  We'll hang tight on this board design then.  Getting some buzz for people asking for it.

aldenhart, 2016-01-15:
Thanks. I'll leave the issue open so it can be updated as progress occurs.

anthonywebb, 2016-01-30:
I have been asked to add an eeprom chip, or perhaps even a microsd slot onto this breakout board for storing conifg (since the due it mounts to has neither)  One thing I love about the smoothieboard is that single config file that lives on the microsd, open in a text editor, tweak, save, etc.  Right now I have to recompile every time I make a tweak to my G2 running on the due.

So I am looking for some advice.  Which route (eeprom or microsd) would be the easiest to integrate into G2?  Would I be forced to create my own fork, or would this be something that could make it into the G2 core codebase?

Thanks @aldenhart @giseburt

anthonywebb, 2016-01-30:
The only other question I have is about the general state of G2 development.  Do you forsee a G2 reference board coming that has support for more than 3 axis?  Are there other people developing something like this?  I'd rather not duplicate efforts.

aldenhart, 2016-01-30:
Anthony, There are a couple of options. There is a MicroSD on the v9 board and Othermachine has a FAT system integrated into it. SO that code is available for use, but of course may require some tweaking. Separately, we are working on a 5 axis board that may serve your purposes. We are also extending the IO system to a very general digital/analog IO in support of these efforts. It may make sense to get n contact to discuss your requirements in more detail.

anthonywebb, 2016-01-30:
Great to hear that a 5 axis board is coming!  That is precisely what I am
looking for.  My PNP machine uses all 5, in addition I need 4 outputs to
trip the solenoids I use to control the vacuum.  Most of the 3D printer
boards out there will work for my application, I just use the fan or heated
bed outputs.  Currently am using the SmoothieboardX5, but if tinyg had a 5
axis option with some outputs I would switch in a heartbeat as I love the
"motion is king" mentality that tinyg focuses on, too many of these
projects are trying to cram webservers, LCD's, etc onto these boards which
I dont necessarily agree with.  The only input I have been asked for so far
is a vacuum pressure sensor to know if a part was picked up or not (if I
did that I would need 2 inputs) but I can accomplish that task with an
upward looking camera too.  Please keep me updated on your 5 axis board
progress, if you need anyone to help you beta things I am happy to help.

https://hackaday.io/project/9319-diy-pick-and-place

aldenhart, 2016-01-30:
I saw your project already, in fact. Earlier today. Very impressive. Check out htis page to see where we are going with the design for the general IO system. The digital inputs are in place, but will have some changes. The second part of the page describes where this is headed. Expect some rapid changes in that part as we are really just using the page for design collaboration at this point. Comments are welcome.

anthonywebb, 2016-01-30:
thanks @aldenhart can you shoot me a link to the page you are referring to? happy to help where I can

aldenhart, 2016-01-30:
Meant to do this before: https://github.com/synthetos/g2/wiki/Digital-IO-(GPIO)

anthonywebb, 2016-02-18:
@aldenhart the 5 axis board you are working on, can you confirm if I will be able to bypass the stepper drivers and have direct access to the step/dir pins?  I use clearpath servo motors which have stepper drivers built in, they just need step/direction signals (5v I think?).

aldenhart, 2016-02-18:
@anthonywebb - the step, direction and enable signals will be available. We have not yet decided if they will be buffered - so they may be 3.3v. Can you find out if this will drive the Clearpaths? If not it's more weight towards adding buffering / level conversion for those outputs.

anthonywebb, 2016-02-18:
@aldenhart yes I did verify, the clearpath, like many other servos and drivers out there supports 5-24v logic levels.  Previously with the due running G2 I had to make a shield to fix this, would be ideal if it was handled by the board itself. We'd be happy to help knock it out if it is to the point where you need some extra hands.  Thanks.

aldenhart, 2016-02-18:
Thanks for checking. So in order to make this function useful it really needs to be level shifted to 5v. I guess that means some more chips are going on the board. Such as it is. 

There's another consideration that could enter into this. The v9 board has a safety system that reacts to an external Shutdown signal. Mind you this is not an Estop function as Estop should always run external to the controller - in case the controller has lost its mind. Shutdown controls what the board should do in the event that an external Estop performed. The external Estop asserts Shutdown.

On Shutdown the v9 does a number of things, one of which is disable steps from getting to the motors. It leaves the direction and enable lines alone as in some cases the shutdown sequence actually wants to keep the motors powered and enabled. Sounds like something similar is warranted for the external step/dir/enable logic as well. Switch the steps and leave enable and direction in their previous state.

Any thoughts on this?

anthonywebb, 2016-02-18:
In the case of a shutdown, the external step/dir should behave as you lined out such that no more steps will get to the motor, but the enable/direction pins should stay energized.  De-energizing especially the Z on say a CNC application may mean that the router would dive into the material.

I dont know how much you have played with these clearpaths, but in addition to the EN/STEP/DIR they have an "HLFB" (high level feedback) output.  The HLFB can be configured to do lots of great stuff.  The one probably used the most is "servo on" which pulls the HLFB high when the motor is enabled, and will remain high whiles the motor is not in a shutdown state.  This is useful for emergency runaway situations or can be used as an endstop for homing. (motor moves to a hard stop and the HLFB is pulled low)

[image]

[image]

aldenhart, 2016-02-18:
Yes, I agree with the output spec you listed, and set it up this way for exactly the same reason - there are some machines that do not react well if you randomly disable the motors.

I have a set of the Clearpaths at home I have been playing with. They are pretty amazing. I especially like the little dance they do for minutes when you calibrate them. I had not looked into utilizing the feedback signal, but I can see the uses for it.

anthonywebb, 2016-02-18:
I have been wondering, related to the great motion planning of TinyG2, if enabling the RAS on these clearpath motors is needed at all? or may even be detrimental?  I think I have my RAS off, but after watching the following video was second guessing maybe turning it on? https://www.motionsolutions.net/store/pc/catalog/images/teknic/videos/11-Step%20&%20Dir%20(720p,%20CBR%200.5Mbps).mp4

aldenhart, 2016-02-18:
I can only figure we should test the RAS function. I should fire them up and see what they do. The RAS manages the 3rd and 4th derivative. TinyG does one more than that, so I'm not sure what we'd see.

aldenhart, 2016-02-23:
We are currently working on the IO system to include outputs. We will announce this and close this issue when it's available for test. Any new issues with the IO should be opened and new issues.

yankee14, 2016-08-18:
How is this coming along?  : )

aldenhart, 2016-09-08:
It's coming along nicely. Please take a look here -->https://github.com/synthetos/g2/wiki/Digital-IO

I am going to close this issue.

closed 2016-09-08 by aldenhart