synthetos/g2 #116: Compile Issues Windows 10

BK64, 2016-01-02:
I am aware this OS is untested.  I am using "Atmel Studio 6.2 sp2 (build 1563) Installer – with .NET" 

I have the following problems (tried with a clone of the files then tried with the Zip of the project).  See the attached text file.

[Compile Errors.txt](https://github.com/synthetos/g2/files/76369/Compile.Errors.txt)

BK64, 2016-01-28:
After looking further under solutions explorer, it seems that the project is trying to pull arduino files from
C:\.....\g2-edge\TinyG2\arduino\ and C:\.....\g2-edge\TinyG2\arduino\USB\.  I was trying to copy files over the arduino libraries and could only find maybe 75% of the files Atmel studio was looking for.  I verified that I have "Arduino IDE for Atmel Studio" (V1.6.6) installed and am completely baffled as to what else I am missing.  I am thinking the problem is plugin related, there is something funky going on with the make file, or the system is trying to use an incorrect or missing compiler because multiple projects on my other computer survived the windows 7 to 10 change and the Atmel Studio 6 to 7 change.  Does anyone have any clues as to what I should look at?

giseburt, 2016-01-28:
It appears you are trying to compile master.

Can you try the edge branch and see if it compiles? Be aware that we are actively working on it, but we usually mention that it own't compile in the commit message.

Thank you!
  -Rob

BK64, 2016-01-28:
I re-cloned edge, making sure to follow the windows compile wiki page step by step, and still ran into the same problems as initially reported.  Heavy googling led me to this forum (http://www.avrfreaks.net/forum/windows-81-compilation-error).  I tried out the solution of replacing the msys-1.0.dll file located within WinAVRXXXXXXXX\utils\bin with the one posted on the website.  Using the replacement file, I was able to get much farther in the compiling process (see attached error log). 
[Compile Errors 2.txt](https://github.com/synthetos/g2/files/108786/Compile.Errors.2.txt)

BK64, 2016-01-31:
I was able to compile finally on Windows 10.  The solution about replacing the msys-1.0 file allowed gnu to start creating the output files.  I ended up uninstalling WinAVR and replacing it with MHV AVR tools as mentioned in the forum link above though I still believe the WinAVR fix would still have worked.  My last problem had to do with the fact that my files were kept in a desktop folder.  I copied the entire project to the "Documents" folder which made it finally able to compile.  I successfully compiled the project on Atmel studio 7 as well to see if the fix still worked.  Attached are the .elf files generated from both projects (delete the .txt extension) to compare and see if they were built correctly.  Neither Atmel studio 6.2 or 7 generated a .bin file.
[TinyG2 Studio 6-2.elf.txt](https://github.com/synthetos/g2/files/111341/TinyG2.Studio.6-2.elf.txt)
[TinyG2 Studio 7-0.elf.txt](https://github.com/synthetos/g2/files/111342/TinyG2.Studio.7-0.elf.txt)

closed 2016-02-23 by aldenhart