synthetos/g2 #119: Compile Fails in config_app.cpp

eltron247, 2016-01-12:
To start off I think I've tracked the bug down but I'm not 100% sure. When attempting to compile, the parameters: MANUAL_FEEDRATE_OVERRIDE_ENABLE and MANUAL_FEEDRATE_OVERRIDE_PARAMETER at the end of lines 486 and 487 in "config_app.cpp" remain undefined.

{ "sys","mfoe",_fipn,0, cm_print_mfoe,get_ui8, set_01,   (float *)&cm.gmx.mfo_enable,           MANUAL_FEEDRATE_OVERRIDE_ENABLE},
	{ "sys","mfo", _fipn,3, cm_print_mfo, get_flt,cm_set_mfo,(float *)&cm.gmx.mfo_parameter,        MANUAL_FEEDRATE_OVERRIDE_PARAMETER},

There was a commit that added these parameters to the shopbot settings file. They're listed on lines 147 and 148. From what I can tell, in this manner you can only compile if you're using the shopbot file. This excludes all others but specifically those that edit their own file from the defaults. 

    #define MANUAL_FEEDRATE_OVERRIDE_ENABLE     false
    #define MANUAL_FEEDRATE_OVERRIDE_PARAMETER  1.00

I'm not experienced enough in coding to know, however, I propose that there be a #ifndef to setup a default on these parameters either in the header or cpp "config_app" files. Is this solution appropriate?

eltron247, 2016-01-12:
I've fixed the issue with this commit:
https://github.com/eltron247/g2/commit/c75aea27ed21c815a2e681451dcc932c8901b3f7

closed 2016-01-12 by eltron247

yankee14, 2016-08-18:
I'm still having this exact issue OP describes.

Steps to reproduce:
1. `git clone -b edge https://github.com/synthetos/g2.git`
2. `cp g2/TinyG2/settings/default_settings.h g2/TinyG2/settings/my_settings.h`
3. `cd g2/TinyG2`
4. `make SETTINGS_FILE=my_settings.h`

aldenhart, 2016-08-21:
You need to define these parameters in the settings profile you are using
#define MANUAL_FEEDRATE_OVERRIDE_ENABLE     false
#define MANUAL_FEEDRATE_OVERRIDE_PARAMETER  1.00

yankee14, 2016-08-21:
Thanks, guys. I got it to compile with your suggestions.
> You need to define these parameters in the settings profile you are using

Wouldn't it be best to set these to a default so that it at least compiles without users having to do commenting or defining? I think a default settings file should have sane enough defaults so it compiles right out of the box. Just my opinion.

aldenhart, 2016-08-21:
The newer code that we are prepping for egde push works as you suggest.