synthetos/g2 #123: Long moves ending in bad deccel?

stg, 2016-01-22:
Ok, so I've been scratching my head regarding this one for months, but I simply can't solve this issue.
I'm using G2 all the time. It typically works perfectly.

However, every time I try to do something big, which involves long transfers I always run into trouble, because some long transfers end in a very specific "pew" sound and the motor looses tracking.
Offending moves are around 500 long.

This never, ever happens when I work on smaller cuts - everything is flawless until I do long moves.

Right now, I have a g-code file which simply will not run without loss of tracking whatever I try.
I've tried low feed rates and low jerk, but the problem persists.

I am running the latest edge (vanilla, except configuration and pin out), but have had this problem for many versions.

Is there any chance at all there may still be some kind of issue with long moves?

Best regards

aldenhart, 2016-02-23:
Still looking into this. Thanks for your patience.

stg, 2016-03-01:
No problem.

Let me know if there is anything I can send you to aid the issue.
Kept a copy of everything this time, because it was failing consistently.
G-code, machine configuration, full source tree.

aldenhart, 2016-03-01:
The files would be helpful. Can you post or link to them somehow? It would also be helpful to know the table dimensions of your machine. --Thanks

ryansturmer, 2016-03-01:
This might be related to our issue, where we saw the bad decel in moves that weren't necessarily long, but only happened in situations with very high offset values (many dozens or hundreds of inches, basically)  - I'll see if I can pull something together.    The "ping" can be observed in this video:

https://www.youtube.com/watch?v=GAjxboUrsTU

stg, 2016-03-02:
Sorry for the delay.

I've gathered everything I believe to be useful in a zip file here.
http://wikisend.com/download/467514/g2_deccel_issue.zip

This includes videos of good/bad behaviour. My codebase. Offending g-code file. Information file with more info on the issue and exactly what (very few) changes I've made to the code (problem was there before though, and I've tried reverting my changes) as well as which settings file I am using.

I hope this is useful.
Anything else, just let me know!

@ryansturmer There is also issue #75 which seem eerily similar.

stg, 2016-03-02:
I also made minor modifications in the makefile (it's included).

Couldn't stand not being able to compile on the command like in Windows and I couldn't bring myself to install a full IDE when I already have a functional ARM toolchain :)

Fairly sure I didn't mess anything up though or that this could relate to the issue, but I missed pointing it out in the information file.

stg, 2016-04-06:
Any news on this at all? Recently, first time, ran into the same issue with job of much smaller size.

guru54321, 2016-04-23:
Hi stg, I recently ran into this issue as well. After a long move at full speed, the deccel would result in lost steps. I am in the process of setting up my machine and I discovered that only 1 of my y-axis motor would stutter though. Lowering my max speeds and feedrate solved the issue for me. Although my machine can rapid at 1500mm/min, rectifying the stutter forced me to go to around 750mm/min. I will still have to test whether changing jerk parameters can change this behavior for me. Moving at 1500mm/min is much nicer than 750mm/min! Hope this helps you as well!

ril3y, 2016-04-23:
I am not sure this is a big but could be caused by your jerk parameters ask
you said. Can you experiment with this a bit?

aldenhart, 2016-04-23:
@guru54321 - Can you please post your parameters and the move that causes the stutter? Thanks.

guru54321, 2016-04-24:
Hi  Alden, I use the following parameters (removed some lines for brevity)
[fb]  firmware build             89.03
[fbs] firmware build "                           build"
[fv]  firmware version            0.98
[cv]  configuration version       7.00
[hp]  hardware platform           3.00
[hv]  hardware version            0.00
[id]  TinyG ID       0213-0215-d453
[ja]  junction aggression         0.25
[ct]  chordal tolerance           0.0100 mm
[sl]  soft limit enable           0 [0=disable,1=enable]
[lim] limit switch enable         1 [0=disable,1=enable]
[saf] safety interlock enable     1 [0=disable,1=enable]
[mt]  motor idle timeout          2.00 seconds
[m48e] overrides enabled          1 [0=disable,1=enable]
[mfoe] manual feed override enab  0 [0=disable,1=enable]
[mfo]  manual feedrate override   1.000 [0.05 < mfo < 2.00]
[spep] spindle enable polarity    1 [0=active_low,1=active_high]
[spdp] spindle direction polarity 0 [0=CW_low,1=CW_high]
[spph] spindle pause on hold      1 [0=no,1=pause_on_hold]
[spdw] spindle dwell time         1.0 seconds
[cofp] coolant flood polarity     1 [0=low is ON,1=high is ON]
[comp] coolant mist polarity      1 [0=low is ON,1=high is ON]
[coph] coolant pause on hold      1 [0=no,1=pause_on_hold]
[tv]  text verbosity              1 [0=silent,1=verbose]
[ej]  enable json mode            0 [0=text,1=JSON]
[jv]  json verbosity              4 [0=silent,1=footer,2=messages,3=configs,4=linenum,5=verbose]
[js]  json serialize style        1 [0=relaxed,1=strict]
[qv]  queue report verbosity      1 [0=off,1=single,2=triple]
[sv]  status report verbosity     1 [0=off,1=filtered,2=verbose]
[si]  status interval           250 ms
[gpl] default gcode plane         0 [0=G17,1=G18,2=G19]
[gun] default gcode units mode    1 [0=G20,1=G21]
[gco] default gcode coord system  1 [1-6 (G54-G59)]
[gpa] default gcode path control  2 [0=G61,1=G61.1,2=G64]
[gdi] default gcode distance mode 0 [0=G90,1=G91]
[1ma] m1 map to axis              1 [0=X,1=Y,2=Z...]
[1sa] m1 step angle               1.800 deg
[1tr] m1 travel per revolution    3.0000 mm
[1mi] m1 microsteps               2 [1,2,4,8,16,32]
[1po] m1 polarity                 1 [0=normal,1=reverse]
[1pm] m1 power management         2 [0=disabled,1=always on,2=in cycle,3=when moving]
[1pl] m1 motor power level        0.375 [0.000=minimum, 1.000=maximum]
[2ma] m2 map to axis              0 [0=X,1=Y,2=Z...]
[2sa] m2 step angle               1.800 deg
[2tr] m2 travel per revolution    3.0000 mm
[2mi] m2 microsteps               2 [1,2,4,8,16,32]
[2po] m2 polarity                 1 [0=normal,1=reverse]
[2pm] m2 power management         2 [0=disabled,1=always on,2=in cycle,3=when moving]
[2pl] m2 motor power level        0.375 [0.000=minimum, 1.000=maximum]
[3ma] m3 map to axis              2 [0=X,1=Y,2=Z...]
[3sa] m3 step angle               1.800 deg
[3tr] m3 travel per revolution    3.0000 mm
[3mi] m3 microsteps               2 [1,2,4,8,16,32]
[3po] m3 polarity                 1 [0=normal,1=reverse]
[3pm] m3 power management         2 [0=disabled,1=always on,2=in cycle,3=when moving]
[3pl] m3 motor power level        0.375 [0.000=minimum, 1.000=maximum]
[4ma] m4 map to axis              1 [0=X,1=Y,2=Z...]
[4sa] m4 step angle               1.800 deg
[4tr] m4 travel per revolution    3.0000 mm
[4mi] m4 microsteps               2 [1,2,4,8,16,32]
[4po] m4 polarity                 1 [0=normal,1=reverse]
[4pm] m4 power management         2 [0=disabled,1=always on,2=in cycle,3=when moving]
[4pl] m4 motor power level        0.375 [0.000=minimum, 1.000=maximum]
-- removed motor 5 and 6 for brevity --
[xam] x axis mode                 1 [standard]
[xvm] x velocity maximum       1500 mm/min
[xfr] x feedrate maximum       1500 mm/min
[xtn] x travel minimum            0.000 mm
[xtm] x travel maximum          700.000 mm
[... 48 lines cut ...]
[chd] c homing direction          0 [0=search-to-negative, 1=search-to-positive]
[csv] c search velocity         600 deg/min
[clv] c latch velocity       100.00 deg/min
[clb] c latch backoff             5.000 deg
[czb] c zero backoff              2.000 deg
-- removed all offsetts for brevity. All offset values are 0 (zero)
tinyg [mm] ok> 

When issuing `"G91 G0 X150"` the axis moves and stops nicely
When issuing `"G91 G0 X-400"` the deceleration is bad.
after
`$xvm=1000`
`{"ej":""}`
`[xvm] x velocity maximum       1000 mm/min`
`tinyg [mm] ok>` 
`$xfr=1000`
`{"ej":""}`
`[xfr] x feedrate maximum       1000 mm/min`
`tinyg [mm] ok> `

The same command "G91 G0 X-400" executes ok. No problems during deceleration. I did only try this on my X-axis, which is single motor. When doing this on the y-axis, 1000mm/min also resulted in bad deceleration (only motor 1), hence the 750 mm/min since I do not like a skewed portal bridge..
FYI: I build the code on a linux environment based on the edge branch. Settings are based on the default settings file with only motor and axis parameters changed.
Hope this helps you guys in determining whats wrong (even if its my settings only :-) )

Mark

guru54321, 2016-04-24:
Hi Alden,

in addition to the post above:
When issuing `"G91 G0 X400"` the deceleration is also bad (with my default parameters). The direction of the move doesn't change the behavior. I'm pretty sure you'd assume that anyway, but now you're sure..

Mark

guru54321, 2016-04-24:
Hi Alden.
I've been playing around a bit with the jerk parameter and it appears there is a relationship between distance traveled and deceleration behavior. First the jerk. I have to lower my max jerk value to 4 ($xjm=4) for my machine to behave at travels 440 mm long. However, even with high jerk values, short travels all work ok. I've been testing up to 60 without any issues with short travels (less than 300mm). I've also been playing with jerk versus travel. I issued several travel commands with varying jerk values. This resulted in the next table:

Jerk	distance (mm)	result
60	300	ok
4	440	ok
5	440	nok
60	350	nok
60	325	ok
60	335	ok
60	345	ok
60	348	ok
60	350	nok
60	349	ok
50	350	ok
50	351	ok
50	360	ok
50	370	ok
50	380	nok
50	379	nok
50	375	nok
50	372	ok
50	373	ok
50	374	nok
40	374	nok
30	374	nok
20	374	ok
20	390	nok
20	380	ok
20	385	ok
20	388	nok
20	387	nok
20	386	nok
20	385	nok

As you can see, the result is not always consistent (385mm with 20 jerk once was ok, later not) but it appears that when I lower the jerk value, I can make longer travels without experiencing a bad deceleration.
Also, for your info, in all cases, the speed of the axis was 1500 mm/min (max value) when the deceleration starts. This speed is achieved after a few mm, so it's strange that a 200 mm move decelerates fine while a 400mm move isn't. The speed at which the deceleration starts is the same. 
This is all measured at different positions of the wagon as well, so as to eliminate different resistance values at different locations on the axis.
I'll be unable to reply for the next week (I'm away from the machine as well) but once back I will gladly pick up any testing requests if needed.

Mark

stg, 2016-04-28:
I have already experimented with jerk and feed-rate to try and resolve this. I typically run at @[x/y/z]jm=2000 without any issues (except for this) and I have tried lowering it all the way down to 200 to little effect. I have also experimented with feed-rates down to 200 in combination with lowered jerk. The problem persists.

I have done significant testing and found that this particular machine starts going haywire at 8000mm/m feed-rate, 4000mm/m3 jerk and decided on 6000/2000 for stable operation. Feed-rate limit is due to mid-band resonance and can be mitigated with DIY inertial dampers to reach over 10000mm/m.

With these specs it seems hard to understand why the motors would still stall at such absurdly low settings - but you are quite right, I cannot be *absolutely sure* there is nothing else going on. This *could* be a case of built up resonance interacting with the declining stepper frequency in some confounded manner. Although persistence throughout testing of non-harmonic frequencies suggest against this I do realize there may be many quite a few resonant modes in a complex and stiff construction such as this.

aldenhart, 2016-04-29:
Interesting conclusions. Have you played with the microsteps settings?

stg, 2016-04-29:
No, I have never changed my microstep setting since the first time I compiled G2.
What do you suggest I try exactly - testing other combinations of microstep/numsteps that yield the same physical results? I could certainly do that.

aldenhart, 2016-04-29:
I'm wondering if it's machine resonances then the uSteps may have a effect on that. I think the settings are at 2x now. It may be interesting to try 4x and 8x.

stg, 2016-04-29:
I have mine at 8x already, as configured on day 1.
But you mean actually changing both hardware and software configuration.
I will do that next time I open up the controller - these settings are not so easily accessible.

aldenhart, 2016-04-29:
My mistake. Yes, it may be useful to see if there are changes at 8x. BTW - Do you still have the file you sent to wikisend and can re-post? Thanks.

stg, 2016-04-29:
Yes, the file is here: http://wikisend.com/download/556396/g2_deccel_issue.zip
The link is valid for 7 days.

aldenhart, 2016-04-29:
Thanks. It was on another computer and the link had expired. I didn't know it was time limited.

stg, 2016-04-29:
Expiring links - for the good of the cybervironment. Prevents the great datavortex in the cybersea from accumulating junkfiles that may kill all the bits and bytes that live there, or even clog up the intertubes.

guru54321, 2016-04-30:
Hi All,
Today I got access to my machine again and have been playing a bit with micro steps. Any changes in the stepping do not affect the deceleration issue though. I tried both 4 and 8 times micro steps. In both cases the issue kept popping up.
Mark

Engineer3DPrototyping, 2016-05-11:
I also have had this happen. Maybe it is just my machine but I have found that decreasing jerk makes the problem worse and my motors made some horrible grinding noise until I pulled the plug. Increasing it toward shapeoko settings reduces the odds of losing position but causes the frame to deflect to much for my endmills to be happy.

diverseg, 2016-06-09:
I, as well, have some sort of an issue involving long traverses(G0 Rapid moves).
I can run complete projects and most turn out fine, however the at the end of the g-code the G0 Rapid move/traverse that returns to x0y0 consistently stops short.
I am running
firmware build 87.01
firmware version 0.98

Thinking it was speed or acceleration related, I have tried testing using faster and faster maximum velocities and jerk max for both the X and Y axis,
moving each individually moderate distances between two known reference points looking for a position error.  I am amazed how fast my axis move, scary fast, and they are consistently accurate.
example,
   g20 (inches units mode)
   g0x0
   g0x24
   g0x0
I repeat the above many times continuing to increase $xvm and never see any position error
I repeat the above many times continuing to increase $xjm and again never see any position error.
so i move to the Y axis
   g20 (inches units mode)
   g0y0
   g0y24
   g0y0
I repeat the above many times continuing to increase $yvm and never see any position error
I repeat the above many times continuing to increase $yjm and again never see any position error.

However, after trying many tests on individual axis and not seeing any issues, i tried using both axis simultaneously.
   g20 (inches units mode)
   g0x0y0
   g0x24y24
**The issue appears!** the move stops short slightly. 
I do not hear any particular strange sounds from either of my steppers.
I will do more testing as time permits.

diverseg, 2016-06-09:
I also agree the issue **rotational axis deceleration error #75** is related.

stg, 2016-09-07:
Any revelations regarding this?

I am in the process of switching out the X+Y axes on a machine, replacing the nema23 motors with Teknic servos. Preliminary calculations indicate a thrust of at least 35kg and speeds of up to 12000mm/s - enough to push some pretty heavy loads, pretty darn fast not to mention a substantial reduction in transfer time for complex jobs.

This puts an entirely new system in the loop and I'm confident it will resolve the decelleration issue for me, regardless of this being a bug or not - unless the deceleration profile is *completely* whack. Temporary glitches in timing can be fatal for the steppers, but with a servo any small errors will be "smoothed" out and if tracking is regained within 4 or 20 msec (configurable) no error will ensue.

The nice thing is that the servos provide scoping of pretty much all signals available, internal and external. so I may be able to provide some further insight into the issue. If you are interested, please let me know if there is anything specific you would like me to measure and report.

stg, 2016-09-11:
Ok, so I've confirmed that the servos solve the issue meaning there are no lost steps.
I also can say that I can't see anything obviously wrong with the decelereration profile - looks quite pretty on torque, acceleration and jerk graphs so if there is a bug it seems it would have to be a minor timing glitch.

However, probably unrelated, I am able to get missing dec-/acc-elerations if I simply manually issue "G0 X-100" "G0 X100" repeatedly in Chilipeppr. Seems to be related to the planner being (near) empty. In contrast, this graphs as a a spike on accelleration and a sharp flank on velocity, as expected.

I would either close this issue, or at the very least consider it suspect.

simonemarin, 2016-09-17:
Hallo I think I ve encountered something similar, everything is ok as long as I don t issue some repeated and overlapping commands from Chilipeppr.
You can see it here:
https://www.youtube.com/watch?v=nL2wEAmYcTo&feature=youtu.be

giseburt, 2016-10-06:
@stg or @diverseg have you retested with builds 100.x (current edge or master)? We'd like to know if this fixes the issue for either or both of you.

Thank you,
-Rob

diverseg, 2016-10-10:
@giseburt I have not retested but I am willing. 
I do not have a development environment set up to compile at the moment. 
Is there a current master 100.x build .bin file somewhere for Due+gShield?
Note: I do not see one at http://synthetos.github.io/g2/
If this were available I could test this this ASAP and report my findings.

guru54321, 2016-10-10:
I have currently the 100.x master compiled and still can reproduce the issue, although it is less prevalent. I seem to be able to fine tune my machine a bit faster (the issue pops up by speeds over 1100 mm/min).  Where set a max of 750 mm/min before I now trust the 1000 mm/min. which is an improvement.

guru54321, 2016-10-10:
If there is anything I can do to help debugging this, please let me know.

giseburt, 2016-10-10:
Thank you @guru54321. Just to be 100%, this ONLY happens during
deceleration? Never during an acceleration, and never during short moves?

-Rob

giseburt, 2016-10-10:
@guru54321, if you could provide these things it would be very helpful:

* What branch/commit/tag you compiled
* what CONFIG and/or BOARD you used
* what settings file you used and and settings you changed
* what settings were set at runtime
* a move that works fine (the G0 X-500 or whatnot)
* a move that fails

Any other conditions required to recreate the problem.

I don't have the same machine so I'll have to look for it from a debugger
and see if I can catch it that way.

Thank you!

stg, 2016-10-10:
I would very much like to test this again.

Since I already have a working toolchain for ARM I am very much not interested in installing a huge development suite just to be able compile. I've previously had to modify your Makefiles to allow regular command line compilation on Windows, but this time I am having more difficult issues.

This is my process:

1) Get g2core and motate in place
2) Set up a symbolic link in motate/tools/win32 to point to my tools (this is just dumb imo, my tools are on the path and should be used unless otherwise instructed)
3) Set up boards.mk according to my configuration:
```
ifeq ("$(CONFIG)","lilDemon")
    ifeq ("$(BOARD)","NONE")
        BOARD=Due
    endif
    SETTINGS_FILE="settings_lilDemon.h"
endif
´´´
4) Set up settings_lilDemon.h
4) Compile with make CONFIG=lilDemon

This invariably results in:
```
"Found that we're in WINDOWS"
Compiling cpp ./canonical_machine.cpp
    -> build/ly6040-Due/./canonical_machine.o
In file included from ./stepper.h:249:0,
                 from ./canonical_machine.cpp:99:
./board/ArduinoDue/board_stepper.h:32:29: fatal error: step_dir_driver.h: No such file or directory
compilation terminated.
make: *** [build/lilDemon-Due/./canonical_machine.o] Error 1
´´´

In fact, trying to compile any combination using BOARD=Due will result in this error.
I am able to compile at least several other configurations where BOARD is g2v9k, printrboardG2v3 or gquintic-a.

Would much appreciate assistance with this and would gladly in return write an article on "How to compile G2 on Windows without installing 6 gigabytes of software you don't need" ;)

giseburt, 2016-10-10:
Hi @stg,

You've run into an interesting issue. The `BOARD=Due` isn't supposed to work, but once I tested it I found that it in fact does. It's Motate that has a `BOARD=Due` target, and I will likely remove that shortly to keep this from happening again.

Use [`BOARD=gShield`](https://github.com/synthetos/g2/blob/edge/g2core/board/ArduinoDue.mk#L26) in your `CONFIG` settings.

I'm sorry for the Due confusion, and thank you for testing this.

-Rob

PS: The symlink at `motate/tools/win32` is working around us explicitly preventing locally-installed tools from being used (including those in Atmel Studio). As long as your tools are recent, that's fine, but if you have issues with compiling at least make sure that the tools you have installed are the same version [we are using](https://github.com/synthetos/Motate/blob/master/Tools/Makefile#L33-L44). (We've had issues with old or broken tools being used, frustrating everyone involved. This also allows us to utilize a reasonably modern compiler.) BTW, you do **not** need Atmel Studio, which I agree is insanely huge. In fact, we don't even use the build tools that come with Atmel Studio, making the space wasted even worse.

It's hard to find a good cross-platform OSS (or free) embedded GUI to support. We're still looking...

giseburt, 2016-10-10:
Hello everyone,

I just pushed branch [dev-123-bad-decel](https://github.com/synthetos/g2/compare/dev-123-bad-decel) that I believe fixes this issue. Could anyone who can test please confirm? Thank you!

  -Rob

guru54321, 2016-10-10:
Hi Gisebrt, I'll try and do a build. (I'll have to brush up my git-fu a bit) I'll try to post any results tomorrow. It's already late over here in Europe :-)

guru54321, 2016-10-10:
Ok, I was really eager to test this out and git is not that hard :-) Did a quick rebuild, uploaded the fw and did som quick moves. For me the issue seems to be fixed. I can now run my machine twice as fast (1500 mm/min instead of 750) Any faster than this and my Nema-23s give it up due to not enough torque. 
Thanks for fixing this!

- Mark

stg, 2016-10-11:
Ok, since I have modified my machine heavily I can no longer test the exact original issue.

However, I've been having issues with my Z axis that I'm quite confident are related.
So a while back I created the nc-file-from-hell to give the Z axis a nice ordeal to work through.
It is basically 15 minutes of torture and this file has ***never*** passed the test reliably.

So - yesterday I was running at jerk:750 and velocity:1000, which has worked well for everyday use but would barely pass the test file half of the time.

Today I am running at jerk:5000 and velocity:5000 and it is passing ***everything*** I throw at it - repeatedly and reliably.

Whatever you've done, you've done very good :)

I believe re-iteration is appropriate, given the impressive improvement in my case:

Jerk bumped from 750 to 5000
Velocity bumped from 1000 to 5000
Reliability bumped from "wat?" to "walks through the fires of hell unscathed"

Also, given that this is a Z axis without gas springs, the load is highly asymmetrical (hard to lift, effortless to lower).
Now, when the axis finally hits it's limits, it always drops down - which makes perfect sense.
Before upgrading, the axis would typically have moved upwards which was good indication that the problem was not exclusively torque/speed related.

@giseburt
Thank you for your help - I do believe I owe you some wiki article'ing (if you don't mind).
Regarding the symlink, I hear you and I completely understand - though I would have preferred an automatic version check and ***option*** to override the default location.
By the way, this is the first G2 makefile that compiles for me without modification - happy about that, as well as the new project structure.

diverseg, 2016-10-14:
I have not retested but I am working on it. 
I have installed a development environment (Atmel Studio 7 on my windows 10 desktop) to compile. 
I have git installed and now have cloned the edge dev-123-bad-decel
Using Atmel Studio 7, I opened g2core.cppproj
Since i have the Due i changed the processor/device to ATSAM3X8E
i selected the gShield configuration and built g2core... (BOARD=gShield for Due+gShield)
Set up boards.mk according to my configuration:
ifeq ("$(CONFIG)","Ox")
    ifeq ("$(BOARD)","NONE")
        BOARD=g2v9k
    endif
    SETTINGS_FILE="settings_ox.h"
endif

Set up settings_ox.h with my known working configuration settings
Compile with make CONFIG=Ox

in C:/Users/Steve/Documents/G2Core/g2core/bin/ there is now Ox-gShield/g2core.bin at 154KB
I try updating thru chilipeppr's SPJS using Arduino Due Programming port 
and 
URL file:///C:/Users/Steve/Documents/G2Core/g2core/bin/Ox-gShield/g2core.bin
Progress goes to download 100% 
but hangs at Program 0% 
Making sure /dev/ttyACM0 port is closed so we have access to it for programming...
Programming...

if i use chilipeppr's SPJS using Arduino Due Programming port URL
and
URL http://synthetos.github.io/g2/binaries/TinyG2_Due-edge-078.03-default.bin
or http://storage.googleapis.com/chilipeppr-bucket/TinyG2-gShield-087edge.bin
the Due programs properly.

I would like to program with a .bin of branch edge dev-123-bad-decel compiled for a Due+gShield
I'm not worried what the config settings are as i could update them with the Ox settings thru a script until i can findout what is wrong with my process.
**If one of you have a working .bin of branch edge dev-123-bad-decel compiled for a Due+gShield they could get me I would be happy to test this as well and post the results.**

Also, @stg I would be very interested in an article on "How to compile G2 on Windows without installing the Atmel Studio 7 software

stg, 2016-10-15:
https://github.com/synthetos/g2/wiki/Compiling-and-configuring-G2-on-Windows

I hope it is not terrible.

giseburt, 2016-10-15:
@diverseg We are working on a new updater/flashing UI. Here are some docs that are less user friendly but will get the firmware on the board: https://github.com/synthetos/g2/wiki/Flashing-G2-with-Windows

commit 510f015a (2016-10-24): Attempt to fix #123.
commit a3db4bb8 (2016-12-12): Attempt to fix #123.
closed 2016-10-24 by giseburt